Hamlib/Hamlib #813: Add password to rigctld

mdblack98, 2021-09-30:
In order to implement some security for using rigctld over the internet introduce a password mechanism.  If a password is set on rigctld then a connecting client must provide password via a passwd command.

Expected behavior
rigctld --password thisisapassword
rigctl --password thisisapassword

Eventually should do this via openssl or such.

N0NB, 2021-10-02:
I think we have avoided this for so long as even though authentication is reasonably straightforward, adding encryption looks like staring into the abyss to me!

I've preferred to direct those who needs such things to SSH as it handles authentication in a secure manner--password or public key--and encryption.  Add to that its tunneling capabilities and it takes care of all we need.  What we don't have is a guide to setting up SSH to be used in this manner.  Examples abound on the Internet but sometimes key details are mixed up.  More advanced use involves reverse tunnels which I have setup to my AllStar repeater so that as long as it can get an outbound network connection it will contact a host in my LAN (firewall port opened, etc.) and set up the tunnel.  On that host I can make a connection to `locahost:port` and be connected.

Just some thoughts.

mdblack98, 2021-10-02:
The need for encryption is way overblown.
Right now there's no security at all so there is a potential hole for any remote connection -- even if done via port redirection which I know some do.A simple password solves 99.9999% of problems -- no encryption needed -- and I'm going to use a self-generated password so it is pseudo-random and users don't reuse some common password.
So it will be something like this:
rigctld --set-passwordpassword: XXXXXXXXThen add a --use-password XXXXXXX option that expects a certain format as rigctld provides. 
So yes...if somebody happens to monitoring the packets the password is in clear text...but nob ody is doing that between a ham operator and their remote rig.  But an open port is subject to scanning.
Mike

mdblack98, 2022-04-29:
Need to make password per connection.