Hamlib/Hamlib #835: RIG_VFO_OTHER not handled in rig_get_cache

mdblack98, 2021-10-22:
[SYSLOG][2021-10-22 00:46:19.548179][00:00:00.000468][info] Log Start
[SYSLOG][2021-10-22 00:46:19.548179][00:00:00.000506][info] WSJT-X   v2.5.1 f30d2c  by K1JT, G4WJS, K9AN, and IV3NWV - Program startup
[SYSLOG][2021-10-22 00:46:19.548179][00:00:00.002876][info] locale: language: English script: Latin country: United2021-10-22 00:46:19.548179][00:00:00.000506][info] WSJT-X   v2.5.1 f30d2c  by K1JT, G4WJS, K9AN, and IV3NWV - P States ui-languages: en-US
[SYSLOG][2021-10-22 00:46:19.548179][00:00:00.003005][info] Loaded Qt translations for current locale from resources
[SYSLOG][2021-10-22 00:46:19.548179][00:00:00.003038][info] Loaded WSJT-X base translation file from :/Translations based on language en
[SYSLOG][2021-10-22 00:46:19.548179][00:00:00.003063][info] Loaded WSJT-X translations for current locale from resources
[SYSLOG][2021-10-22 00:46:19.563804][00:00:00.017083][info] shmem size: 48275456
[RIGCTRL][2021-10-22 00:46:19.579444][00:00:00.032455][info] Hamlib version: Hamlib 4.4~git Wed Oct 20 21:21:43 2021 +0000 SHA=655337
[RIGCTRL][2021-10-22 00:46:21.451673][00:00:01.900719][error] icom_get_freq: unknown VFO?  VFO=None
[RIGCTRL][2021-10-22 00:46:21.451673][00:00:01.900846][error] rig_get_cache: unknown vfo?, vfo=otherVFO
[RIGCTRL][2021-10-22 00:46:21.451673][00:00:01.900886][error] icom_set_vfo: unsupported VFO otherVFO
[RIGCTRL][2021-10-22 00:46:21.520673][00:00:01.970025][error] rig_get_cache: unknown vfo?, vfo=otherVFO
[RIGCTRL][2021-10-22 00:46:21.583168][00:00:02.032196][error] rig_get_vfo: no get_vfo
[SYSLOG][2021-10-22 00:46:44.398701][00:00:24.845757][warning] Detected dropped audio source samples: 624 (0.013 S)[SYSLOG][2021-10-22 00:54:19.628599][00:08:00.043419][info] Log Finish

g4wjs, 2021-11-01:
Mike,

can I test this fix? I have an IC_910H and an IC-756.

73.

mdblack98, 2021-11-01:
Sure...go ahead...

g4wjs, 2021-11-01:
Hi Mike,

OK, my question was asking if it is applicable to either of those rigs? 
Also do I need to do anything particular to reproduce it?

73
Bill
G4WJS.

On 01/11/2021 12:44, Michael Black wrote:
> Sure...go ahead...
>
>
>
> On Monday, November 1, 2021, 07:37:50 AM CDT, Bill Somerville 
> ***@***.***> wrote:
>
>
>
>
> Mike,
>
> can I test this fix? I have an IC_910H and an IC-756.

mdblack98, 2021-11-01:
Actually I was just testing something else completely unrelated and now I can reproduce the problem so it['s not fixed yet.
It appears to me to apply to all Icom rigs if you want to see if you have the error too.l

g4wjs, 2021-11-01:
Hi Mike,

OK, will do.

73
Bill
G4WJS.

On 01/11/2021 12:59, Michael Black wrote:
> Actually I was just testing something else completely unrelated and
> now I can reproduce the problem so it['s not fixed yet.
> It appears to me to apply to all Icom rigs if you want to see if you [...]

mdblack98, 2021-11-01:
Should have that one fixed today.
Have to find a Yaesu owner to test https://github.com/Hamlib/Hamlib/issues/838
Mike

 
 

Hi Mike,

I confirm this is reproducible on both the IC-910H and the IC-756. It 
only happens once just after startup. More details like logs available 
is that helps.

closed 2021-11-03 by mdblack98