Hamlib/Hamlib #858: FT-991A unsupported VFO

mdblack98, 2021-11-07:
[2021-11-06 17:36:01.622247][00:00:02.674691][RIGCTRL:debug] rig_get_freq(2098) called vfo=currVFO
[2021-11-06 17:36:01.622343][00:00:02.674832][RIGCTRL:trace] vfo_fixup: vfo=currVFO, vfo_curr=VFOA
[2021-11-06 17:36:01.622406][00:00:02.674846][RIGCTRL:trace] vfo_fixup: Leaving currVFO alone
[2021-11-06 17:36:01.622413][00:00:02.674930][RIGCTRL:debug] rig.c(2107) vfo=currVFO, curr_vfo=VFOA
[2021-11-06 17:36:01.622520][00:00:02.674962][RIGCTRL:trace] rig_get_freq: VFOA cache hit age=39ms, freq=14074000
[2021-11-06 17:36:01.622540][00:00:02.674982][RIGCTRL:trace] rig_get_freq: elapsed=1ms
[2021-11-06 17:36:01.622549][00:00:02.675054][RIGCTRL:debug] rig.c(2170):rig_get_freq return(0)
[2021-11-06 17:36:01.622624][00:00:02.675098][RIGCTRL:trace] rig_get_freq frequency=14074000
[2021-11-06 17:36:01.622796][00:00:02.675310][RIGCTRL:trace] frequency: 14074000
[2021-11-06 17:36:01.622993][00:00:02.675549][RIGCTRL:debug] rig_get_freq(2098) called vfo=Sub
[2021-11-06 17:36:01.623269][00:00:02.675773][RIGCTRL:trace] vfo_fixup: vfo=Sub, vfo_curr=VFOA
[2021-11-06 17:36:01.623347][00:00:02.675851][RIGCTRL:debug] rig.c(2107) vfo=Sub, curr_vfo=VFOA
[2021-11-06 17:36:01.623437][00:00:02.675954][RIGCTRL:trace] rig_get_freq: cache miss age=10101ms, cached_vfo=Sub, asked_vfo=Sub
[2021-11-06 17:36:01.623609][00:00:02.676098][RIGCTRL:debug] flrig.c(1105):flrig_get_freq entered
[2021-11-06 17:36:01.623741][00:00:02.676182][RIGCTRL:trace] flrig_get_freq: vfo=Sub
[2021-11-06 17:36:01.623795][00:00:02.676251][RIGCTRL:debug] flrig.c(265):check_vfo return(0)
[2021-11-06 17:36:01.623896][00:00:02.676400][RIGCTRL:error] flrig_get_freq: unsupported VFO Sub
[2021-11-06 17:36:01.624065][00:00:02.676645][RIGCTRL:debug] flrig.c(1114):flrig_get_freq return(-1)
[2021-11-06 17:36:01.624248][00:00:02.676706][RIGCTRL:trace] rig_get_freq: elapsed=3ms
[2021-11-06 17:36:01.624336][00:00:02.676840][RIGCTRL:debug] rig.c(2279):rig_get_freq return(-1)
[2021-11-06 17:36:01.624501][00:00:02.676990][RIGCTRL:error] error: Invalid parameter
rig_get_freq(2276): freqMainB=0, modeMainB=, widthMainB=0
rig_get_freq: elapsed=3ms
rig.c(2279):rig_get_freq return(-1)
[2021-11-06 17:36:01.624993][00:00:02.677454][RIGCTRL:trace] startup what: Hamlib error: Invalid parameter
rig_get_freq(2276): freqMainB=0, modeMainB=, widthMainB=0
rig_get_freq: elapsed=3ms
rig.c(2279):rig_get_freq return(-1) while getting other VFO frequency
[2021-11-06 17:36:01.625037][00:00:02.677510][RIGCTRL:trace] reason: Hamlib error: Invalid parameter
rig_get_freq(2276): freqMainB=0, modeMainB=, widthMainB=0
rig_get_freq: elapsed=3ms
rig.c(2279):rig_get_freq return(-1) while getting other VFO frequency
[2021-11-06 17:36:01.625201][00:00:02.677674][SYSLOG:error] handle_transceiver_failure: reason: Hamlib error: Invalid parameter
rig_get_freq(2276): freqMainB=0, modeMainB=, widthMainB=0
rig_get_freq: elapsed=3ms
rig.c(2279):rig_get_freq return(-1) while getting other VFO frequency

Johnny-Sierra, 2021-11-12:
I'm working on a radio GUI and just find this issue with my FT-991A.
By rig CAT manual, should be possible to read/write separately VFOA and B, CAT command FA and FB. But hamlib 4.3.1 supports only current VFO. Please update the frontend.

Some notes: the 991 has a single receiver and the current VFO is always the VFOA, the VFOB is used only to trasmit when in split mode. When VFO swap function is used, the focus doesn't change from one VFO to another as other radio models, but as said before, A becames B and viceversa, so the active VFO is only VFOA.

mdblack98, 2021-11-13:
There is no "set_vfo" for the FT991 -- but it does have targetable freq and mode.
So all you have to do is 
rig_set_freq(rig, RIG_VFO_A, freq);
rig_set_mode(rig, RIG_VFO_A, mode, RIG_PASSBAND_NOCHANGE);
rig_set_freq(rig, RIG_VFO_B, freq);
rig_set_mode(rig, RIG_VFO_B, mode, RIG_PASSBAND_NOCHANGE);

mdblack98, 2021-11-13:
OH...and from rigctld you need to add the --vfo option to the command line so you can send the vfo with the command.
f VFOA 14074000
f VFOB 14075000
m VFOA USB -1
m VFOB USB -1
The -1 means no passband change.

I suppose we could add a "virtual VFO" to the 991 -- we've done that in other places where the rig is fixed and it's just for the API like rigctld so it does not need the --vfo argument.

Johnny-Sierra, 2021-11-13:
Hi Michael,
yes, I use -o option in the rigctld.

The issue is that set or get command you mentioned, with VFOB parameter doesn't work.

For example:
`rig_get_freq(my_rig, RIG_VFO_B, &freqB);`

Return error 11: Function not available. The same for rig_get_mode

What is strange is that using rigctl 
`f VFOB`
it works!

Johnny-Sierra, 2021-11-13:
Also `rig_vfo_op(my_rig, RIG_VFO_A, RIG_OP_XCHG)` doesn't work, error 5. Same with RIG_OP_CPY, both using library or rigctl

mdblack98, 2021-11-13:
Can you please send me debug logs?

Johnny-Sierra, 2021-11-14:
rigctl connected to rigctld with option -o, v. 4.1.3

Rig command: G VFOA XCHG
rigctl_parse: input_line: G VFOA XCHG
rig_parse_vfo called
rigctl_parse: vfo_opt=1
rigctl_parse.c(3615):rigctl_vfo_op entered
rig_parse_vfo_op called
rig.c(5772):rig_vfo_op entered
rig.c(5739):rig_has_vfo_op entered
rig.c(5747):rig_has_vfo_op return(2)
netrigctl_vfo_op called
netrigctl_transaction: called len=7
rig_flush: called for network device
network_flush called
write_block(): TX 7 bytes
0000    4a 20 58 43 48 47 0a                                J XCHG.         
read_string called, rxmax=1024
read_string(): Timed out 10.009 seconds after 0 chars
rig.c(5790):rig_vfo_op return(-5)
rigctl_parse.c(3632):rigctl_vfo_op return(-5)
vfo_op: error = Communication timed out
read_string(): Timed out 10.009 seconds after 0 chars
rig.c(5790):rig_vfo_op return(-5)
rigctl_parse.c(3632):rigctl_vfo_op return(-5)
rigctl_parse.c(1758):rigctl_parse return(-5)
main: i/o error
rig.c(1124):rig_close entered
netrigctl_close called
netrigctl_transaction: called len=2
rig_flush: called for network device
network_flush called
write_block(): TX 2 bytes
0000    71 0a                                               q.              
read_string called, rxmax=1024
read_string(): RX 8 characters
0000    52 50 52 54 20 2d 31 0a                             RPRT -1.        
netrigctl_close: close error Invalid parameter
read_string(): RX 8 characters
0000    52 50 52 54 20 2d 31 0a                             RPRT -1.        
netrigctl_close: close error Invalid parameter
read_string(): RX 8 characters
0000    52 50 52 54 20 2d 31 0a                             RPRT -1.        
0000    52 50 52 54 20 2d 31 0a                             RPRT -1.        
iofunc.c(200):port_close entered
network.c(379):network_close entered
network_close: close socket ret=0
network.c(402):network_close return(0)
iofunc.c(234):port_close return(0)
rig.c(243):remove_opened_rig return(0)
rig.c(1280):rig_close return(0)
main: rig_close retcode=0
rig.c(684):rig_open entered
rig_open: using network address 127.0.0.1:4532
iofunc.c(68):port_open entered
network.c(171):network_open entered
network_open: hoststr=127.0.0.1, portstr=4532
network_open: client port=30875
network.c(308):network_open return(0)
iofunc.c(186):port_open return(0)
rig.c(220):add_opened_rig return(0)
netrigctl_open called
netrigctl_transaction: called len=9
rig_flush: called for network device
network_flush called
write_block(): TX 9 bytes
0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
read_string called, rxmax=1024
read_string(): RX 2 characters
0000    31 0a                                               1.              
netrigctl_open: vfo_mode=1
netrigctl_transaction: called len=12
rig_flush: called for network device
network_flush called
write_block(): TX 12 bytes
0000    5c 64 75 6d 70 5f 73 74 61 74 65 0a                 \dump_state.   
...

mdblack98, 2021-11-14:
Try the latest master if you would please.We're working on this.
Mike W9MDB

mdblack98, 2021-11-14:
Try the current master and see if it works now.  Does 1 retry.
Mike W9MDB

Johnny-Sierra, 2021-11-14:
Just compiled hamlib-4.4~git-4e717d23-20211114.tar.gz 
<http://n0nb.users.sourceforge.net/hamlib-4.4~git-4e717d23-20211114.tar.gz>

In rigctl:

- set and get_vfo now are working;

- No change for vfo_op, still not working with same error 5.

Using dll

- get_freq now is working also for VFOB;

- vfo_op not working.

Johnny IZ8EWD

mdblack98, 2021-11-14:
XCHG should be working as of most recent commit.Any other VFO_OPS that don't work?
Mike W9MDB

 
 

<http://n0nb.users.sourceforge.net/hamlib-4.4~git-4e717d23-20211114.tar.gz>

Johnny-Sierra, 2021-11-14:
Update:

- set and get_vfo now are working;

- No change for vfo_op, still not working with same error 5;

- set and get_mode work for VFOA and currVFO=VFOA, don't work for VFOB 
and currVFO=VFOB.

Il 14/11/2021 18:53, Michael Black ha scritto:
> Try the current master and see if it works now.  Does 1 retry.
> Mike W9MDB

Johnny-Sierra, 2021-11-14:
I test only XCHG, but with commit 4e717d23.

Anyway, tomorrow I will try again with your last commit.

TNX

Johnny-Sierra, 2021-11-15:
Test with snapshot f044c2bf.
vfo_op command still not working for any of the supported operations.

```
vfo_op: error = Communication timed out
read_string(): Timed out 10.010 seconds after 0 chars
rig.c(5983):rig_vfo_op return(-5)
rigctl_parse.c(3639):rigctl_vfo_op return(-5)
```

mdblack98, 2021-11-15:
I need a debug log please.
Mike W9MDB

Johnny-Sierra, 2021-11-16:
Snapshot d9d247

Rig command: G
rigctl_parse: input_line: G
VFO: currVFO
rig_parse_vfo called
Mem/VFO Op: CPY
rigctl_parse: vfo_opt=1
rigctl_parse.c(3622):rigctl_vfo_op entered
rig_parse_vfo_op called
rig.c(5990):rig_vfo_op entered
rig.c(5957):rig_has_vfo_op entered
rig.c(5965):rig_has_vfo_op return(1)
netrigctl_vfo_op called
netrigctl_transaction: called len=6
rig_flush: called for network device
network_flush called
write_block(): TX 6 bytes
0000    4a 20 43 50 59 0a                                   J CPY.          
read_string called, rxmax=1024
read_string(): Timed out 10.010 seconds after 0 chars
rig.c(6008):rig_vfo_op return(-5)
rigctl_parse.c(3639):rigctl_vfo_op return(-5)
vfo_op: error = Communication timed out
read_string(): Timed out 10.010 seconds after 0 chars
rig.c(6008):rig_vfo_op return(-5)
rigctl_parse.c(3639):rigctl_vfo_op return(-5)
rigctl_parse.c(1762):rigctl_parse return(-5)
main: i/o error
rig.c(1156):rig_close entered
netrigctl_close called
netrigctl_transaction: called len=2
rig_flush: called for network device
network_flush called
write_block(): TX 2 bytes
0000    71 0a                                               q.              
read_string called, rxmax=1024
read_string(): RX 8 characters
0000    52 50 52 54 20 2d 31 0a                             RPRT -1.        
netrigctl_close: close error Invalid parameter
read_string(): RX 8 characters
0000    52 50 52 54 20 2d 31 0a                             RPRT -1.        
netrigctl_close: close error Invalid parameter
read_string(): RX 8 characters
0000    52 50 52 54 20 2d 31 0a                             RPRT -1.        
0000    52 50 52 54 20 2d 31 0a                             RPRT -1.        
iofunc.c(200):port_close entered
network.c(381):network_close entered
network_close: close socket ret=0
network.c(404):network_close return(0)
iofunc.c(234):port_close return(0)
rig.c(272):remove_opened_rig return(0)
rig.c(1312):rig_close return(0)
main: rig_close retcode=0
rig.c(713):rig_open entered
rig_open: using network address 127.0.0.1:4532
iofunc.c(68):port_open entered
network.c(171):network_open entered
network_open: hoststr=127.0.0.1, portstr=4532
network_open: client port=57579
network.c(310):network_open return(0)
iofunc.c(186):port_open return(0)
rig.c(249):add_opened_rig return(0)
netrigctl_open called
netrigctl_transaction: called len=9
rig_flush: called for network device
network_flush called
write_block(): TX 9 bytes
0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
read_string called, rxmax=1024
read_string(): RX 2 characters
0000    31 0a                                               1.              
netrigctl_open: vfo_mode=1
netrigctl_transaction: called len=12
rig_flush: called for network device
network_flush called
write_block(): TX 12 bytes
0000    5c 64 75 6d 70 5f 73 74 61 74 65 0a                 \dump_state.  
...
```

Johnny-Sierra, 2021-11-20:
It is a general issue, see #874

Johnny-Sierra, 2021-11-23:
Hi, the VFOs control on Yaesu 991 now is working well.
I think the issue could be closed. TNX

closed 2021-11-23 by mdblack98