Hamlib/Hamlib #860: rigctld does not handle USB-errors

OE4AMW, 2021-11-08:
I am using rigctld with a FT-857D with an USB-adapter using following command line:
`rigctld -m 122 -r /dev/ttyUSB0  --set-conf=serial_speed=9600  -vvv`

When the USB adapter is disconnected (or fails), rigctld logs an error:
`write_block():491 failed -1 - Input/output error`
but continues reading:
```
...
[pid 17598] read(5, "", 5)              = 0
[pid 17598] select(6, [5], NULL, [5], {tv_sec=0, tv_usec=200000}) = 1 (in [5], left {tv_sec=0, tv_usec=199999})
[pid 17598] read(5, "", 5)              = 0
[pid 17598] select(6, [5], NULL, [5], {tv_sec=0, tv_usec=200000}) = 1 (in [5], left {tv_sec=0, tv_usec=199999})
[pid 17598] read(5, "", 5)              = 0
[pid 17598] select(6, [5], NULL, [5], {tv_sec=0, tv_usec=200000}) = 1 (in [5], left {tv_sec=0, tv_usec=199999})
[pid 17598] read(5, "", 5)              = 0
[pid 17598] select(6, [5], NULL, [5], {tv_sec=0, tv_usec=200000}) = 1 (in [5], left {tv_sec=0, tv_usec=199999})
[pid 17598] read(5, "", 5)              = 0
[pid 17598] select(6, [5], NULL, [5], {tv_sec=0, tv_usec=200000}) = 1 (in [5], left {tv_sec=0, tv_usec=199999})
[pid 17598] read(5, "", 5)              = 0
[pid 17598] select(6, [5], NULL, [5], {tv_sec=0, tv_usec=200000}) = 1 (in [5], left {tv_sec=0, tv_usec=199999})
[pid 17598] read(5, "", 5)              = 0
[pid 17598] select(6, [5], NULL, [5], {tv_sec=0, tv_usec=200000}) = 1 (in [5], left {tv_sec=0, tv_usec=199999})
[pid 17598] read(5, "", 5)              = 0
^C[pid 17598] select(6, [5], NULL, [5], {tv_sec=0, tv_usec=200000}strace: Process 17099 detached
strace: Process 17598 detached
 <detached ...>
```
on the already deleted filedescriptor:
```
ls -l  /proc/17099/fd
insgesamt 0
lrwx------ 1 user user 64 Nov  8 21:15 0 -> /dev/pts/12
lrwx------ 1 user user 64 Nov  8 21:15 1 -> /dev/pts/12
lrwx------ 1 user user 64 Nov  8 21:15 2 -> /dev/pts/12
lrwx------ 1 user user 64 Nov  8 21:15 3 -> 'socket:[391807]'
lrwx------ 1 user user 64 Nov  8 21:15 4 -> 'socket:[395790]'
lrwx------ 1 user user 64 Nov  8 21:15 5 -> '/dev/ttyUSB0 (deleted)'
```
During that time rigctld consumes 100% of a CPU-core:
```
    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     ZEIT+ BEFEHL                                                                                                     
  17099 user     20   0   88880   6640   3700 S 100,0   0,0   0:18.27 rigctld
```
and does not send any responses on the TCP-socket (while incoming packets are ACKed on TCP-level).

In this state, rigctld does does not react on "Ctrl+C" (and must be terminated using kill).

Found on libhamlib-utils 3.3-10build1 on Ubuntu.
Verified with today's master (manually compiled b17d681fc28f48e6bda9103f78552c348c87ab56).

as3ii, 2023-10-08:
I found a similar problem, where if the serial disconnects or the radio turns off, rigctl (4.5.5) gets stuck probably in an infinite loop inside or around `rig_get_freq()` saturating a CPU core and never recovering (even if the serial come back alive)

Output example:
```
Rig command: f
 1:rigctl_parse.c(2098):rigctl_get_freq entered
rig_get_freq(2033) called vfo=currVFO
rig.c(2042) vfo=currVFO, curr_vfo=VFOA
rig_get_freq(2129): vfo_opt=0, model=1020
ft817_get_freq: called, vfo=VFOA, ptt=0, split=0
ft817_get_status: called
read_string_generic(): read failed, direct=1 - Success
tcflush
write_block():1054 failed -1 - Input/output error
read_block_generic called, direct=1
[here the output stops and it starts eating a lot of CPU cycles]
```

mdblack98, 2023-10-08:
What OS?
Mike  W9MDB

as3ii, 2023-10-08:
I'm running arch linux, kernel 6.5.5

as3ii, 2023-10-08:
Let me know if and how I can help debugging this issue

mdblack98, 2023-10-08:
I think this may be fixed in the latest Hamlib.
https://n0nb.users.sourceforge.net/

Mike W9MDB

as3ii, 2023-10-08:
I've compiled and tested the master branch of this repo: I can confirm that the bug is still present