Hamlib/Hamlib #862: Issue in compiling yesterday GIT version on Raspberry PI3+

UgoPoddine, 2021-11-09:
Hello,

I'm trying to install yesterday version (from GIT) of  Hamlib, from source, on Raspberrt PI3+ running PI OS 32bits "bullseye" (5.10.63-v7+).
Configuration and compilation is successful, installation and "make check" is ok, ldconfig seems ok, but when I try : 

rigctl -V

the result is :

rigctld: symbol lookup error: rigctld: undefined symbol: debugmsgsave

What I'm doing wrong ?
Any help will be appreciated
Than-you, 73
IU1IPB Ugo

mdblack98, 2021-11-20:
That means you have and old libhamlib.so around somewhere that doesn't contain the new code.

closed 2021-11-20 by mdblack98

UgoPoddine, 2021-11-22:
Hello mdblack98,

thank.you for your support.
I was afraid about that because I have had installed libhamlib from Debian repository on this same machine in the past.
But :

a) the only libhamlib.so is located in ./usr/local/lib/libhamlib.so, simlink to libhamlib.so.4.0.4 
b) sudo apt-get remove libhamlib* returns for all packages : Package 'libhamlib2' is not installed, so not removed and so on

It seems that no multiple libhamlib.so is there...

Thank-you fot your help
U

mdblack98, 2021-11-22:
Just to prove it's there do this
nm src/rig.o | grep debugmsg
0000000000000000 D debugmsgsave
0000000000005dc0 D debugmsgsave2
000000000000bb80 D debugmsgsave3

Also do this to find any stray libhamlib
find / -name "libhamlib*"

You can also try configuring as a static build
./configure CFLAGS="-static -g -O0 -fPIC" --no-create --no-recursion