Hamlib/Hamlib #869: FreeBSD build for 4.x?

techn0mad, 2021-11-14:
As the FreeBSD port is way back at 3.3, I thought I'd try building a more current version:

```
% uname -a
FreeBSD pinawa 13.0-RELEASE-p3 FreeBSD 13.0-RELEASE-p3 #0: Tue Jun 29 19:46:20 UTC 2021     root@amd64-builder.daemonology.net:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

% c++ --version
FreeBSD clang version 11.0.1 (git@github.com:llvm/llvm-project.git llvmorg-11.0.1-0-g43ff75f2c3fe)
Target: x86_64-unknown-freebsd13.0
Thread model: posix
InstalledDir: /usr/bin

% git log --pretty=oneline -5
a67f885c853c9824cd021123c70375fca4911d30 (HEAD -> master, origin/master, origin/HEAD) Implement SV command in newcat.c https://github.com/Hamlib/Hamlib/issues/858#
470d3b7c4fc22fbb7b8883c5065d9b927f48b3fe All icom_open to do 1 retry https://github.com/Hamlib/Hamlib/issues/868
4e717d2386201845da77e2ffff4fa0ef182ac1f7 Add set/get_vfo for ft991 so rigctld can be used wihout --vfo argument Hopefully fixes rig_vfo_op function too https://github.com/Hamlib/Hamlib/issues/858
dbb9f90abf7ccae3c9b5c603a1180524a26f4047 Remove VFOA/B copy from ft1000mp_rig_set_split_vfo https://github.com/Hamlib/Hamlib/issues/867
d6110a8b8bb8f62c745556a2784a57334d80f467 Add rx_vfo to state structure More to be added to implement this https://github.com/Hamlib/Hamlib/issues/866

% ./configure
checking for gcc... no
checking for cc... cc
checking whether the C compiler works... yes
. . .
checking whether to build USB dependent backends... yes
checking for libusb_init in -lusb-1.0... no
configure: WARNING: libusb_init was not found in libusb-1.0--USB backends will be disabled
checking libusb.h usability... yes
. . .
checking readline/history.h usability... no
checking readline/history.h presence... no
checking for readline/history.h... no
checking whether to use INDI in rigctl/rotctl... checking for nova library... checking libnova/libnova.h usability... no
checking libnova/libnova.h presence... no
checking for libnova/libnova.h... no
no
configure: WARNING: libnova support not found, required by INDI.
checking whether to build HTML rig feature matrix... check
checking gd.h usability... no
checking gd.h presence... no
. . .
config.status: creating android/Makefile
config.status: creating amplifiers/elecraft/Makefile
config.status: creating hamlib.pc
config.status: creating include/config.h
config.status: executing depfiles commands
config.status: executing libtool commands
----------------------------------------------------------------------

 Hamlib Version 4.0 configuration:

 Prefix 	/usr/local
 Preprocessor	cc -E
 C Compiler	cc -g -O2
 C++ Compiler	c++ -g -O2

 Package features:
    With C++ binding		    yes
    With Perl binding		    no
    With Python binding 	    no
    With TCL binding		    no
    With Lua binding		    no
    With rigmem XML support	    no
    With Readline support	    yes
    With INDI support		    no

    Enable HTML rig feature matrix  no
    Enable WinRadio		    yes
    Enable USRP 		    no
    Enable USB backends 	    yes
    Enable shared libs		    yes
    Enable static libs		    yes

-----------------------------------------------------------------------

% make
Making all in macros
cd .. && make  am--refresh
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/sh '/home/larry/projects/Hamlib/build-aux/missing' aclocal-1.16 -I macros --install
 cd . && /bin/sh /home/larry/projects/Hamlib/build-aux/missing automake-1.16 --gnu
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/sh '/home/larry/projects/Hamlib/build-aux/missing' autoconf
/bin/sh ./config.status --recheck
running CONFIG_SHELL=/bin/sh /bin/sh ./configure --no-create --no-recursion
checking for gcc... no
checking for cc... cc
checking whether the C compiler works... yes
. . .
checking whether c++ supports C++11 features with -std=c++11... yes
Build_OS is freebsd13.0
checking that generated files are newer than configure... done
configure: creating ./config.status
----------------------------------------------------------------------

 Hamlib Version 4.4~git configuration:

 Prefix 	/usr/local
 Preprocessor	cc -E
 C Compiler	cc -g -O2
 C++ Compiler	c++ -std=c++11 -g -O2

 Package features:
    With C++ binding		    yes
    With Perl binding		    no
    With Python binding 	    no
    With TCL binding		    no
    With Lua binding		    no
    With rigmem XML support	    no
    With Readline support	    yes
    With INDI support		    no

    Enable HTML rig feature matrix  no
    Enable WinRadio		    yes
    Enable USRP 		    no
    Enable USB backends 	    no
    Enable shared libs		    yes
    Enable static libs		    yes

-----------------------------------------------------------------------
 /bin/sh ./config.status
config.status: creating Makefile
config.status: creating macros/Makefile
. . .
config.status: executing depfiles commands
config.status: executing libtool commands
Making all in include
(CDPATH="${ZSH_VERSION+.}:" && cd .. && /bin/sh '/home/larry/projects/Hamlib/build-aux/missing' autoheader)
rm -f stamp-h1
touch ./config.h.in
cd .. && /bin/sh ./config.status include/config.h
config.status: creating include/config.h
make  all-am
Making all in lib
  CC       termios.lo
  CC       dummy.lo
  CCLD     libmisc.la
Making all in rigs/adat
  CC       adt_200a.lo
  CC       adat.lo
  CCLD     libhamlib-adat.la
Making all in rigs/alinco
. . .
  CC       transfox.lo
  CC       flex.lo
In file included from flex.c:31:
In file included from ./kenwood.h:29:
../../src/misc.h:98:64: warning: declaration of 'struct timeval' will not be visible outside of this function [-Wvisibility]
extern HAMLIB_EXPORT(int) rig_check_cache_timeout(const struct timeval *tv,
                                                               ^
../../src/misc.h:101:59: warning: declaration of 'struct timeval' will not be visible outside of this function [-Wvisibility]
extern HAMLIB_EXPORT(void) rig_force_cache_timeout(struct timeval *tv);
                                                          ^
2 warnings generated.
  CCLD     libhamlib-kenwood.la
Making all in rigs/kit
  CC       elektor304.lo
  CC       drt1.lo
. . .
  CC       sleep.lo
  CC       sprintflst.lo
make[2]: don't know how to make ../rotators/satel/libhamlib-satel.la. Stop

make[2]: stopped in /home/larry/projects/Hamlib/src
*** Error code 2

Stop.
make[1]: stopped in /home/larry/projects/Hamlib/src
*** Error code 1

Stop.
make: stopped in /home/larry/projects/Hamlib
```

I'm having a hard time finding any information as to whether anyone has built a version for FreeBSD beyond 3.3.

techn0mad, 2021-11-20:
Taking another run at it:

```
% uname -a
FreeBSD pinawa 13.0-RELEASE-p4 FreeBSD 13.0-RELEASE-p4 #0: Tue Aug 24 07:33:27 UTC 2021     root@amd64-builder.daemonology.net:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

% git show --oneline -s
12f7bb4d (HEAD -> master, origin/master, origin/HEAD) Change getnameinfo to return numeric IP instead of hostname to avoid Windows long delay on reverse lookups https://github.com/Hamlib/Hamlib/issues/873

% gmake --version
GNU Make 4.3
Built for amd64-portbld-freebsd13.0
Copyright (C) 1988-2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

% autoconf --version
autoconf (GNU Autoconf) 2.69
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+/Autoconf: GNU GPL version 3 or later
<http://gnu.org/licenses/gpl.html>, <http://gnu.org/licenses/exceptions.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by David J. MacKenzie and Akim Demaille.

% automake --version
automake (GNU automake) 1.16.3
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv2+: GNU GPL version 2 or later <https://gnu.org/licenses/gpl-2.0.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Tom Tromey <tromey@redhat.com>
       and Alexandre Duret-Lutz <adl@gnu.org>.

% libtool --version
libtool (GNU libtool) 2.4.6
Written by Gordon Matzigkeit, 1996

Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

% ./bootstrap
Running 'autoreconf -i' to process configure.ac
and generate the configure script.
aclocal: installing 'macros/libtool.m4' from '/usr/local/share/aclocal/libtool.m4'
aclocal: installing 'macros/ltoptions.m4' from '/usr/local/share/aclocal/ltoptions.m4'
aclocal: installing 'macros/ltsugar.m4' from '/usr/local/share/aclocal/ltsugar.m4'
aclocal: installing 'macros/ltversion.m4' from '/usr/local/share/aclocal/ltversion.m4'
aclocal: installing 'macros/lt~obsolete.m4' from '/usr/local/share/aclocal/lt~obsolete.m4'
aclocal: installing 'macros/pkg.m4' from '/usr/local/share/aclocal/pkg.m4'
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize: copying file 'build-aux/ltmain.sh'
configure.ac:95: installing 'build-aux/ar-lib'
configure.ac:26: installing 'build-aux/compile'
configure.ac:126: installing 'build-aux/config.guess'
configure.ac:126: installing 'build-aux/config.sub'
configure.ac:35: installing 'build-aux/install-sh'
configure.ac:35: installing 'build-aux/missing'
amplifiers/elecraft/Makefile.am: installing 'build-aux/depcomp'
bindings/Makefile.am:106: installing 'build-aux/py-compile'
parallel-tests: installing 'build-aux/test-driver'

% ./configure
checking for gcc... no
checking for cc... cc
checking whether the C compiler works... yes
[... 1001 lines cut ...]
Making all in rotators/prosistel
  CC       prosistel.lo
  CCLD     libhamlib-prosistel.la
Making all in rotators/ioptron
  CC       rot_ioptron.lo
  CCLD     libhamlib-ioptron.la
Making all in rotators/satel
  CC       satel.lo
  CCLD     libhamlib-satel.la
Making all in rotators/radant
  CC       radant.lo
  CCLD     libhamlib-radant.la
Making all in amplifiers/elecraft
  CC       kpa1500.lo
  CC       kpa.lo
  CCLD     libhamlib-elecraft.la
Making all in src
make[1]: don't know how to make ./hamlibdatetime.h. Stop

make[1]: stopped in /home/larry/projects/Hamlib/src
*** Error code 1

Stop.
make: stopped in /home/larry/projects/Hamlib
```

mdblack98, 2021-11-20:
Use the tar file from here.  It doesn't need the bootstrap...just the configure...and should already have the hamlibdatetime.h
http://n0nb.users.sourceforge.net/

Mike W9MDB 

 
 

FreeBSD pinawa 13.0-RELEASE-p4 FreeBSD 13.0-RELEASE-p4 #0: Tue Aug 24 07:33:27 UTC 2021     ***@***.***:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

<http://gnu.org/licenses/gpl.html>, <http://gnu.org/licenses/exceptions.html>

mdblack98, 2021-11-20:
The tar file from http://n0nb.users.sourceforge.net/ builds perfectly for me on FreeBSD 13.
Also...I found gmake works on the git checkout version.
pkg install gmake
Seems bsd make can't handle generating the hamlibdatetime.h generation.

closed 2021-11-20 by mdblack98

techn0mad, 2021-11-21:
Hi Mike:

I tried the hamlib-4.4~git-cf527c8d-20211121.tar.gz tarfile and still got
this result:

copying selected object files to avoid basename conflicts...
Making all in c++
  CXX      rigclass.lo
  CXX      rotclass.lo
  CXX      ampclass.lo
  CXXLD    libhamlib++.la
c++: warning: argument unused during compilation: '-pthread'
[-Wunused-command-line-argument]
Making all in tests
  CC       rigctl-rigctl.o
rigctl.c:598:9: error: use of undeclared identifier 'rl_readline_name'
        rl_readline_name = "rigctl";
        ^
rigctl.c:601:9: warning: implicit declaration of function 'using_history'
is invalid in C99 [-Wimplicit-function-declaration]
        using_history();    /* Initialize Readline History */
        ^
1 warning and 1 error generated.
*** Error code 1

Stop.
make[1]: stopped in /home/larry/projects/hamlib-4.4~git/tests
*** Error code 1

Stop.
make: stopped in /home/larry/projects/hamlib-4.4~git

73,

On Sat, 20 Nov 2021 at 12:30, Michael Black ***@***.***>
wrote:

> Use the tar file from here.  It doesn't need the bootstrap...just the
> configure...and should already have the hamlibdatetime.h
> http://n0nb.users.sourceforge.net/
>
>
>
> Mike W9MDB
>

-- 
Larry Gadallah, lgadallah AT gmail DOT com
PGP Sig: AE93 1785 6874 7111 48AD  63A6 2136 3651 981C F87B

mdblack98, 2021-11-21:
I did a fresh install of FreeBSD 13 and did not have these problems.So I don't know what your problem is and perhaps a fresh install is needed.
Mike W9MDB

techn0mad, 2021-11-21:
Hi Mike:

Starting to look like one of those "works for me" scenarios: Ran a
completely fresh install using Vagrant and ran into the same problem.

It's also odd that a declaration of rl_readline_name doesn't seem to be
present anywhere.

calgary:vagrant larry$ vagrant --version
Vagrant 2.2.19

calgary:vagrant larry$ vagrant init freebsd/FreeBSD-13.0-RELEASE
A `Vagrantfile` has been placed in this directory. You are now
ready to `vagrant up` your first virtual environment! Please read
the comments in the Vagrantfile as well as documentation on
`vagrantup.com` for more information on using Vagrant.

calgary:vagrant larry$ vagrant up
Bringing machine 'default' up with 'virtualbox' provider...
==> default: Checking if box 'freebsd/FreeBSD-13.0-RELEASE' version
'2021.04.09' is up to date...
==> default: Clearing any previously set forwarded ports...
==> default: Clearing any previously set network interfaces...
==> default: Preparing network interfaces based on configuration...
    default: Adapter 1: nat
==> default: Forwarding ports...
    default: 22 (guest) => 2222 (host) (adapter 1)
==> default: Booting VM...
==> default: Waiting for machine to boot. This may take a few minutes...
    default: SSH address: 127.0.0.1:2222
    default: SSH username: vagrant
    default: SSH auth method: private key
==> default: Machine booted and ready!
==> default: Checking for guest additions in VM...
==> default: Mounting shared folders...
    default: /vagrant => /Users/larry/vagrant
==> default: Machine already provisioned. Run `vagrant provision` or use
the `--provision`
==> default: flag to force provisioning. Provisioners marked to run always
will still run.

calgary:vagrant larry$ vagrant ssh
Last login: Sun Nov 21 21:45:21 2021 from 10.0.2.2
FreeBSD 13.0-RELEASE-p4 (GENERIC) #0: Tue Aug 24 07:33:27 UTC 2021

Welcome to FreeBSD!

Release Notes, Errata: https://www.FreeBSD.org/releases/
Security Advisories:   https://www.FreeBSD.org/security/
FreeBSD Handbook:      https://www.FreeBSD.org/handbook/
FreeBSD FAQ:           https://www.FreeBSD.org/faq/
Questions List:
https://lists.FreeBSD.org/mailman/listinfo/freebsd-questions/
FreeBSD Forums:        https://forums.FreeBSD.org/

Documents installed with the system are in the /usr/local/share/doc/freebsd/
directory, or can be installed later with:  pkg install en-freebsd-doc
For other languages, replace "en" with a language code like de or fr.

Show the version of FreeBSD installed:  freebsd-version ; uname -a
Please include that output and any error messages when posting questions.
Introduction to manual pages:  man man
FreeBSD directory layout:      man hier

To change this login announcement, see motd(5).
In order to search for a string in some files, use 'grep' like this:

grep "string" filename1 [filename2 filename3 ...]

This will print out the lines in the files that contain the string.  grep
[... 115 lines cut ...]

Stop.
make[1]: stopped in /home/vagrant/hamlib-4.4~git/tests
*** Error code 1

Stop.
make: stopped in /home/vagrant/hamlib-4.4~git

***@***.***:~/hamlib-4.4~git % find . -depth -type f -exec grep -H
'rl_readline_name' {} \;
./tests/rotctl.c:        rl_readline_name = "rotctl";
./tests/ampctl.c:        rl_readline_name = "ampctl";
./tests/rigctl.c:        rl_readline_name = "rigctl";

On Sun, 21 Nov 2021 at 09:44, Michael Black ***@***.***>
wrote:

> I did a fresh install of FreeBSD 13 and did not have these problems.So I
> don't know what your problem is and perhaps a fresh install is needed.
> Mike W9MDB
>

-- 
Larry Gadallah, lgadallah AT gmail DOT com
PGP Sig: AE93 1785 6874 7111 48AD  63A6 2136 3651 981C F87B

techn0mad, 2021-11-22:
FWIW, I went through exactly the same sequence of steps, but for an Ubuntu 20.04 Vagrant box, and while the build completes sucessfully, I can't explain why since there still does not seem to be a declaration for 'rl_readline_name'. Based on the Gnu documentation for Readline, they indicate that this needs to be set to a unique string.

Comparing the Ubuntu and FreeBSD builds, I notice that configure enables Readline, whereas for the Ubuntu build Readline is disabled. By running "./configure --without-readline" prior to the FreeBSD build, the build ran to completion successfully.

mdblack98, 2021-11-22:
readline is working on the FreeBSD install I have.

Mike W9MDB