Hamlib/Hamlib #872: Kenwood mode set problems

mdblack98, 2021-11-18:
TS590S is setting mode when mode does not need to be set causing lots of relay chatter when transmitting with WSJTX.

barjac, 2021-11-18:
Oh what a mess!
Hi Mike,
Split was working fine for me with the TS-450S at commit 52e0840 a few weeks ago after your fixes. Still is, I checked.
However I have just tested at c229c2f5 having noticed that you have been poking at Kenwood code again.
Now split mode on the rig turns off for one TX period, then turns on for the next etc. When it turns on it switches on and off before finally staying on for the TX (rig beeping merrily!). Then it turns off and the next TX is on the same VFO (B) as RX.

Here is a log from start of rigctld, followed by start of wsjtx, then two cycles of FT4 CQ.
[rigctld_c229c2f54_2.log](https://github.com/Hamlib/Hamlib/files/7565714/rigctld_c229c2f54_2.log)

Cheers,
Barry

mdblack98, 2021-11-18:
Would you please send me a log from 52e0840 too?
Mike W9MDB

 
 

However I have just tested at c229c2f having noticed that you have been poking at Kenwood code again.

barjac, 2021-11-19:
Hi Mike,
I think I did at the end of the relevant bug report to show you what it was doing after the fixes.

mdblack98, 2021-11-19:
If you are building the static version of wsjtx then yes...you have to rebuild every time.
If you are building the shared library version then you just need to install hamlib -- and make sure you remove the system hamlib package.
You can do "ldd wsjtx" to see if libhamlib is referenced and the location found.

barjac, 2021-11-19:
No I am not building static at all. I package wsjtx and hamlib for Mageia, static builds are not allowed.
I am not building directly on my system, but building packages in build chroot for testing so there is nothing left on the test system from previous packages after updating from one to another.
I just realized that I do have my working system with  52e840 build installed so I am attaching a log.
Band switched several times, then near the end tuned a few times and the started a TX cycle, stopped it and killed rigctld
[rigctl_52e0840_1.log](https://github.com/Hamlib/Hamlib/files/7571221/rigctl_52e0840_1.log)
.

mdblack98, 2021-11-19:
Try the master repo again please.
Mike W9MDB

barjac, 2021-11-20:
The alternating 'split' then 'not split' at each TX  has now gone, but each transmit is preceded by a switch to 'not split' at the rig using VFOA. This causes delay and  4 beeps whereas with 52e840 there was no change from 'split mode' at the rig once it was set and generally only one beep going into TX iirc.
[rigctl_4b3bbab_4.log](https://github.com/Hamlib/Hamlib/files/7574828/rigctl_4b3bbab_4.log)

barjac, 2021-11-20:
Maybe these two small logs of simply running wsjtx and performing two TUNE operations may be easier to debug. Using kdiff3 the much extended portions of the log when using 4b3bbab can be seen.
[rigctl_52e0840_5.log](https://github.com/Hamlib/Hamlib/files/7574886/rigctl_52e0840_5.log)
[rigctl_4b3bbab_5.log](https://github.com/Hamlib/Hamlib/files/7574887/rigctl_4b3bbab_5.log)

barjac, 2021-11-21:
Still seeing switch to 'non-split' before TX.
[rigctld_cf527c8_10.log](https://github.com/Hamlib/Hamlib/files/7576160/rigctld_cf527c8_10.log)

mdblack98, 2021-11-21:
For rigs that don't have a way to directly set VFOB mode you will see split turn off, VFOB activated, VFOB mode set, VFOA activated, split turned on again.
I'm think about not querying VFOB mode and using the cached value to determine if it should change.  That would mean if the user makes a mode change on VFOB they are on their own and hamlib won't change it back until a restart.  But it would avoid all the display flashing that's occurring now on different rigs.
Mike W9MDB

barjac, 2021-11-21:
> For rigs that don't have a way to directly set VFOB mode you will see split turn off, VFOB activated, VFOB mode set, VFOA activated, split turned on again. I'm think about not querying VFOB mode and [...]

That sounds much more sensible to me. If the user is relying on hamlib for rig control he should not be 'knob twiddling'! If he does it's his problem!

I have been doing a bit of digging and it seems that you introduced this to fix a Yaesu problem in 603223d - the commit right after the one that was/is working fine for me. I was trying to understand why it was continually turning split off and back on now where previously it was set once and left alone.

Since the rig type is identified by hamlib could it not be conditionally handled to not do the 'split dance' for Kenwood? I don't know what this would involve but I'm sure software 'juggling' would be a lot faster than the 'split dance' with beeps and rattling relays :)

 Mike W9MDB

barjac, 2021-11-23:
I guess this is a work in progress but here is a quick test at d94c8c9 where each transmit is still preceded by a switch to non-split. (Rig sends 'i e' in morse).
[rigctld_d94c8c9_23.log](https://github.com/Hamlib/Hamlib/files/7589942/rigctld_d94c8c9_23.log)

mdblack98, 2021-11-23:
Try adding --vfo option.  Things behave better that way anyways -- a lot less traffic and vfo swapping going on.
Not sure I can fix the non-vfo version to avoid that without messing up other paths.  One possibility is to make the VFO selection just an internal thing in hamlib rather than doing anything with the rig.   That might work.
In non-vfo mode the client explicitly sets VFOA/B and that turns off split and we have to turn it back on again.  
Mike W9MDB

barjac, 2021-11-23:
Damn! I thought I was, as I read somewhere that -v was the short version. Hence my -v -vvvvv. So yes I will change that and re-test shortly.

mdblack98, 2021-11-23:
-o is the short version.See rigctld -h
Mike W9MDB

barjac, 2021-11-23:
It does not seem to make any difference.
Just to be sure I used the long version ;)
[rigctld_d94c8c9_24.log](https://github.com/Hamlib/Hamlib/files/7590240/rigctld_d94c8c9_24.log)
.

barjac, 2021-11-23:
I just tested the good version 52e0840 from Oct 1st again with -o and it is worse than without it. 

With:
[rigctld_52e0840_30.log](https://github.com/Hamlib/Hamlib/files/7591035/rigctld_52e0840_30.log)

Without
[rigctld_52e0840_31.log](https://github.com/Hamlib/Hamlib/files/7591036/rigctld_52e0840_31.log)
:

barjac, 2021-11-24:
Hi MIke,
Looks good :)
With --vfo just doing a short test at e58db5c, the split switching dance is gone. I will do more testing later and include a short log. The one from my 'short test' is too big to attach, and we are just about to eat.

barjac, 2021-11-24:
Tested as follows:
Start rigctld  (e58db5c) with rig powered up.
Start wsjtx
Change band twice. After first band change which produced a good few beeps, subsequent ones gave M or TT.
Transmitted two CQ cycles of FT4 which now does not beep at all nor change split mode at rig.
Well done!
[rigctld_e58db5c_35.log](https://github.com/Hamlib/Hamlib/files/7598356/rigctld_e58db5c_35.log)

Out of interest I will try this again without --vfo shortly.

barjac, 2021-11-24:
Without --vfo the result is the same as above, similar test.
[rigctld_e58db5c_36.log](https://github.com/Hamlib/Hamlib/files/7598384/rigctld_e58db5c_36.log)

mdblack98, 2021-11-24:
I assume that means all is OK?
Mike W9MDB

barjac, 2021-11-25:
> I assume that means all is OK?
Yes it seems fine now. I also tested without rigctld, using rs232 settings in wsjtx and klog and that also performed OK.
Thanks for all your work on this - much appreciated. :)

barjac, 2021-12-04:
I just built and installed 4.4 final and have a serious issue that has been introduced since e58db5c.
I can't yet find a way to reproduce it reliably enough to grab a log, but the effect is that the rig (TS450S) sits there switching from split to reverse split on an exact 2 second cycle (toggles every second) while not transmitting.
It has happened twice now when I cancelled a TX cycle in FT4. I had klog running in read only mode polling hamlib at a 500ms rate at the time, and stopping klog did stop the split flipping.
I will try to get some logs, but I thought you should know in advance.

barjac, 2021-12-04:
This log demonstrates the problem I hope!
Probably 10-15seconds before the end of the log.
I had to keep killing TX during the many TX cycles to trigger it.
Seems to only happen if klog is started after wsjtx but by no means certain.
[rigctld_4.4_5.log](https://github.com/Hamlib/Hamlib/files/7654464/rigctld_4.4_5.log)

mdblack98, 2021-12-04:
It looks like KLog is sending "V VFOB" which will reverse split on the rig.
grep "V " rigctld_4.4_5.log | grep VFOB
2021-12-04:18:50:22.692477: rigctl(d): V 'currVFO' 'VFOB' '' ''
2021-12-04:18:50:22.792175: rigctl(d): V 'currVFO' 'VFOB' '' ''
2021-12-04:18:50:26.021190: rigctl(d): V 'currVFO' 'VFOB' '' ''
2021-12-04:18:50:52.674620: rigctl(d): V 'currVFO' 'VFOB' '' ''
rigctl(d): V 'currVFO' 'VFOB' '' ''
2021-12-04:18:51:07.681342: rigctl(d): V 'currVFO' 'VFOB' '' ''
rigctl(d): V 'currVFO' 'VFOB' '' ''
2021-12-04:18:51:12.370091: 2021-12-04:18:51:12.370136: rigctl(d): V 'currVFO' 'VFOB' '' ''

Confirm that you do NOT see this when running just WSJTX.
I suspect it's only when running klog.
KLog should not be asking for VFOB mode on rigs that do not have TARGETABLE_MODE.

barjac, 2021-12-04:
Hi Mike,
Yes I did enough testing earlier to confirm that this is not happening without klog running, however how would klog know anything about the rig type if it is using Net_rigctl?

mdblack98, 2021-12-04:
Asking for VFOB freq  and mode should be options in KLog.  Some rigs can do VFOB freq and some can do both freq+mode.I do believe KLog is using the direct rigctld interface which require the client to control these things.
Mike W9MDB

closed 2021-12-04 by mdblack98

barjac, 2021-12-04:
Sorry Mike, I maybe misunderstood your question. I don't see the toggling of split/reverse-split when klog is not running, but I do see:
[baz@localhost ~]$ grep "V " rigctld_4.4_8.log | grep VFOB
2021-12-04:23:05:08.097263: rigctl(d): V 'currVFO' 'VFOB' '' ''
2021-12-04:23:05:08.202186: rigctl(d): V 'currVFO' 'VFOB' '' ''
2021-12-04:23:05:11.321813: rigctl(d): V 'currVFO' 'VFOB' '' ''
2021-12-04:23:05:22.637151: rigctl(d): V 'currVFO' 'VFOB' '' ''
2021-12-04:23:05:37.636219: rigctl(d): V 'currVFO' 'VFOB' '' ''
2021-12-04:23:05:40.837182: rigctl(d): V 'currVFO' 'VFOB' '' ''
2021-12-04:23:05:52.637144: rigctl(d): V 'currVFO' 'VFOB' '' ''
etc.
[rigctld_4.4_8.log](https://github.com/Hamlib/Hamlib/files/7654889/rigctld_4.4_8.log)
.

mdblack98, 2021-12-04:
Then the question is what is KLog doing extra?

mdblack98, 2021-12-05:
Also -- try using --vfo mode
I think KLog supports that and it may fix the problem.
Mike W9MDB

barjac, 2021-12-05:
I am using --vfo for all these tests.
rigctld -o -vvvvv -Z -m 2003 -r /dev/ttyUSB0 -T 127.0.0.1 -t 4532 --set-conf=serial_handshake=None