Hamlib/Hamlib #879: Yaesu newcat if shift

Johnny-Sierra, 2021-12-02:
Enhancement:
Add if shift capability in Yaesu newcat backend (command IS).

mdblack98, 2021-12-02:
We don't have an API for this....do you have a proposal?

Johnny-Sierra, 2021-12-03:
The if-shift is a common feature, so I suggest to create a new RIG_LEVEL constant and use the existing rig_set_level rig_get_level functions.

For Yaesu (newcat) the command prototype is:

Set IS P1 +/- P2 P2 P2 P2 ;
Read IS P1 ;
Answer IS P1 +/- P2 P2 P2 P2 ;

where P1 = 0,
P2 = -1200 to +1200 (20 Hz steps)

mdblack98, 2021-12-03:
LEVEL_IF is already implemented.
I just updated the rig.h to make it a little clearer
Mike W9MDB

Johnny-Sierra, 2021-12-03:
Very well, I missed it. HI
And also already implemented in the Yaesu newcat API.

Yes it isn't clear, from wiki:
<html><body>
<h2 class="memtitle">RIG_LEVEL_IF</h2>

<div class="memitem">
<div class="memproto">
      

#define RIG_LEVEL_IF   CONSTANT_64BIT_FLAG(6)
--

</div><div class="memdoc">
<p><code>IF</code> – IF, arg int (Hz) </p>

</div>
</div>
</body>
</html>

closed 2021-12-03 by mdblack98