Hamlib/Hamlib #882: Nightly MinGW build failure in function `rigctl_set_clock`

N0NB, 2021-12-07:
Hi Mike.

It looks like MinGW doesn't like the changes in the set_clock function:

```
Making install in tests
make[1]: Entering directory '/home/nate/builds/hamlib-4.5~git/tests'
  CC       rigctl-rigctl.o
  CC       rigctl-dumpcaps.o
  CC       rigctl-rigctl_parse.o
  CC       rigctld-rigctld.o
  CC       rigctld-rigctl_parse.o
  CC       rigctld-dumpcaps.o
  CC       rigmem-rigmem.o
  CC       rigmem-memsave.o
  CC       rigmem-memload.o
  CC       rigmem-memcsv.o
  CC       rigsmtr.o
  CC       rigswr.o
  CC       rotctl-rotctl.o
  CC       rotctl-rotctl_parse.o
  CC       rotctl-dumpcaps_rot.o
  CC       rotctld-rotctld.o
  CC       rotctld-rotctl_parse.o
  CC       rotctld-dumpcaps_rot.o
  CC       rigctlcom-rigctlcom.o
  CC       rigctlcom-rigctl_parse.o
  CC       rigctlcom-dumpcaps.o
  CC       ampctl-ampctl.o
  CC       ampctl-ampctl_parse.o
  CC       ampctl-dumpcaps_amp.o
  CC       ampctld-ampctld.o
  CC       ampctld-ampctl_parse.o
  CC       ampctld-dumpcaps_amp.o
  CC       rigtestlibusb-rigtestlibusb.o
  CCLD     rigctl.exe
  CCLD     rigctld.exe
  CCLD     rigmem.exe
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_clock':
/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:5238: undefined reference to `date_strget'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:5230: undefined reference to `date_strget'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:1018: rigctl.exe] Error 1
make[1]: *** Waiting for unfinished jobs....
/usr/bin/i686-w64-mingw32-ld: rigctld-rigctl_parse.o: in function `rigctl_set_clock':
/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:5238: undefined reference to `date_strget'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:5230: undefined reference to `date_strget'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:1026: rigctld.exe] Error 1
make[1]: Leaving directory '/home/nate/builds/hamlib-4.5~git/tests'
```

N0NB, 2021-12-07:
This is from commit c67f21bf.

mdblack98, 2021-12-07:
Sounds like the wrong library is getting linked.
date_strget did not move though the reference in rigctl-parse is new

Defined herenm misc.o | grep date_strget0000000000004897 T date_strget

And should be in libhamlib.anm libhamlib.a | grep date_strget                 U date_strget
0000000000004897 T date_strget
                 U date_strget

And libhamlib.sonm libhamlib.so | grep date_strget00000000000a9adf T date_strget

Mike W9MDB

mdblack98, 2021-12-07:
Darned text formatting....resending plain text.
 Sounds like the wrong library is getting linked.
date_strget did not move though the reference in rigctl-parse is new

Defined herenm misc.o | grep date_strget0000000000004897 T date_strget

And should be in libhamlib.anm libhamlib.a | grep date_strget                 U date_strget
0000000000004897 T date_strget
                 U date_strget

And libhamlib.sonm libhamlib.so | grep date_strget00000000000a9adf T date_strget

Mike W9MDB

mdblack98, 2021-12-07:
> Again, this is in the virtual machine where the directory is purged after each run and Hamlib has never been installed.
>
> I just ran the build manually and the failure looks the same:

Please check misc.o and libhamlib
nm misc.onm libhamlib.sonm libhamlib.a
All should show date_strget with the T label.
0000000000004897 T date_strget

Mike W9MDB 

  CC       rigctl-dumpcaps.o

N0NB, 2021-12-07:
I see this:

```
(1)[1]nate@buster:~/builds/hamlib-4.5~git/src/.libs 
$ nm misc.o libhamlib-4.dll | grep date_strget
00003cb0 T _date_strget
6772c120 T _date_strget
```

mdblack98, 2021-12-07:
Looks like it's having a problem with rigmem
Can you do a "make --trace" on just that program?

make --trace 
Makefile:1390: update target 'rigmem-rigmem.o' due to: rigmem.c
echo "  CC      " rigmem-rigmem.o;gcc -DHAVE_CONFIG_H -I. -I../include  -DIN_HAMLIB -I../include -I../src -I../lib  -Wall  -I../src -g -O0 -fPIC -D_FORTIFY_SOURCE=2  -MT rigmem-rigmem.o -MD -MP -MF .deps/rigmem-rigmem.Tpo -c -o rigmem-rigmem.o `test -f 'rigmem.c' || echo './'`rigmem.c
  CC       rigmem-rigmem.o
mv -f .deps/rigmem-rigmem.Tpo .deps/rigmem-rigmem.Po
Makefile:1040: update target 'rigmem' due to: rigmem-rigmem.o
rm -f rigmem
echo "  CCLD    " rigmem;/bin/bash ../libtool --silent --tag=CC   --mode=link gcc -Wall  -I../src -g -O0 -fPIC -D_FORTIFY_SOURCE=2    -o rigmem rigmem-rigmem.o rigmem-memsave.o rigmem-memload.o rigmem-memcsv.o  ../src/libhamlib.la ../lib/libmisc.la -ldl 
  CCLD     rigmem

 
 

I see this:

N0NB, 2021-12-07:
I get:

```Making all in tests
make[1]: Entering directory '/home/nate/builds/hamlib-4.5~git/tests'
Makefile:1017: update target 'rigctl.exe' due to: rigctl-rigctl.o rigctl-rigctl_parse.o rigctl-dumpcaps.o ../src/libhamlib.la ../lib/libmisc.la
rm -f rigctl.exe
echo "  CCLD    " rigctl.exe;/bin/bash ../libtool --silent --tag=CC   --mode=link i686-w64-mingw32-gcc -Wall -g -O2  -Wl,--enable-auto-import -L/home/nate/builds/libusb-1.0.24/MinGW32/dll -o rigctl.exe rigctl-rigctl.o rigctl-rigctl_parse.o rigctl-dumpcaps.o   ../src/libhamlib.la ../lib/libmisc.la  -lpthread 
  CCLD     rigctl.exe
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_clock':
/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:5238: undefined reference to `date_strget'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:5230: undefined reference to `date_strget'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:1018: rigctl.exe] Error 1
make[1]: Leaving directory '/home/nate/builds/hamlib-4.5~git/tests'
```

N0NB, 2021-12-07:
It looks like 759918a fixed this issue.  The dailies have been built and uploaded.

mdblack98, 2021-12-07:
It should be fixed with last patch I posed a little while ago.

 
 

I get:

closed 2021-12-07 by mdblack98