Hamlib/Hamlib #883: Add agc_level_e from value and string conversion

Johnny-Sierra, 2021-12-07:
In addition to internal API rig_stragclevel please add the following:

```
agc_level_e levelagcvalue (int agcValue)
{
    agc_level_e agcLevel;
    switch (agcValue)
    {
    case 0: agcLevel = RIG_AGC_OFF; break;
    case 1: agcLevel = RIG_AGC_SUPERFAST; break;
    case 2: agcLevel = RIG_AGC_FAST; break;
    case 3: agcLevel = RIG_AGC_SLOW; break;
    case 4: agcLevel = RIG_AGC_USER; break;
    case 5: agcLevel = RIG_AGC_MEDIUM; break;
    case 6: agcLevel = RIG_AGC_AUTO; break;
    default: agcLevel = RIG_AGC_AUTO; break;
    }
    return agcLevel;
}

agc_level_e levelagcstr (string agcString)
{
    agc_level_e agcLevel;
    if (agcString == "OFF") agcLevel = RIG_AGC_OFF;
    else if (agcString == "SUPERFAST") agcLevel = RIG_AGC_SUPERFAST;
    else if (agcString == "FAST") agcLevel = RIG_AGC_FAST;
    else if (agcString == "SLOW") agcLevel = RIG_AGC_SLOW;
    else if (agcString == "USER") agcLevel = RIG_AGC_USER;
    else if (agcString == "MEDIUM") agcLevel = RIG_AGC_MEDIUM;
    else agcLevel = RIG_AGC_AUTO;
    return agcLevel;
}
```

mdblack98, 2021-12-07:
Do you have a pull request you want to submit?
Are these going to be used somewhere?  I assume in rigctl?
Mike W9MDB

Johnny-Sierra, 2021-12-07:
Sorry, I'm not yet skilled with GitHub, so I don't know how to pull a request.

Please, if you agree and think that my proposal could be useful, you can add it into **misc.c** where are _rig_stragclevel_ and other utilities.
Sorry, some makeup are required, for example rename them with suffux _rig__ and convert string to maybe char *

TNX

mdblack98, 2021-12-07:
But where are they referred to?
What function is calling them?
Mike W9MDB

Johnny-Sierra, 2021-12-08:
No Mike, they are utilities for the hamlib users, but not referred from other function of the library.

mdblack98, 2021-12-08:
So you mean you want them exposed for the API and the shared library?

Johnny-Sierra, 2021-12-10:
I suppose that internal API are already exposed to external use and intended as utility http://hamlib.sourceforge.net/manuals/4.3/group__rig__internal.html
Infact I'm using some of them in my GUI, for example rig_strmode, rig_strmodes, rig_stragclevel, etc.
But if it is not the case, please discard my proposal.

closed 2021-12-10 by mdblack98

Johnny-Sierra, 2021-12-12:
Thanks.

I have another function, if you want to add it

```
value_t rig_valueagclevel (agc_level_e agcLevel)
{
    value_t value;

    if (agcLevel == RIG_AGC_OFF) value.i = 0;
    else if (agcLevel == RIG_AGC_SUPERFAST) value.i = 1;
    else if (agcLevel == RIG_AGC_FAST) value.i = 2;
    else if (agcLevel == RIG_AGC_SLOW) value.i = 3;
    else if (agcLevel == RIG_AGC_USER) value.i = 4;
    else if (agcLevel == RIG_AGC_MEDIUM) value.i = 5;
    else value.i = 6; //RIG_AGC_AUTO

    return value;
}

```