Hamlib/Hamlib #885: Icom USB echo not working when echo is off

mdblack98, 2021-12-12:
2021-12-12T10:28:23.386547-0500: icom_rig_open: IC-9700 v20211120.5
2021-12-12T10:28:23.386547-0500: icom_get_usb_echo_off called
2021-12-12T10:28:23.386547-0500: icom_get_usb_echo_off: retry temp set to 0
2021-12-12T10:28:23.386547-0500: frame.c(409):icom_transaction entered
2021-12-12T10:28:23.386547-0500: icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
2021-12-12T10:28:23.386547-0500: frame.c(139):icom_one_transaction entered
2021-12-12T10:28:23.386547-0500: frame.c(54):make_cmd_frame entered
2021-12-12T10:28:23.386547-0500: frame.c(88):make_cmd_frame return(6)
2021-12-12T10:28:23.386547-0500: rig_flush: called for serial device
2021-12-12T10:28:23.386547-0500: serial.c(645):serial_flush entered
2021-12-12T10:28:23.386547-0500: read_string called, rxmax=4095
2021-12-12T10:28:23.402172-0500: serial.c(713):serial_flush return(0)
2021-12-12T10:28:23.402172-0500: write_block(): TX 6 bytes
2021-12-12T10:28:23.402172-0500: 0000    fe fe a2 e0 03 fd                                  ......
2021-12-12T10:28:23.402172-0500: frame.c(459):read_icom_frame entered
2021-12-12T10:28:23.402172-0500: read_string called, rxmax=80
2021-12-12T10:28:23.418153-0500: read_string(): RX 11 characters
2021-12-12T10:28:23.418153-0500: 0000    fe fe e0 a2 03 00 40 17 44 01 fd                    ......@.D..
2021-12-12T10:28:23.418153-0500: frame.c(496):read_icom_frame return(11)
2021-12-12T10:28:23.418153-0500: frame.c(226):icom_one_transaction return(-8)
2021-12-12T10:28:23.418153-0500: icom_transaction: retry=0: Protocol error
frame.c(496):read_icom_frame return(11)
frame.c(226):icom_one_transaction return(-8)
icom_transaction: retry=0: Protocol error
frame.c(496):read_icom_frame return(11)
frame.c(226):icom_one_transaction return(-8)
frame.c(226):icom_one_transaction return(-8)
2021-12-12T10:28:23.528242-0500: icom_transaction: failed: Protocol error
frame.c(226):icom_one_transaction return(-8)
icom_transaction: retry=0: Protocol error
frame.c(496):read_icom_frame return(11)
frame.c(226):icom_one_transaction return(-8)
frame.c(226):icom_one_transaction return(-8)icom_transaction: failed: Protocol error
frame.c(226):icom_one_transaction return(-8)
icom_transaction: retry=0: Protocol error
frame.c(496):read_icom_frame return(11)
frame.c(226):icom_one_transaction return(-8)
frame.c(226):icom_one_transaction return(-8)icom_transaction: retry=0: Protocol error
frame.c(496):read_icom_frame return(11)
frame.c(226):icom_one_transaction return(-8)
frame.c(226):icom_one_transaction return(-8)
2021-12-12T10:28:23.528242-0500: frame.c(439):icom_transaction return(-8)
2021-12-12T10:28:23.528242-0500: icom_get_usb_echo_off: ack_len=0
2021-12-12T10:28:23.528242-0500: icom_get_usb_echo_off: USB echo off detected
2021-12-12T10:28:23.528242-0500: icom.c(768):icom_get_usb_echo_off return(1)
2021-12-12T10:28:23.528242-0500: icom_rig_open: rig error getting frequency retry=1, err=Invalid parameter
icom_get_usb_echo_off: USB echo off detected
icom.c(768):icom_get_usb_echo_off return(1)
icom_rig_open: rig error getting frequency retry=1, err=Invalid parameter
icom_get_usb_echo_off: USB echo off detected
icom.c(768):icom_get_usb_echo_off return(1)
icom.c(768):icom_get_usb_echo_off return(1)
2021-12-12T10:28:24.045357-0500: icom_get_usb_echo_off called
2021-12-12T10:28:24.045887-0500: icom_get_usb_echo_off: retry temp set to 0
2021-12-12T10:28:24.046896-0500: frame.c(409):icom_transaction entered
2021-12-12T10:28:24.047896-0500: icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
2021-12-12T10:28:24.048895-0500: frame.c(139):icom_one_transaction entered
2021-12-12T10:28:24.049894-0500: frame.c(54):make_cmd_frame entered
2021-12-12T10:28:24.049894-0500: frame.c(88):make_cmd_frame return(6)
2021-12-12T10:28:24.050895-0500: rig_flush: called for serial device
2021-12-12T10:28:24.050895-0500: serial.c(645):serial_flush entered
2021-12-12T10:28:24.050895-0500: read_string called, rxmax=4095
2021-12-12T10:28:24.061179-0500: serial.c(713):serial_flush return(0)
2021-12-12T10:28:24.063528-0500: write_block(): TX 6 bytes
2021-12-12T10:28:24.063528-0500: 0000    fe fe a2 e0 03 fd                                  ......
2021-12-12T10:28:24.064537-0500: frame.c(459):read_icom_frame entered
2021-12-12T10:28:24.064537-0500: read_string called, rxmax=80
2021-12-12T10:28:24.076845-0500: read_string(): RX 11 characters
2021-12-12T10:28:24.078539-0500: 0000    fe fe e0 a2 03 00 40 17 44 01 fd                    ......@.D..
2021-12-12T10:28:24.079537-0500: frame.c(496):read_icom_frame return(11)
2021-12-12T10:28:24.080537-0500: frame.c(226):icom_one_transaction return(-8)
2021-12-12T10:28:24.081948-0500: icom_transaction: retry=0: Protocol error
frame.c(496):read_icom_frame return(11)
frame.c(226):icom_one_transaction return(-8)
icom_transaction: retry=0: Protocol error
frame.c(496):read_icom_frame return(11)
frame.c(226):icom_one_transaction return(-8)
frame.c(226):icom_one_transaction return(-8)
2021-12-12T10:28:24.185549-0500: icom_transaction: failed: Protocol error
frame.c(226):icom_one_transaction return(-8)
icom_transaction: retry=0: Protocol error
frame.c(496):read_icom_frame return(11)
frame.c(226):icom_one_transaction return(-8)
frame.c(226):icom_one_transaction return(-8)icom_transaction: failed: Protocol error
frame.c(226):icom_one_transaction return(-8)
icom_transaction: retry=0: Protocol error
frame.c(496):read_icom_frame return(11)
frame.c(226):icom_one_transaction return(-8)
frame.c(226):icom_one_transaction return(-8)icom_transaction: retry=0: Protocol error
frame.c(496):read_icom_frame return(11)
frame.c(226):icom_one_transaction return(-8)
frame.c(226):icom_one_transaction return(-8)
2021-12-12T10:28:24.186256-0500: frame.c(439):icom_transaction return(-8)
2021-12-12T10:28:24.187285-0500: icom_get_usb_echo_off: ack_len=0
2021-12-12T10:28:24.188253-0500: icom_get_usb_echo_off: USB echo off detected
2021-12-12T10:28:24.189067-0500: icom.c(768):icom_get_usb_echo_off return(1)
2021-12-12T10:28:24.189067-0500: icom_rig_open: rig error getting frequency retry=0, err=Invalid parameter
icom_get_usb_echo_off: USB echo off detected
icom.c(768):icom_get_usb_echo_off return(1)
icom_rig_open: rig error getting frequency retry=0, err=Invalid parameter
icom_get_usb_echo_off: USB echo off detected
icom.c(768):icom_get_usb_echo_off return(1)
icom.c(768):icom_get_usb_echo_off return(1)
2021-12-12T10:28:24.189067-0500: icom.c(992):icom_rig_open return(1)
2021-12-12T10:28:24.190076-0500: rig.c(1028):rig_open return(1)
rig_open: error = Invalid parameter
icom_rig_open: rig error getting frequency retry=0, err=Invalid parameter
icom_get_usb_echo_off: USB echo off detected
icom.c(768):icom_get_usb_echo_off return(1)
icom.c(768):icom_get_usb_echo_off return(1)icom.c(992):icom_rig_open return(1)
rig.c(1028):rig_open return(1) com11 No error

mdblack98, 2021-12-13:
Not working
rigctl -m 3073 -r /dev/ttyUSB0 -vvvvv –Z  (no-autopower,txt, no USB echo in radio, startup rigctl fails)

021-12-12T16:31:06.793079-0800: rigctl.c(444) Startup: rigctl -m 3073 -r /dev/ttyUSB0 -vvvvv -Z

2021-12-12T16:31:06.812753-0800: icom_init called
2021-12-12T16:31:06.812835-0800: icom_init: done
2021-12-12T16:31:06.812913-0800: icom.c(684):icom_init return(0)
2021-12-12T16:31:06.812994-0800: rigctl_parse.c(1940):set_conf entered
2021-12-12T16:31:06.813073-0800: rigctl_parse.c(1974):set_conf return(0)
2021-12-12T16:31:06.813159-0800: rig.c(685):rig_open entered
2021-12-12T16:31:06.813276-0800: iofunc.c(68):port_open entered
2021-12-12T16:31:06.813411-0800: serial.c(144):serial_open entered
2021-12-12T16:31:06.813490-0800: serial_open: /dev/ttyUSB0
2021-12-12T16:31:06.817852-0800: serial.c(291):serial_setup entered
2021-12-12T16:31:06.817937-0800: serial_setup: tcgetattr
2021-12-12T16:31:06.818024-0800: serial_setup: cfmakeraw
2021-12-12T16:31:06.818109-0800: serial_setup: cfsetispeed=115200,0x1002
2021-12-12T16:31:06.818194-0800: serial_setup: cfsetospeed=115200,0x1002
2021-12-12T16:31:06.818276-0800: serial_setup: data_bits=8
2021-12-12T16:31:06.818355-0800: serial_setup: parity=0
2021-12-12T16:31:06.818444-0800: serial_setup: tcsetattr TCSANOW
2021-12-12T16:31:06.818576-0800: serial.c(631):serial_setup return(0)
2021-12-12T16:31:06.818663-0800: serial.c(645):serial_flush entered
2021-12-12T16:31:06.818772-0800: read_string called, rxmax=4095
2021-12-12T16:31:06.819972-0800: serial.c(713):serial_flush return(0)
2021-12-12T16:31:06.870189-0800: serial.c(266):serial_open return(0)
2021-12-12T16:31:06.870232-0800: iofunc.c(186):port_open return(0)
2021-12-12T16:31:06.870270-0800: rig.c(221):add_opened_rig return(0)
2021-12-12T16:31:06.870330-0800: icom.c(937):icom_rig_open entered
2021-12-12T16:31:06.870368-0800: icom_rig_open: IC-7300 v20211120.6
2021-12-12T16:31:06.870405-0800: icom_get_usb_echo_off called
2021-12-12T16:31:06.870440-0800: icom_get_usb_echo_off: retry temp set to 0
2021-12-12T16:31:06.870477-0800: frame.c(409):icom_transaction entered
2021-12-12T16:31:06.870512-0800: icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
2021-12-12T16:31:06.870550-0800: frame.c(139):icom_one_transaction entered
2021-12-12T16:31:06.870589-0800: frame.c(54):make_cmd_frame entered
2021-12-12T16:31:06.870624-0800: frame.c(88):make_cmd_frame return(6)
2021-12-12T16:31:06.870661-0800: rig_flush: called for serial device
2021-12-12T16:31:06.870695-0800: serial.c(645):serial_flush entered
2021-12-12T16:31:06.870730-0800: read_string called, rxmax=4095
2021-12-12T16:31:06.871843-0800: serial.c(713):serial_flush return(0)
2021-12-12T16:31:06.871906-0800: write_block(): TX 6 bytes
2021-12-12T16:31:06.871947-0800: 0000    fe fe 94 e0 03 fd                                   ......          
2021-12-12T16:31:06.871987-0800: frame.c(459):read_icom_frame entered
2021-12-12T16:31:06.872022-0800: read_string called, rxmax=80
2021-12-12T16:31:06.874080-0800: read_string(): RX 11 characters
2021-12-12T16:31:06.874120-0800: 0000    fe fe e0 94 03 80 96 11 18 00 fd                    ...........     
2021-12-12T16:31:06.874155-0800: frame.c(496):read_icom_frame return(11)
2021-12-12T16:31:06.874192-0800: icom_one_transaction: frm_len=11, frm_len-1=fd, frm_len-2=00
2021-12-12T16:31:06.874232-0800: frame.c(387):icom_one_transaction return(0)
2021-12-12T16:31:06.874266-0800: frame.c(439):icom_transaction return(0)
2021-12-12T16:31:06.874301-0800: icom_get_usb_echo_off: ack_len=6
2021-12-12T16:31:06.874335-0800: icom_get_usb_echo_off: USB echo off detected
2021-12-12T16:31:06.874370-0800: icom.c(771):icom_get_usb_echo_off return(1)
2021-12-12T16:31:06.874449-0800: icom_rig_open: rig error getting frequency retry=1, err=Invalid parameter
icom_get_usb_echo_off: USB echo off detected
icom.c(771):icom_get_usb_echo_off return(1)
icom_rig_open: rig error getting frequency retry=1, err=Invalid parameter
icom_get_usb_echo_off: USB echo off detected
icom.c(771):icom_get_usb_echo_off return(1)
icom.c(771):icom_get_usb_echo_off return(1)
2021-12-12T16:31:07.374598-0800: icom_get_usb_echo_off called
2021-12-12T16:31:07.374652-0800: icom_get_usb_echo_off: retry temp set to 0
2021-12-12T16:31:07.374693-0800: frame.c(409):icom_transaction entered
2021-12-12T16:31:07.374732-0800: icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
2021-12-12T16:31:07.374774-0800: frame.c(139):icom_one_transaction entered
2021-12-12T16:31:07.374813-0800: frame.c(54):make_cmd_frame entered
2021-12-12T16:31:07.374850-0800: frame.c(88):make_cmd_frame return(6)
2021-12-12T16:31:07.374890-0800: rig_flush: called for serial device
[... 7 lines cut ...]
2021-12-12T16:31:07.378335-0800: read_string(): RX 11 characters
2021-12-12T16:31:07.378378-0800: 0000    fe fe e0 94 03 80 96 11 18 00 fd                    ...........     
2021-12-12T16:31:07.378414-0800: frame.c(496):read_icom_frame return(11)
2021-12-12T16:31:07.378452-0800: icom_one_transaction: frm_len=11, frm_len-1=fd, frm_len-2=00
2021-12-12T16:31:07.378488-0800: frame.c(387):icom_one_transaction return(0)
2021-12-12T16:31:07.378525-0800: frame.c(439):icom_transaction return(0)
2021-12-12T16:31:07.378560-0800: icom_get_usb_echo_off: ack_len=6
2021-12-12T16:31:07.378597-0800: icom_get_usb_echo_off: USB echo off detected
2021-12-12T16:31:07.378634-0800: icom.c(771):icom_get_usb_echo_off return(1)
[... cut ...]
icom.c(771):icom_get_usb_echo_off return(1)
2021-12-12T16:31:07.378748-0800: icom.c(995):icom_rig_open return(1)
2021-12-12T16:31:07.378785-0800: rig.c(1028):rig_open return(1)
rig_open: error = Invalid parameter
icom_rig_open: rig error getting frequency retry=0, err=Invalid parameter
icom_get_usb_echo_off: USB echo off detected
icom.c(771):icom_get_usb_echo_off return(1)
icom.c(771):icom_get_usb_echo_off return(1)icom.c(995):icom_rig_open return(1)
rig.c(1028):rig_open return(1) /dev/ttyUSB0 Success 
root@rigpi3:/usr/local/bin#

closed 2021-12-13 by mdblack98

gridbasedotnet, 2023-11-11:
I cannot find the command in rigctl --help to turn off echo. How do I turn off echo?

 Additionally, should I be running-c, --civaddr=ID since I am using CI-V? And, how do i get the CI-V addr?

mdblack98, 2023-11-11:
Menu 36 CI-V TRN
Commonly called "echo" for Icom rigs.