Hamlib/Hamlib #889: Not able to use Sub without split mode ICOM.

cantudo, 2021-12-19:
https://github.com/Hamlib/Hamlib/blob/b78ec43e2e8c391f0eca5761f7d059c013b23173/rigs/icom/icom.c#L2673-L2684

Hello, I have an ICOM 9100. I wanted to change to Sub using Hamlib, and I noticed that you can only change to Sub (refer to code above) if you have Split mode enabled, otherwise it forces you to use VFOB, i get that you can just use VFOB if you are not doing split mode, but why force the user?

Thank you.

mdblack98, 2021-12-19:
Try using SUB_A and SUB_B for direct access.
Mike W9MDB

 
 

https://github.com/Hamlib/Hamlib/blob/b78ec43e2e8c391f0eca5761f7d059c013b23173/rigs/icom/icom.c#L2673-L2684

cantudo, 2021-12-19:
Thank you very much, I can change without split mode enabled now using "V MainA" and "V SubA" in rigctl. :+1:

closed 2021-12-19 by cantudo

cantudo, 2021-12-19:
@mdblack98 Should this be added to man page? It does not mention "MainA", "SubA", etc.
https://github.com/Hamlib/Hamlib/blob/b78ec43e2e8c391f0eca5761f7d059c013b23173/doc/man1/rigctld.1#L486-L487