Hamlib/Hamlib #909: Xiegu G90 improvements

mdblack98, 2021-12-29:
Add more capability to Xiegu G90 now that we have the CI-V info.
Question is how this applies to other Xeigu's.

rubienr, 2024-01-03:
Please consider the new display unit FW version 1.8 (2023-05-07) which adds CI-V commands
- `0x25`,
- `0x26` and
- `0x1d 0x19`.

CI-V Reference: https://xiegu.eu/wp-content/uploads/sites/4/2023/08/XIEGU-RADIO-CI-V-REFERENCE.pdf
FW release notes: https://xiegu.eu/downloads/ -> "G90 Firmware 1.8 (2023-05-07)"

mdblack98, 2024-01-03:
Current hamlib should be using 0x25 and 0x26 if it is supported...it tests during startup to see if it is.

rubienr, 2024-01-04:
I landed here, and this might be off topic, because i faced issues with `rigctld` + `wsjtx` although `flrig` + `wsjtx` works perfectly fine. I expected that both use the same rig protocol.
Because of annoying troubles with `flrig` i want to use `rigctld`. However, with `wsjtx` connected to `rigctld` it reports this error when band is changed: 

Wsjtx `2.6.1`:
```
Hamlib error: vfo_fixup: final vfo=VFOB
rig_set_split_vfo: final rxvfo=VFOA, txvfo=VFOB, split=0
../../src/rig.c(5012) trace
../../src/rig.c(5035) trace
netrigctl_set_split_vfo called vfo=VFOA, vfotx=VFOB, split=1
netrigctl_vfostr: called vfo=VFOA
netrigctl_vfostr: vfo_opt=0
netrigctl_transaction: called len=9
rig_flush: called for network device
network_flush called
write_block(): TX 9 bytes, method=2
0000    53 20 31 20 56 46 4f 42 0a                          S 1 VFOB.       
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): RX 9 characters, direct=1
0000    52 50 52 54 20 2d 31 31 0a                          RPRT -11.       
 1:rig_set_split_vfo: elapsed=0ms
 1:rig.c(5047):rig_set_split_vfo returning(-11) Feature not available

Feature not available
Feature not available
 while setting/unsetting split mode
```

Rigctld `Hamlib 4.5.5 Apr 05 11:43:08Z 2023 SHA=6eecd3`:
```
$ rigctld -l | grep Xiegu | grep -i G90
  3088  Xiegu                  G90                     20230109.4      Stable      RIG_MODEL_G90

$ rigctld --model 3088 --rig-file /dev/ttyUSB0 --serial-speed 19200 -C rts_state=OFF,dtr_state=OFF -vvv
rig_get_cache(302): unknown vfo=otherVFO
icom_set_vfo: unknown vfo 'currVFO'
icom_set_vfo: unsupported VFO otherVFO
Opened rig model 3088, 'G90'
rigctl_set_ptt: ptt=0  <-- this is when wsjtx is started and connects to rigctld
rigctl_set_ptt: ptt=0  <-- this is when band is changed via wsjtx
```

In both cases (`flrig` + `wsjtx`, `rigctld` + `wsjtx`) the "Split Operation" in Wsjtx is set to "Rig". The only setting that changes is "Rig" which is either "Hamlib NET rigctl" or "FLRig FLRig" accordingly.

Working Flrig `1.4.7` config:
[image]

Wsjtx Radio config:
[image]
`Test Cat` and `Test PTT` seem to work, but the rest of CAT communication is erroneous right up to where Wsjtx crashes.

mdblack98, 2024-01-04:
That should be fixed in the latest hamlib.
https://n0nb.users.sourceforge.net/

N0NB, 2025-07-30:
@rubienr Is this still an issue?

rubienr, 2025-09-09:
:/ 
I can't tell because the Xiegu G90 was replaced by Kenwood TS-890
I suggest to close the issue and re-open if it occurs again.

N0NB, 2025-09-09:
Thanks for the update.

closed 2025-09-09 by N0NB