Hamlib/Hamlib #914: "w64" exe file is actually 32-bit

nautix, 2022-01-02:
Upon downloading https://github.com/Hamlib/Hamlib/releases/download/4.4/hamlib-w64-4.4.exe and initiating the installation, I saw that it wanted to install into C:\Program Files(x86)\..., which made me suspicious. I aborted the install and checked the properties of hamlib-w64-4.4.exe using the steps outlined at https://www.digitalcitizen.life/3-ways-learn-whether-windows-program-64-bit-or-32-bit/, method 3. Specifically, when choosing the Compatibility Mode, Windows 95 was the first option, so I concluded that the subject file was 32-bit, though its name implied 64-bit.

Workaround: I downloaded https://github.com/Hamlib/Hamlib/releases/download/4.4/hamlib-w32-4.4.zip and unzipped it. At least hamlib-w64-4.4/bin/rigctl.exe is 64-bit, as tested by the same method as above. I have not tested all of the executables.

N0NB, 2022-01-02:
I'm a bit confused by your work around, are you saying the 64 bit files are in the archives named for w32?

If so, I've got something terribly wrong with my script!

nautix, 2022-01-02:
No.

The executables (well, rigctl.exe at least) in hamlib-w64-4.4.zip appear to
be 64-bit, but the installation executable, hamlib-w64-4.4.exe appears to
be 32-bit and, when I try to install from hamlib-w64-4.4.exe, it defaults
to installing under the 32-bit folder (C:\Program Files(x86)\).

So, if anyone REALLY wants the 64-bit executables, they can workaround the
issue by downloading the .zip file and manually installing the various
files. But rebuilding the w64 installation executable with proper 64-bit
executables would make it MUCH easier.

⟶ *Brian Haney*   ***@***.***)

*This email might contain paid links.*

N0NB, 2022-01-02:
Ahhh, that makes more sense!

I use the nsis package in Debian to create the executable archive.  I probably have missed an important option to create a 64 bit executable archive.

mdblack98, 2022-01-02:
Apparently your 64-bit build is not working -- you should be able to confirm this on your build system too.
$ file *.*ampctl.exe:          PE32 executable (console) Intel 80386 (stripped to external PDB), for MS Windowsampctld.exe:         PE32 executable (console) Intel 80386 (stripped to external PDB), for MS Windowslibgcc_s_sjlj-1.dll: PE32 executable (DLL) (console) Intel 80386, for MS Windowslibhamlib-4.dll:     PE32 executable (DLL) (console) Intel 80386 (stripped to external PDB), for MS Windowslibusb-1.0.dll:      PE32 executable (DLL) (console) Intel 80386, for MS Windowslibwinpthread-1.dll: PE32 executable (DLL) (console) Intel 80386, for MS Windowsrigctl.exe:          PE32 executable (console) Intel 80386 (stripped to external PDB), for MS Windowsrigctlcom.exe:       PE32 executable (console) Intel 80386 (stripped to external PDB), for MS Windowsrigctld.exe:         PE32 executable (console) Intel 80386 (stripped to external PDB), for MS Windowsrigmem.exe:          PE32 executable (console) Intel 80386 (stripped to external PDB), for MS Windowsrigsmtr.exe:         PE32 executable (console) Intel 80386 (stripped to external PDB), for MS Windowsrigswr.exe:          PE32 executable (console) Intel 80386 (stripped to external PDB), for MS Windowsrigtestlibusb.exe:   PE32 executable (console) Intel 80386 (stripped to external PDB), for MS Windowsrotctl.exe:          PE32 executable (console) Intel 80386 (stripped to external PDB), for MS Windowsrotctld.exe:         PE32 executable (console) Intel 80386 (stripped to external PDB), for MS Windows
On my build locally$ cd src/hamlib/src/mingw64-inst/bin
***@***.*** ~/src/hamlib/src/mingw64-inst/bin$ lsampctl.exe*   libhamlib-4.dll*  rigctlcom.exe*  rigmem.exe*   rigswr.exe*         rotctl.exe*ampctld.exe*  rigctl.exe*       rigctld.exe*    rigsmtr.exe*  rigtestlibusb.exe*  ***@***.*** ~/src/hamlib/src/mingw64-inst/bin$ file *ampctl.exe:        PE32+ executable (console) x86-64, for MS Windowsampctld.exe:       PE32+ executable (console) x86-64, for MS Windowslibhamlib-4.dll:   PE32+ executable (DLL) (console) x86-64, for MS Windowsrigctl.exe:        PE32+ executable (console) x86-64, for MS Windowsrigctlcom.exe:     PE32+ executable (console) x86-64, for MS Windowsrigctld.exe:       PE32+ executable (console) x86-64, for MS Windowsrigmem.exe:        PE32+ executable (console) x86-64, for MS Windowsrigsmtr.exe:       PE32+ executable (console) x86-64, for MS Windowsrigswr.exe:        PE32+ executable (console) x86-64, for MS Windowsrigtestlibusb.exe: PE32+ executable (console) x86-64, for MS Windowsrotctl.exe:        PE32+ executable (console) x86-64, for MS Windowsrotctld.exe:       PE32+ executable (console) x86-64, for MS Windows

N0NB, 2022-01-03:
Brian,

Please check http://n0nb.users.sourceforge.net/hamlib-w64-4.5~git-18548ef2-20220102.exe and be sure you're seeing it is 64 bit now for the installer EXE.  I ran down the undocumented setting to make this work, fortunately, Debian builds the package to enable 64 bit but doesn't document the command.

I have checked in a Win 10 64 system here through the File Manager properties and it shows to only be compatible with Vista and later.

nautix, 2022-01-03:
I downloaded
http://n0nb.users.sourceforge.net/hamlib-w64-4.5~git-18548ef2-20220102.exe
and checked it's Properties > Compatibility. It appears to be 64-bit.

I executed hamlib-w64-4.5~git-18548ef2-20220102.exe with Administrator
privileges (to get around the Windows Evil Software Guardians) and
installed the suite under C:\Program Files\hamlib-w64-4.5rc\. Then, under
...\bin\, I checked the properties of rigctl.exe and rigctd.exe, both of
which appear to be 64-bit executables. It looks like you fixed it.

Thank you for being so responsive! Only in the Amateur Radio / FOSS
community can I report an issue at lunchtime on a Sunday and see it
effectively fixed before I go to bed. Bravo!

⟶ *Brian Haney*   ***@***.***)

mdblack98, 2022-01-03:
Yup -- all files are now 64-bit except libgcc_s_sjlj-1.dll and I don't see where any of the executables or dlls depend on it

ampctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsampctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowslibgcc_s_sjlj-1.dll: PE32 executable (DLL) (console) Intel 80386, for MS Windowslibhamlib-4.dll:     PE32+ executable (DLL) (console) x86-64 (stripped to external PDB), for MS Windowslibusb-1.0.dll:      PE32+ executable (DLL) (console) x86-64, for MS Windowslibwinpthread-1.dll: PE32+ executable (DLL) (console) x86-64, for MS Windowsrigctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigctlcom.exe:       PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigmem.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigsmtr.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigswr.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigtestlibusb.exe:   PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrotctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrotctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows
Mike W9MDB
Message ID: ***@***.***>

N0NB, 2022-01-03:
* On 2022 02 Jan 22:50 -0600, Michael Black wrote:
> Yup -- all files are now 64-bit except libgcc_s_sjlj-1.dll and I don't see where any of the executables or dlls depend on it

At one time there was a dependency on that or a similar file.  As I did
upgrades through years I simply figured out the new version and have
included it as a matter of course.

> ampctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsampctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS [...]

That is tough to parse!

If you think I can simply not include the libgcc_s_sjlj-1.dll file and
everything will work, I'll do so.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

N0NB, 2022-01-03:
Mike, I see your output looks fine here.  One thing that helps pasted output like this is bracket it between lines of three grave accent characters.

Thanks for the feedback, Brian.

Now, I should probably rebuild the EXE installer for 4.4 but I will leave the older versions alone.

mdblack98, 2022-01-03:
Never heard of using the grave accent chars....let's see if this works.
~~~ampctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsampctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowslibgcc_s_sjlj-1.dll: PE32 executable (DLL) (console) Intel 80386, for MS Windowslibhamlib-4.dll:     PE32+ executable (DLL) (console) x86-64 (stripped to external PDB), for MS Windowslibusb-1.0.dll:      PE32+ executable (DLL) (console) x86-64, for MS Windowslibwinpthread-1.dll: PE32+ executable (DLL) (console) x86-64, for MS Windowsrigctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigctlcom.exe:       PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigmem.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigsmtr.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigswr.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigtestlibusb.exe:   PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrotctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrotctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows~~~

 

    On Monday, January 3, 2022, 07:25:47 AM CST, Nate Bargmann ***@***.***> wrote:  
 
 

* On 2022 02 Jan 22:50 -0600, Michael Black wrote:
> Yup -- all files are now 64-bit except libgcc_s_sjlj-1.dll and I don't see where any of the executables or dlls depend on it

At one time there was a dependency on that or a similar file. As I did
upgrades through years I simply figured out the new version and have
included it as a matter of course.

> ampctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsampctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowslibgcc_s_sjlj-1.dll: PE32 executable (DLL) (console) Intel 80386, for MS Windowslibhamlib-4.dll:     PE32+ executable (DLL) (console) x86-64 (stripped to external PDB), for MS Windowslibusb-1.0.dll:      PE32+ executable (DLL) (console) x86-64, for MS Windowslibwinpthread-1.dll: PE32+ executable (DLL) (console) x86-64, for MS Windowsrigctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigctlcom.exe:       PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigmem.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigsmtr.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigswr.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigtestlibusb.exe:   PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrotctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrotctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows

That is tough to parse!

If you think I can simply not include the libgcc_s_sjlj-1.dll file and
everything will work, I'll do so.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds. The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

—
Reply to this email directly, view it on GitHub, or unsubscribe.
Triage notifications on the go with GitHub Mobile for iOS or Android.
You are receiving this because you commented.Message ID: ***@***.***>

mdblack98, 2022-01-03:
Well that didn't work
Mike W9MDB

 

    On Monday, January 3, 2022, 07:35:28 AM CST, Michael Black ***@***.***> wrote:  
 
 

Never heard of using the grave accent chars....let's see if this works.
~~~ampctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsampctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowslibgcc_s_sjlj-1.dll: PE32 executable (DLL) (console) Intel 80386, for MS Windowslibhamlib-4.dll:     PE32+ executable (DLL) (console) x86-64 (stripped to external PDB), for MS Windowslibusb-1.0.dll:      PE32+ executable (DLL) (console) x86-64, for MS Windowslibwinpthread-1.dll: PE32+ executable (DLL) (console) x86-64, for MS Windowsrigctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigctlcom.exe:       PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigmem.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigsmtr.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigswr.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigtestlibusb.exe:   PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrotctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrotctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows~~~

 

 On Monday, January 3, 2022, 07:25:47 AM CST, Nate Bargmann ***@***.***> wrote: 
 
 

* On 2022 02 Jan 22:50 -0600, Michael Black wrote:
> Yup -- all files are now 64-bit except libgcc_s_sjlj-1.dll and I don't see where any of the executables or dlls depend on it

At one time there was a dependency on that or a similar file. As I did
upgrades through years I simply figured out the new version and have
included it as a matter of course.

> ampctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsampctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowslibgcc_s_sjlj-1.dll: PE32 executable (DLL) (console) Intel 80386, for MS Windowslibhamlib-4.dll:     PE32+ executable (DLL) (console) x86-64 (stripped to external PDB), for MS Windowslibusb-1.0.dll:      PE32+ executable (DLL) (console) x86-64, for MS Windowslibwinpthread-1.dll: PE32+ executable (DLL) (console) x86-64, for MS Windowsrigctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigctlcom.exe:       PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigmem.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigsmtr.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigswr.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigtestlibusb.exe:   PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrotctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrotctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows

That is tough to parse!

If you think I can simply not include the libgcc_s_sjlj-1.dll file and
everything will work, I'll do so.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds. The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

—
Reply to this email directly, view it on GitHub, or unsubscribe.
Triage notifications on the go with GitHub Mobile for iOS or Android.
You are receiving this because you commented.Message ID: ***@***.***>
 
—
Reply to this email directly, view it on GitHub, or unsubscribe.
Triage notifications on the go with GitHub Mobile for iOS or Android.
You are receiving this because you are subscribed to this thread.Message ID: ***@***.***>

N0NB, 2022-01-03:
The grave accents highlight text in a constant width font which is implemented by Markdown.

They can either be `inline` with a single grave accent on either side of the text or as a block with a line of three grave accents preceding and succeeding the block:

```
ampctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows
ampctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS  Windows
libgcc_s_sjlj-1.dll: PE32 executable (DLL) (console) Intel 80386, for MS Windows
libhamlib-4.dll:     PE32+ executable (DLL) (console) x86-64 (stripped to external PDB), for MS Windows
libusb-1.0.dll:      PE32+ executable (DLL) (console) x86-64, for MS Windows
libwinpthread-1.dll: PE32+ executable (DLL) (console) x86-64, for MS Windows
rigctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows
rigctlcom.exe:       PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows
rigctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows
rigmem.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows
rigsmtr.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows
rigswr.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows
rigtestlibusb.exe:   PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows
rotctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows
rotctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows
```

The block formatting works best for copying and pasting from a terminal window.

Play with formatting by using the Preview tab above the text entry box.

N0NB, 2022-01-03:
BTW, what tool did you use for that output, Mike?

N0NB, 2022-01-03:
> Thank you for being so responsive! Only in the Amateur Radio / FOSS community can I report an issue at lunchtime on a Sunday and see it effectively fixed before I go to bed. Bravo!

Truth be told, I had just finished setting a 2m repeater up that had gotten back from repair last week.  I had gone through and brought the duplexer into line and got it all up and running.  I then checked my email and saw the new issue.  Also, I am waiting for any final 2021 donations to our church before I devote the next week to ten days finalizing the 2021 financial report.  In other words, right place, right time!

mdblack98, 2022-01-03:
The "file" command under mingw
Now I pasted the mingw copy  into notepad++ and copy/pasted from notepad++ here...hopefully this works and get the CR in there.
ampctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsampctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowslibgcc_s_sjlj-1.dll: PE32 executable (DLL) (console) Intel 80386, for MS Windowslibhamlib-4.dll:     PE32+ executable (DLL) (console) x86-64 (stripped to external PDB), for MS Windowslibusb-1.0.dll:      PE32+ executable (DLL) (console) x86-64, for MS Windowslibwinpthread-1.dll: PE32+ executable (DLL) (console) x86-64, for MS Windowsrigctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigctlcom.exe:       PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigmem.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigsmtr.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigswr.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrigtestlibusb.exe:   PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrotctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windowsrotctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows

Mike W9MDB

N0NB, 2022-01-03:
I see libgcc_s_sjlj-1.dll isn't even an x86-64 format so it's rather useless in this archive.

mdblack98, 2022-01-03:
I changed the mingw window to use full html...interesting how to see how this appears as it looks just like the window now while I'm sending.
ampctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows                                               
ampctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows                                               
libgcc_s_sjlj-1.dll: PE32 executable (DLL) (console) Intel 80386, for MS Windows                                                                
libhamlib-4.dll:     PE32+ executable (DLL) (console) x86-64 (stripped to external PDB), for MS Windows                                         
libusb-1.0.dll:      PE32+ executable (DLL) (console) x86-64, for MS Windows                                                                    
libwinpthread-1.dll: PE32+ executable (DLL) (console) x86-64, for MS Windows                                                                    
rigctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows                                               
rigctlcom.exe:       PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows                                               
rigctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows                                               
rigmem.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows                                               
rigsmtr.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows                                               
rigswr.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows                                               
rigtestlibusb.exe:   PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows                                               
rotctl.exe:          PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows                                               
rotctld.exe:         PE32+ executable (console) x86-64 (stripped to external PDB), for MS Windows

N0NB, 2022-01-03:
BTW, the Markdown formatting only works from the GitHub Web page.  From email the system doesn't process anything it would appear.

N0NB, 2022-01-12:
Apparently this is fixed so closing.

I've decided not to tamper with already released files so this fix applies to the daily snapshots and any forthcoming releases.

closed 2022-01-12 by N0NB