Hamlib/Hamlib #923: Add Barrett 4050

MrSon888, 2022-01-07:
Hi,

The new (2017) Barrett 4050 SDR is a IP based radio which allows for RS232 over IP. It also has the audio streams via IP.

Is it possible to have the Barrett 4050 added? I have all the RS232 commands if required and the documentation.

mdblack98, 2022-01-07:
Please send the command info.  I don't see it on their website.

Mike W9MDB

MrSon888, 2022-01-07:
Hi Mike, how can I email you directly?

Sent from ProtonMail mobile

-------- Original Message --------

mdblack98, 2022-01-07:
***@***.***

Mike W9MDB

mdblack98, 2022-01-07:
Look on my QRZ page.

Seems email addresses are being blocked now.

Mike W9MDB

MrSon888, 2022-01-07:
I have sent you an email mate, please let me know if you have received it.

mdblack98, 2022-01-08:
Working now as long as rig is on a non-protected channel.
Have now set default channel to 9999 to see if it works.

We should be able to add audio streaming alongside the new IC-7300 streaming.

mdblack98, 2022-01-09:
4050 is supposed to recognize temporary freq commands like this.
TR14075000
E0

TT14075000
E0
We don't want to use the "P" command as that writes  to flash memory which has limited write lifetime.
Waiting for a response from Barrett to see why this isn't working.
Fallback is to use channelized bands like in the Barrett 950 to minimize writes.

closed 2022-01-18 by mdblack98

MrSon888, 2022-04-18:
Hi Mate,

Hope all is well, I'd like to pick this up again.

Barrett said they removed the T commands. However I updated to their latest firmware 1.7.8 = wondering if anything has changed.

Where too from here?

Cheers

Jamie

Sent with [ProtonMail](https://protonmail.com/) secure email.

mdblack98, 2022-04-18:
Is it not working now?  I'll have to contact Barret if they are removing that command...would want to know why.

Mike W9MDB

mdblack98, 2022-04-18:
Actually I did make the 4050 channelized instead of using the T command.
So I hope it is still working, right?
Mike W9MDB