Hamlib/Hamlib #927: MinGW build failure in `kit` backend

N0NB, 2022-01-16:
I see the following from GCC in the `kit` backend:

```
Making install in rigs/kit
make[1]: Entering directory '/home/nate/builds/hamlib-4.5~git/rigs/kit'
  CC       elektor304.lo
  CC       drt1.lo
  CC       dwt.lo
  CC       usrp.lo
  CC       elektor507.lo
  CC       dds60.lo
  CC       miniVNA.lo
  CC       si570avrusb.lo
  CC       funcube.lo
  CC       fifisdr.lo
elektor507.c:673:62: error: unknown type name ‘val_len’; did you mean ‘va_end’?
 int elektor507_get_conf2(RIG *rig, token_t token, char *val, val_len)
                                                              ^~~~~~~
                                                              va_end
elektor507.c: In function ‘elektor507_get_conf’:
elektor507.c:698:12: warning: implicit declaration of function ‘elektor507_get_conf2’; did you mean ‘elektor507_get_conf’? [-Wimplicit-function-declaration]
     return elektor507_get_conf2(rig, token, val, 128);
            ^~~~~~~~~~~~~~~~~~~~
            elektor507_get_conf
make[1]: *** [Makefile:538: elektor507.lo] Error 1
make[1]: *** Waiting for unfinished jobs....
si570avrusb.c: In function ‘si570xxxusb_get_conf’:
si570avrusb.c:965:12: warning: implicit declaration of function ‘si570xxusb_get_conf2’; did you mean ‘si570xxxusb_get_conf2’? [-Wimplicit-function-declaration]
     return si570xxusb_get_conf2(rig, token val, 128);
            ^~~~~~~~~~~~~~~~~~~~
            si570xxxusb_get_conf2
si570avrusb.c:965:43: error: expected ‘)’ before ‘val’
     return si570xxusb_get_conf2(rig, token val, 128);
                                           ^~~~
                                           )
make[1]: *** [Makefile:538: si570avrusb.lo] Error 1
make[1]: Leaving directory '/home/nate/builds/hamlib-4.5~git/rigs/kit'
```

mdblack98, 2022-01-16:
Fixed.

N0NB, 2022-01-16:
All built okay now.

Thanks, Mike.

closed 2022-01-16 by N0NB