Hamlib/Hamlib #941: rigctld does not stay running when rig is powered off

mdblack98, 2022-01-22:
rigctld used to stay running if rig was powered off so set_powerstat=1 would work and set_powerstat=0 would turn off rig but leave rigctld running still.  This makes remote operations impossible.
This does not work as of 2022-01-22

mikaelnousiainen, 2022-01-22:
In icom.c:

- If `icom_get_usb_echo_off()` and/or `rig_get_freq()` fail in `icom_rig_open()`, it will simply make rig_open fail on line: https://github.com/Hamlib/Hamlib/blob/be992b8a36a999726306f0214d847bbadd6587ea/rigs/icom/icom.c#L1008 -- this has changed during the recent echo off check code refactoring.

Additionally, rigctld will not accept failing `rig_open()` initially, but will do `exit(2)` on line: https://github.com/Hamlib/Hamlib/blob/be992b8a36a999726306f0214d847bbadd6587ea/tests/rigctld.c#L697

 This has not changed in a long time, but we could basically allow rig_open() to fail and let the "main handler loop" for clients handle rig_open() / rig_close() like it does now.

@mdblack98 I'm already working on some improvements to rigctld.c (e.g. locking rig_open/rig_close), so I can try to fix this if we just agree how.

My proposal would be to:

1) Allow `icom_rig_open()` in `icom.c` to succeed after 1 retry of getting echo off / frequency, but probably with `priv->poweron = 0`.

2) Remove the `exit(2)` in `rigctld.c` in case `rig_open()` fails -- or potentially have a conf setting to allow this. We can move `rig_opened` to a global variable and have it reflect the initial result of `rig_open()`, so that the client handler will re-attempt opening.

I can push out a PR soon about this proposal.

mdblack98, 2022-01-22:
I don't think we need #1.  #2 sounds OK as long as priv->poweron = 0.  If they are asking for the rig to power-on at startup and it fails it should fail hard.
Mike W9MDB

mikaelnousiainen, 2022-01-24:
These issues were addressed in #943

closed 2022-01-26 by mdblack98