Hamlib/Hamlib #946: Some more MinGW issues

N0NB, 2022-01-31:
Hi Mike.

I just tried building from SHA1 2e218c56 on MinGW and get the following:

```
Making install in src
make[1]: Entering directory '/home/nate/builds/hamlib-4.5~git/src'
make  install-am
make[2]: Entering directory '/home/nate/builds/hamlib-4.5~git/src'
  CC       serial.lo
  CC       misc.lo
  CC       register.lo
  CC       event.lo
  CC       cal.lo
  CC       conf.lo
  CC       tones.lo
  CC       rotator.lo
  CC       locator.lo
  CC       rot_reg.lo
  CC       rot_conf.lo
  CC       rot_settings.lo
  CC       rot_ext.lo
  CC       iofunc.lo
  CC       ext.lo
  CC       mem.lo
  CC       settings.lo
  CC       parallel.lo
  CC       usb_port.lo
  CC       debug.lo
  CC       network.lo
  CC       cm108.lo
  CC       gpio.lo
  CC       microham.lo
  CC       amplifier.lo
In file included from network.c:79:
network.c: In function ‘network_init’:
misc.h:166:91: error: ‘rig’ undeclared (first use in this function)
                         rig_debug(RIG_DEBUG_VERBOSE, "%.*s%d:%s(%d):%s return(%ld) %s\n", rig->state.depth, spaces(), rig->state.depth, __FILENAME__, __LINE__, __func__, (long int) (rctmp), rctmp<0?rigerror(rctmp):""); \
                                                                                           ^~~
../include/hamlib/rig.h:3276:212: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) do { strncpy(debugmsgsave3, debugmsgsave2,sizeof(debugmsgsave3));strncpy(debugmsgsave2, debugmsgsave, sizeof(debugmsgsave2));snprintf(debugmsgsave,sizeof(debugmsgsave),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); } while(0);
                                                                                                                                                                                                                    ^~~~~~~~~~~
network.c:184:13: note: in expansion of macro ‘RETURNFUNC’
             RETURNFUNC(-RIG_EIO);
             ^~~~~~~~~~
misc.h:166:91: note: each undeclared identifier is reported only once for each function it appears in
                         rig_debug(RIG_DEBUG_VERBOSE, "%.*s%d:%s(%d):%s return(%ld) %s\n", rig->state.depth, spaces(), rig->state.depth, __FILENAME__, __LINE__, __func__, (long int) (rctmp), rctmp<0?rigerror(rctmp):""); \
                                                                                           ^~~
../include/hamlib/rig.h:3276:212: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) do { strncpy(debugmsgsave3, debugmsgsave2,sizeof(debugmsgsave3));strncpy(debugmsgsave2, debugmsgsave, sizeof(debugmsgsave2));snprintf(debugmsgsave,sizeof(debugmsgsave),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); } while(0);
                                                                                                                                                                                                                    ^~~~~~~~~~~
network.c:184:13: note: in expansion of macro ‘RETURNFUNC’
             RETURNFUNC(-RIG_EIO);
             ^~~~~~~~~~
In file included from network.c:81:
network.c: In function ‘multicast_publisher_create_data_pipe’:
misc.h:159:27: error: ‘rig’ undeclared (first use in this function)
 #define ENTERFUNC {     ++rig->state.depth; \
                           ^~~
misc.h:159:27: note: in definition of macro ‘ENTERFUNC’
 #define ENTERFUNC {     ++rig->state.depth; \
                           ^~~
network.c: In function ‘multicast_publisher_close_data_pipe’:
misc.h:159:27: error: ‘rig’ undeclared (first use in this function)
 #define ENTERFUNC {     ++rig->state.depth; \
                           ^~~
misc.h:159:27: note: in definition of macro ‘ENTERFUNC’
 #define ENTERFUNC {     ++rig->state.depth; \
                           ^~~
In file included from network.c:79:
network.c: In function ‘multicast_publisher_write_data’:
misc.h:166:91: error: ‘rig’ undeclared (first use in this function)
                         rig_debug(RIG_DEBUG_VERBOSE, "%.*s%d:%s(%d):%s return(%ld) %s\n", rig->state.depth, spaces(), rig->state.depth, __FILENAME__, __LINE__, __func__, (long int) (rctmp), rctmp<0?rigerror(rctmp):""); \
                                                                                           ^~~
../include/hamlib/rig.h:3276:212: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) do { strncpy(debugmsgsave3, debugmsgsave2,sizeof(debugmsgsave3));strncpy(debugmsgsave2, debugmsgsave, sizeof(debugmsgsave2));snprintf(debugmsgsave,sizeof(debugmsgsave),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); } while(0);
                                                                                                                                                                                                                    ^~~~~~~~~~~
network.c:490:9: note: in expansion of macro ‘RETURNFUNC’
         RETURNFUNC(-RIG_EIO);
         ^~~~~~~~~~
network.c: In function ‘multicast_publisher_read_data’:
misc.h:166:91: error: ‘rig’ undeclared (first use in this function)
                         rig_debug(RIG_DEBUG_VERBOSE, "%.*s%d:%s(%d):%s return(%ld) %s\n", rig->state.depth, spaces(), rig->state.depth, __FILENAME__, __LINE__, __func__, (long int) (rctmp), rctmp<0?rigerror(rctmp):""); \
                                                                                           ^~~
../include/hamlib/rig.h:3276:212: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) do { strncpy(debugmsgsave3, debugmsgsave2,sizeof(debugmsgsave3));strncpy(debugmsgsave2, debugmsgsave, sizeof(debugmsgsave2));snprintf(debugmsgsave,sizeof(debugmsgsave),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); } while(0);
                                                                                                                                                                                                                    ^~~~~~~~~~~
network.c:515:9: note: in expansion of macro ‘RETURNFUNC’
         RETURNFUNC(result);
         ^~~~~~~~~~
make[2]: *** [Makefile:606: network.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
  CC       amp_reg.lo
make[2]: Leaving directory '/home/nate/builds/hamlib-4.5~git/src'
make[1]: *** [Makefile:716: install] Error 2
make[1]: Leaving directory '/home/nate/builds/hamlib-4.5~git/src'
make: *** [Makefile:606: install-recursive] Error 1
```

mdblack98, 2022-01-31:
> Hi Mike.
>
> I just tried building from SHA1 2e218c5 on MinGW and get the following:

Just fixed it.
Mike W9MDB

N0NB, 2022-01-31:
Indeed those are fixed, now I am seeing the following errors in `rigctl_parse` causing `rigtcl.exe` and `rigctld.exe` to fail:

```
make[1]: Leaving directory '/home/nate/builds/hamlib-4.5~git/src'
Making install in tests
make[1]: Entering directory '/home/nate/builds/hamlib-4.5~git/tests'
  CC       rigctl-rigctl_parse.o
  CC       rigctl-rigctl.o
  CC       rigctld-rigctld.o
  CC       rigctl-dumpcaps.o
  CC       rigctld-rigctl_parse.o
  CC       rigctld-dumpcaps.o
  CC       rigmem-rigmem.o
  CC       rigmem-memsave.o
  CC       rigmem-memload.o
  CC       rigmem-memcsv.o
  CC       rigsmtr.o
  CC       rigswr.o
  CC       rotctl-rotctl.o
  CC       rotctl-rotctl_parse.o
  CC       rotctl-dumpcaps_rot.o
  CC       rotctld-rotctld.o
  CC       rotctld-rotctl_parse.o
  CC       rotctld-dumpcaps_rot.o
  CC       rigctlcom-rigctlcom.o
  CC       rigctlcom-rigctl_parse.o
  CC       rigctlcom-dumpcaps.o
  CC       ampctl-ampctl.o
  CC       ampctl-ampctl_parse.o
  CC       ampctl-dumpcaps_amp.o
  CC       ampctld-ampctld.o
  CC       ampctld-ampctl_parse.o
  CC       ampctld-dumpcaps_amp.o
  CC       rigtestlibusb-rigtestlibusb.o
  CCLD     rigctl.exe
  CCLD     rigctld.exe
  CCLD     rigmem.exe
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_set_password':
/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4938: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4938: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_halt':
/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4890: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4890: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_chk_vfo':
/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4862: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o:/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4862: more undefined references to `spaces' follow
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:1025: rigctl.exe] Error 1
make[1]: *** Waiting for unfinished jobs....
/usr/bin/i686-w64-mingw32-ld: rigctld-rigctl_parse.o: in function `rigctl_set_password':
/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4938: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4938: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: rigctld-rigctl_parse.o: in function `rigctl_halt':
/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4890: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4890: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: rigctld-rigctl_parse.o: in function `rigctl_chk_vfo':
/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4862: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: rigctld-rigctl_parse.o:/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4862: more undefined references to `spaces' follow
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:1033: rigctld.exe] Error 1
make[1]: Leaving directory '/home/nate/builds/hamlib-4.5~git/tests'
make: *** [Makefile:606: install-recursive] Error 1
```

mdblack98, 2022-01-31:
Try again...
Mike

N0NB, 2022-01-31:
There are still a few more:

```
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_chk_vfo':
/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4862: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4862: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4873: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4873: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o: in function `rigctl_get_trn':
/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4049: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: rigctl-rigctl_parse.o:/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4049: more undefined references to `spaces' follow
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:1025: rigctl.exe] Error 1
make[1]: *** Waiting for unfinished jobs....
/usr/bin/i686-w64-mingw32-ld: rigctld-rigctl_parse.o: in function `rigctl_chk_vfo':
/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4862: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4862: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4873: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: /home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4873: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: rigctld-rigctl_parse.o: in function `rigctl_get_trn':
/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4049: undefined reference to `spaces'
/usr/bin/i686-w64-mingw32-ld: rigctld-rigctl_parse.o:/home/nate/builds/hamlib-4.5~git/tests/rigctl_parse.c:4049: more undefined references to `spaces' follow
collect2: error: ld returned 1 exit status
```

mdblack98, 2022-01-31:
Workin' it....had to change to static build to see them.
Mike W9MDB

N0NB, 2022-01-31:
Everything completed without error as of commit f08c7d7e.

Thanks, Mike.

closed 2022-01-31 by N0NB