Hamlib/Hamlib #956: IO Error

kf0ezb, 2022-02-06:
Fresh install of the latest Linux Mint 20.3, fresh install of Hamlib, trying to get my YAESU FT-991A to connect to WSJTx (all fresh installs) I have been following this guide (http://pa0rob.vandenhoff.info/article/hamlib-rigcltd) and in configuring the ttyUSB ports when I try to open the ports I get an IO Error. Testing by using telnet to localhost fails. The ports show up when the radio is plugged in and disappear when it is not so I think that part is good. Radio is set to 4800 baud. this is the error I get in terminal using the testing method above:  rig_open: error = IO error.

Thanks for any assistance in advance.
Doug Lewis
KF0EZB

mdblack98, 2022-02-06:
Did you add yourself to the dialout group?

In order to run some of the commands featured on this website, you will need to give yourself sudo access, and add yourself to the dialout group. This applies to the primary machine, and any virtual machines running CentOS / Fedora. If you install VirtualBox to run virtualised images, you will also need to add your uid to the vboxusers group.

Open a terminal window and enter: su - to become super-user. You will be prompted for your root password.

Add yourself to the wheel and dialout groups with the following command: usermod -G wheel,dialout uid
- where uid is your id as found from the command: whoami.

In the same terminal window as above, edit /etc/sudoers and un-comment either of the %wheel options, depending on whether you want to input your password each time you run a sudo command, or not. I prefer not, so I un-comment: %wheel    ALL=(ALL)    NOPASSWD: ALL

You will need to log-out and log-in again for the above changes to take effect.
Mike W9MDB

kf0ezb, 2022-02-06:
Yes, after a few more reboots i finally got it to run. However when i test wsjtx the button never goes green, the radio transmits but doesn’t  appear to have any power out.  I have not added myself to the sudoers file yet, ill do that and report back.

Thank you
Doug

kf0ezb, 2022-02-06:
> Android [4].
> Links:

Mike, I made the changes you recommended, Linux Mint doesn't have a 
'Wheel' group they use the 'Sudo' group for that, I rebooted and still 
have the same issue, I wonder if it's a WSHTx issue, I never get a green 
button when testing the radio setup however it enables the test transmit 
button which does cause the radio to key up, however when trying to make 
contacts when the radio keys up my meters show no power output.

Thank You
Doug

mdblack98, 2022-02-06:
If the radio is in TX but no power then your audio path is not working.
Path to sound on Linux and volume level.Sound card
Sound input to rigRig sound settings.
So four things to check.
Mike W9MDB

 
 

Mike, I made the changes you recommended, Linux Mint doesn't have a 
'Wheel' group they use the 'Sudo' group for that, I rebooted and still 
have the same issue, I wonder if it's a WSHTx issue, I never get a green 
button when testing the radio setup however it enables the test transmit 
button which does cause the radio to key up, however when trying to make 
contacts when the radio keys up my meters show no power output.

Thank You
Doug

closed 2022-02-07 by mdblack98