Hamlib/Hamlib #959: TS-590 power ON not work at master (Python)

rn3kk, 2022-02-10:
I tested power on/off for TS-590

```
my_rig.set_powerstat(1)
my_rig.set_powerstat(0)
```
Hamlib 4.4 all OK! But at master my_rig.set_powerstat(1) not work, my_rig.set_powerstat(0) is work. What can be?

rn3kk, 2022-02-11:
I pulled last origin/master. Rebuild hamlib. 
Code:
`   my_rig = Hamlib.Rig(Hamlib.RIG_MODEL_TS590S)
    my_rig.set_conf("rig_pathname", "/dev/ttyUSB0")
    my_rig.set_conf("serial_speed", "115200")
    my_rig.open()

    my_rig.set_powerstat(1)
    time.sleep(5)
    my_rig.close()

Log:
`/usr/bin/python3.6 /home/user/Projects/RemoteAudio/py_server/hamlib_test.py
rig_check_rig_caps: p1=0x7fa576668d20, p2=0x7fa57666f2a0, rig_model=0x7fa576668d20, macro_name=0x7fa57666f2a0
rig_init: backend for TS-590S does not contain hamlib_check_rig_caps
rig_init(460): 0x14fe1fc rs->comm_state==0?=0
rig_init: rig has VFO_A
rig_init: rig has VFO_B
rig_init: rig has VFO_MEM
kenwood_init called, version 20220206/20220206.2
kenwood_init: if_len = 37
kenwood.c(745):kenwood_init return(0) 
rig_token_lookup called for rig_pathname
rig_confparam_lookup called for rig_pathname
rig_set_conf called
rig_confparam_lookup called for 1073741834
rig_set_conf: rig_pathname='/dev/ttyUSB0'
rig_token_lookup called for serial_speed
rig_confparam_lookup called for serial_speed
rig_set_conf called
rig_confparam_lookup called for 1073741844
rig_set_conf: serial_speed='115200'
 1:rig.c(707):rig_open entered
serial_open: /dev/ttyUSB0
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=115200,0x1002
serial_setup: cfsetospeed=115200,0x1002
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: Handshake=Hardware
serial_setup: tcsetattr TCSANOW
read_string_generic called, rxmax=4095 direct=1
  2:rig.c(6942):async_data_handler_start entered
async_data_handler_start: async data support disabled
  2:rig.c(6947):async_data_handler_start return(0) 
rig.c(239):add_opened_rig return(0) 
rig_open: 0x14fe1fc rs->comm_state==1?=1
  2:kenwood.c(767):kenwood_open entered
   3:kenwood.c(1031):kenwood_get_id entered
kenwood_transaction called
kenwood_transaction: cmdstr = ID
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1
write_block(): TX 3 bytes, method=2
0000    49 44 3b                                            ID;             
read_string_generic called, rxmax=128 direct=1
read_string_generic: minlen now 6
read_string_generic(): RX 6 characters, direct=1
0000    49 44 30 32 31 3b                                   ID021;          
kenwood_transaction: read_string(len=6)='ID021;'
kenwood_transaction: returning RIG_OK, retval=0
kenwood.c(584):kenwood_transaction return(0) 
   3:kenwood.c(1033):kenwood_get_id return(0) 
kenwood_open: got ID so try PS
   3:rig.c(5572):rig_get_powerstat entered
../../Hamlib/src/rig.c(5589) trace
    4:kenwood.c(4770):kenwood_get_powerstat entered
kenwood_safe_transaction called
kenwood_transaction called
[... 65 lines cut ...]
ser_close: restoring options
ser_close: restoring options 1:rig.c(5535):rig_set_powerstat entered
 1:rig.c(5535):rig_set_powerstat entered
 1:rig.c(1187):rig_close entered
 1:rig.c(1199):rig_close return(-1) Invalid parameter
 1:rig.c(5539):rig_set_powerstat return(-1) Invalid parameter
 1:rig.c(1098):rig_open return(-5) Communication timed out
ser_close called
ser_close: restoring options
ser_close: restoring options 1:rig.c(5535):rig_set_powerstat entered
 1:rig.c(5535):rig_set_powerstat entered 1:rig.c(1187):rig_close entered
 1:rig.c(1199):rig_close return(-1) Invalid parameter
 1:rig.c(5539):rig_set_powerstat return(-1) Invalid parameter
 1:rig.c(1098):rig_open return(-5) Communication timed out
ser_close called
ser_close: restoring options
ser_close: restoring options 1:rig.c(5535):rig_set_powerstat entered
 1:rig.c(5535):rig_set_powerstat entered 1:rig.c(1187):rig_close entered
 1:rig.c(1187):rig_close entered
 1:kenwood.c(750):kenwood_cleanup entered
 1:kenwood.c(755):kenwood_cleanup return(0) 

Process finished with exit code 0

`

rn3kk, 2022-02-11:
In log I see
`0000 46 56 3b FV;
read_string_generic called, rxmax=8 direct=1
read_string_generic(): Timed out 0.500 seconds after 0 chars, direct=1
kenwood_transaction: read_string(len=0)=''
kenwood_transaction: read_string retval < 0, retval = -5, retry_read=0, rs->rigport.retry=0
kenwood.c(584):kenwood_transaction return(-5) Communication timed out
kenwood_transaction: read_string(len=0)=''
kenwood_transaction: read_string retval < 0, retval = -5, retry_read=0, rs->rigport.retry=0
kenwood.c(584):kenwood_transaction return(-5) Communication timed out
kenwood_transaction: read_string(len=0)=''
kenwood_transaction: read_string retval < 0, retval = -5, retry_read=0, rs->rigport.retry=0
kenwood_transaction: read_string retval < 0, retval = -5, retry_read=0, rs->rigport.retry=0
kenwood_open: cannot get f/w version`

but if radio is OFF,  cmd **FV;** is not performed. the radio does not return anything

rn3kk, 2022-02-11:
And, if use 590SG, turn ON and turn OFF is work (ON - long time)
my_rig = Hamlib.Rig(Hamlib.RIG_MODEL_TS590SG)

rn3kk, 2022-02-11:
TS-590S, before my_rig.open() need be ON. But if set "Auto power ON",  my_rig.open() work. 

`  my_rig = Hamlib.Rig(Hamlib.RIG_MODEL_TS590S)
    my_rig.set_conf("rig_pathname", "/dev/ttyUSB0")
    my_rig.set_conf("serial_speed", "115200")
    my_rig.set_conf("1073741948", "1") #poweron - need for correct open ts590
    my_rig.open()`

I will think.

mdblack98, 2022-02-11:
Do you know the firmware version on your rig?And it is a 590S and not 590SG?
It appears the it does not support the "FV" command at all so maybe it's an early serial number?
Pull the latest hamlib and see if it works now.
Mike W9MDB

rn3kk, 2022-02-12:
> 

TS-590S
firmware 2.05
s/n B3A00133

Build from master, but not work.. 

cmd FV work only if radio is powered on. And no answer, if radio powered is off.

rn3kk, 2022-02-12:
```
    my_rig = Hamlib.Rig(Hamlib.RIG_MODEL_TS590S)
    my_rig.set_conf("rig_pathname", "/dev/ttyUSB0")
    my_rig.set_conf("serial_speed", "115200")
    my_rig.open()
    my_rig.set_powerstat(1)

    while not terminate:
        time.sleep(0.1)

    my_rig.close()
```

`/usr/bin/python3.6 /home/user/Projects/RemoteAudio/py_server/hamlib_test.py
rig_check_rig_caps: p1=0x7f51ae0ded20, p2=0x7f51ae0e52a0, rig_model=0x7f51ae0ded20, macro_name=0x7f51ae0e52a0
rig_init(460): 0x288632c rs->comm_state==0?=0
rig_init: rig has VFO_A
rig_init: rig has VFO_B
rig_init: rig has VFO_MEM
kenwood_init called, version 20220211/20220211.2
kenwood_init: if_len = 37
kenwood.c(745):kenwood_init return(0) 
rig_token_lookup called for rig_pathname
rig_confparam_lookup called for rig_pathname
rig_set_conf called
rig_confparam_lookup called for 1073741834
rig_set_conf: rig_pathname='/dev/ttyUSB0'
rig_token_lookup called for serial_speed
rig_confparam_lookup called for serial_speed
rig_set_conf called
rig_confparam_lookup called for 1073741844
rig_set_conf: serial_speed='115200'
 1:rig.c(707):rig_open entered
serial_open: /dev/ttyUSB0
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=115200,0x1002
serial_setup: cfsetospeed=115200,0x1002
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: Handshake=Hardware
serial_setup: tcsetattr TCSANOW
read_string_generic called, rxmax=4095 direct=1
  2:rig.c(6942):async_data_handler_start entered
async_data_handler_start: async data support disabled
  2:rig.c(6947):async_data_handler_start return(0) 
rig.c(239):add_opened_rig return(0) 
rig_open: 0x288632c rs->comm_state==1?=1
  2:kenwood.c(767):kenwood_open entered
   3:kenwood.c(1034):kenwood_get_id entered
kenwood_transaction called
kenwood_transaction: cmdstr = ID
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1
write_block(): TX 3 bytes, method=2
0000    49 44 3b                                            ID;             
read_string_generic called, rxmax=128 direct=1
read_string_generic: minlen now 6
read_string_generic(): RX 6 characters, direct=1
0000    49 44 30 32 31 3b                                   ID021;          
kenwood_transaction: read_string(len=6)='ID021;'
kenwood_transaction: returning RIG_OK, retval=0
kenwood.c(584):kenwood_transaction return(0) 
   3:kenwood.c(1036):kenwood_get_id return(0) 
kenwood_open: got ID so try PS
   3:rig.c(5572):rig_get_powerstat entered
../../Hamlib/src/rig.c(5589) trace
    4:kenwood.c(4775):kenwood_get_powerstat entered
kenwood_safe_transaction called
kenwood_transaction called
[... 153 lines cut ...]
kenwood_open: no response from rig
  2:rig.c(6984):async_data_handler_stop entered
  2:rig.c(7014):async_data_handler_stop return(0) 
ser_close called
ser_close: restoring options
 1:rig.c(1098):rig_open return(-5) Communication timed out
ser_close called
ser_close: restoring options
 1:rig.c(1098):rig_open return(-5) Communication timed out
ser_close called
ser_close: restoring options
ser_close: restoring options
 1:rig.c(5535):rig_set_powerstat entered
 1:rig.c(5539):rig_set_powerstat return(-1) Invalid parameter
 1:rig.c(1098):rig_open return(-5) Communication timed out
ser_close called
ser_close: restoring options
ser_close: restoring options 1:rig.c(5535):rig_set_powerstat entered
 1:rig.c(5539):rig_set_powerstat return(-1) Invalid parameter
 1:rig.c(1098):rig_open return(-5) Communication timed out
ser_close called
ser_close: restoring options
ser_close: restoring options 1:rig.c(5535):rig_set_powerstat entered
 1:rig.c(5535):rig_set_powerstat entered
`

rn3kk, 2022-02-14:
my_rig.open()
my_rig.set_conf("auto_power_on", "1")

works.

closed 2022-02-14 by mdblack98