Hamlib/Hamlib #961: 991A

k0100311, 2022-02-12:
start C:\GPredict\gpredict.exe

start C:\GPredict\hamlib\bin\rigctld.exe -v -r \\.\com3 -m 1035 -s 9600 -t 4532 -C "serial_speed=9600,stop_bits=1,rts_state=ON,dtr_state=OFF,serial_handshake=None" 

When using this command I need it to stay in fm mode cannot make it work tried many times thank you

mdblack98, 2022-02-12:
> Please send debug log
> C:\GPredict\hamlib\bin\rigctld.exe -vvvvv -r \.\com3 -m 1035 -s 9600 -t
> f >log.txt 2>&1
> And send me the log.txt
> Mike W9MDB

Download the DLL or package 20220213 when it's build later tonight.
Or if you're running the 64-bit you can test this DLL.
https://www.dropbox.com/s/7aejdv74nfjxr0q/libhamlib-4.dll?dl=0

Mike W9MDB

 
 

Thanks

rigctld.c(579) Startup: C:\GPredict\hamlib\bin\rigctld.exe -vvvvv -r \.\com3 -m 1035 -s 9600 -t 4532 -C serial_speed=9600,stop_bits=1,rts_state=ON,dtr_state=OFF,serial_handshake=None f
rigctld Hamlib 4.4 Thu Dec 02 23:46:51 2021 +0000 SHA=5f8c4c
Report bugs to ***@***.***>

rig_check_rig_caps: p1=0000000052ED0BE0, p2=0000000052ED7158, rig_model=0000000052ED0BE0, macro_name=0000000052ED7158
rig.c(359):rig_check_rig_caps return(0)
yaesu: initrigs4_yaesu called
rig_register called
rig_register: rig_register (1021)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1027)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1010)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1005)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1006)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1007)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1039)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1009)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1020)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1001)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1022)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1023)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1011)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1015)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1013)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1014)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1028)
register.c(225):rig_register return(0)
rig_register called
[... 97 lines cut ...]
rig_confparam_lookup called
rig_set_conf called
rig_confparam_lookup called
rig_set_conf: dtr_state='OFF'
rig_token_lookup called
rig_confparam_lookup called
rig_set_conf called
rig_confparam_lookup called
rig_set_conf: serial_handshake='None'
rigctl_parse.c(1974):set_conf return(0)
main: twiddle=0, uplink=0, twiddle_rit=0
rig.c(688):rig_open entered
rig_open: using network address \.\com3
iofunc.c(68):port_open entered
network.c(171):network_open entered
network_init: WSAStartup OK
network_open: hoststr=\.\com3, portstr=
network_open: cannot get host "\.\com3": No such host is known. 
network.c(254):network_open return(-2)
iofunc.c(177):port_open return(-2)
rig.c(800):rig_open return(-2)
rig_open: error = Invalid configuration
network.c(254):network_open return(-2)
iofunc.c(177):port_open return(-2)
rig.c(800):rig_open return(-2) \.\com3 No error

mdblack98, 2022-02-13:
Image did not come through....do you have the debug log?
Mike W9MDB

 
 

[image: image.png]
error code comes up

k0100311, 2022-02-13:
does not degenerate err in log file for me

mdblack98, 2022-02-14:
So does that mean things are working now?

k0100311, 2022-02-15:
No the error occurs still not working locks up

mdblack98, 2022-02-15:
Please send a debug log again.
And use com3 instead of \.\com3

k0100311, 2022-02-15:
It does not create a log the program sends
error message --->Entrypoint not found rigctrld.exe

k0100311, 2022-02-17:
It is not
I am unable to create a logfile program hangs b4 it is created

mdblack98, 2022-02-17:
I'd like to hook with you using AnyDesk from https://anydesk.com/en
Contact me at mdblack98@yahoo.com

k0100311, 2022-02-17:
invalid email address ***@***.***

mdblack98, 2022-02-17:
What's your callsign?
Look me up QRZ and get my email address from there.
Mike W9MDB

k0100311, 2022-02-18:
Your callsign is correct when I click invite and add you r address it tells
me it is invalid.

closed 2022-06-29 by mdblack98