Hamlib/Hamlib #966: Hamlib persistence

mdblack98, 2022-02-17:
Add the ability for hamlib clients to store data in a config file.
Plan to use $HOME for Unix and %HOMEPATH% for Windows.
$HOME/hamlib.cfg
Will store --set-conf commands.
Config file will be read on open call and will attempt to set_conf items and ignore errors since they may not apply to current selection.
Command line parameters need to override config file.

GeoBaltz, 2022-06-03:
Please, no more lone files in $HOME.  The FreeDesktop spec recommends these go in $XDG_CONFIG_HOME (default $HOME./config/ ), so ~/.config/hamlib/hamlib.cfg.

mdblack98, 2022-06-03:
I see no such environment variable defined by default.

GeoBaltz, 2022-06-03:
Yeah, I was going by the spec, not the real world.  Looks like everybody has to know/use the default.  Sigh.

"Standards are wonderful Everybody should have one." :-)

GeoBaltz, 2022-06-03:
FWIW - On this computer, out of the Ham related packages -
$HOME has these hidden directories ~/.fldigi, ~/.fllog, ~/.fltk, ~/.klog, ~/.tr4w, ~/.tqsl, ~/.qrq
$HOME/.config has these directories - cqrlog/, ON4QZ/, and file WSJT-X.ini

So the rest of the ham software world is still split.  But I still have no visible files in $HOME, and about 110 in $HOME/.config/.

N0NB, 2022-06-03:
* On 2022 03 Jun 10:12 -0500, GeoBaltz wrote:
> Please, no more lone files in $HOME.  The FreeDesktop spec recommends
> these go in $XDG_CONFIG_HOME (default $HOME./config/ ), so
> ~/.config/hamlib/hamlib.cfg.

I agree that when practical the $HOME/.config should be preferred.  I
don't where that environment variable is supposed to be defined.  Even
on Gnome on Debian is is not defined:

$ set | grep XDG
XDG_CURRENT_DESKTOP=GNOME
XDG_DATA_DIRS=/usr/local/share/:/usr/share/:/var/lib/snapd/desktop
XDG_MENU_PREFIX=gnome-
XDG_RUNTIME_DIR=/run/user/1000
XDG_SESSION_CLASS=user
XDG_SESSION_DESKTOP=gnome
XDG_SESSION_TYPE=wayland
    local -a dirs=(${BASH_COMPLETION_USER_DIR:-${XDG_DATA_HOME:-$HOME/.local/share}/bash-completion}/completions);
    for dir in ${XDG_DATA_DIRS:-/usr/local/share:/usr/share};

I'm not 100% sure what convention is regarding creation of
$HOME/.config.  It seems like the first application that wants to put
something there creates it if it doesn't exist.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

mdblack98, 2022-06-03:
I don't see a .config on my Windows system.  So nothing is using it there.
Mike W9MDB

 
 

I agree that when practical the $HOME/.config should be preferred. I
don't where that environment variable is supposed to be defined. Even
on Gnome on Debian is is not defined:

$ set | grep XDG
XDG_CURRENT_DESKTOP=GNOME
XDG_DATA_DIRS=/usr/local/share/:/usr/share/:/var/lib/snapd/desktop
XDG_MENU_PREFIX=gnome-
XDG_RUNTIME_DIR=/run/user/1000
XDG_SESSION_CLASS=user
XDG_SESSION_DESKTOP=gnome
XDG_SESSION_TYPE=wayland
 local -a dirs=(${BASH_COMPLETION_USER_DIR:-${XDG_DATA_HOME:-$HOME/.local/share}/bash-completion}/completions);
 for dir in ${XDG_DATA_DIRS:-/usr/local/share:/usr/share};

I'm not 100% sure what convention is regarding creation of
$HOME/.config. It seems like the first application that wants to put
something there creates it if it doesn't exist.

73, Nate

GeoBaltz, 2022-06-03:
At least on this system (openSUSE Leap 15.3), .config appears in /etc/skel, so it is created when the user's original $HOME is created.  Every user should have one.

jmerdich, 2022-06-29:
> I don't see a .config on my Windows system.  So nothing is using it there.

That's because every OS sadly has a different standard for this. As someone who's had to deal with this for work, here are the standards:

 - [Linux](https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html): somewhere in $XDG_CONFIG_HOME, falling back to ~/.config
 - Windows: %APPDATA%.
 - [MacOS](https://developer.apple.com/library/archive/documentation/FileManagement/Conceptual/FileSystemProgrammingGuide/FileSystemOverview/FileSystemOverview.html#//apple_ref/doc/uid/TP40010672-CH2-SW1): "~/Library/Application Support".
 - [Android](https://developer.android.com/training/data-storage/app-specific): Context.getFilesDir(). You also have getExternalFilesDir, which is user accessible but might be on a removable card.
 - Other platforms: ???

Messy? Yes it is. Naturally, a lot of apps either predated the many standards or wanted One Thing To Rule All Platforms, so they added a hidden file/dir (dot-prefix = hidden for mac/linux) in the home directory. That works everywhere except maybe android.

N0NB, 2022-10-11:
* On 2022 03 Jun 12:55 -0500, Michael Black wrote:
> I don't see a .config on my Windows system.  So nothing is using it there.

Nor would I expect to see it there.  This is for Linux/BSD configuration.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

N0NB, 2022-10-11:
* On 2022 03 Jun 13:33 -0500, GeoBaltz wrote:
> At least on this system (openSUSE Leap 15.3), .config appears in
> /etc/skel, so it is created when the user's original $HOME is created.
> Every user should have one.

That's the difference between distributions.  Debian lacks it:

$ ls -la /etc/skel/
total 28
drwxr-xr-x   2 root root  4096 Aug 15  2021 ./
drwxr-xr-x 151 root root 12288 Jun  2 07:17 ../
-rw-r--r--   1 root root   220 Feb 25  2020 .bash_logout
-rw-r--r--   1 root root  3526 Feb 25  2020 .bashrc
-rw-r--r--   1 root root   807 Feb 25  2020 .profile

I just checked an Arch Linux virtual machine and it is similar though
having .bash_profile instead of .profile.

Yet ~/.config has been created and populated on both my Arch and Debian
installations.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819