Hamlib/Hamlib #968: Add send/get raw command

Johnny-Sierra, 2022-02-18:
Add in the library functions to send/get raw command to the connected radio, something like the -w -W command in rigctl utility.
It will be useful for debug, specific commands or user's macro.

mdblack98, 2022-02-18:
Already there -- the w will time out but the W command takes a length argument which works for fixed length commands
Could also make W accept a terminator char instead of a length if needed.  Some commands are variable length responses.
Rig command: W ID; 6
Reply: ID900;

Can also do hexadecimal

Rig command: W \0x49\0x44\0x3b 6
Reply: ID900;

mdblack98, 2022-02-19:
I guess you mean adding a function in rig.h to do this.
Sounds reasonable
Proposed call
int rig_send_raw(rig, unsigned char *send, int send_len, unsigned char *reply, int reply_len, unsigned char term);
send contains the raw command data
send_len is the # of bytes to send
If reply is NULL no answer is expected
reply should be as long as need for any reply
term is the command termination char -- could be semicolon, CR, or 0xfd for Icom rigs

Johnny-Sierra, 2022-02-19:
Yes, I mean into rig.h
Your proposal is good.

PianetaRadio, 2022-06-01:
I'm studying the code to help you in adding this function, but I'm not able to know enough the rig.h and rig.c to write a working code. Sorry

closed 2022-06-08 by mdblack98