Hamlib/Hamlib #982: Connection is unstable

Supermagnum, 2022-03-08:
Hello, for some reason I got a lot of these:
Hamlib error: IO error
port_open: serial_open(/dev/ttyUSB0) status=-6, err=No such file or directory
iofunc.c(81):port_open return(-6)
rig.c(800):rig_open return(-6) while opening connection to rig

Timestamp: 2022-03-08T09:00:08.838Z

Hamlib error: IO error
port_open: serial_open(/dev/ttyUSB0) status=-6, err=No such file or directory
iofunc.c(81):port_open return(-6)
rig.c(800):rig_open return(-6) while opening connection to rig

Timestamp: 2022-03-08T09:05:52.387Z

Hamlib error: IO error
port_open: serial_open(/dev/ttyUSB0) status=-6, err=No such file or directory
iofunc.c(81):port_open return(-6)
rig.c(800):rig_open return(-6) while opening connection to rig

Timestamp: 2022-03-08T09:07:11.743Z

It seems that the connection for some reason goes up and down, 
I then reconfigure the radio with exact the same settings in wsjtx, then it works for a while and the same happens.
It is not related to any RFI because it happens even if I am only listening.

The rig is a FT-991. I am running Linux mint.
Could this be a bad USB cable ?

mdblack98, 2022-03-08:
You can still have RFI even if you aren't transmitting but I agree that it's unlikely as it would be a much lower level.  You can get 60Hz for example from wall warts, LED lights, grow lights, power company transformers, and such.If there's "no such file" then the device has disappeared and if you run "dmesg" you'll probably see the USB messages on the device disappearing and reappearing.
I would run overnight with WSJT-X not running and see if the USB cable still disconnects (in dmesg).
Then I would try another USB cable.
Mike W9MDB

Supermagnum, 2022-03-09:
It looks like the issue is related to wsjtx-z, because it doesn't happen with the normal newest wsjt and Hamlib 4.4

Supermagnum, 2022-03-09:
Here is the relevant log from dmesg, it does reset itself. I will try to get hold of a new USB cable.
Hopefully that fixes the issue.

[    8.206915] usb 1-6: USB disconnect, device number 4
[    8.206918] usb 1-6.1: USB disconnect, device number 5
[    8.207122] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[    8.207144] cp210x 1-6.1:1.0: device disconnected
[    8.207282] cp210x ttyUSB1: cp210x converter now disconnected from ttyUSB1
[    8.207300] cp210x 1-6.1:1.1: device disconnected
[    8.212754] usb 1-6.2: USB disconnect, device number 6
[    8.256813] EDAC amd64: Node 0: DRAM ECC disabled.
[    8.256816] EDAC amd64: ECC disabled in the BIOS or no ECC capability, module will not load.
                Either enable ECC checking or force module loading by setting 'ecc_enable_override'.
                (Note that use of the override may cause unknown side effects.)
[    8.258673] kvm: disabled by bios
[    8.354510] EDAC amd64: Node 0: DRAM ECC disabled.
[    8.354513] EDAC amd64: ECC disabled in the BIOS or no ECC capability, module will not load.
                Either enable ECC checking or force module loading by setting 'ecc_enable_override'.
                (Note that use of the override may cause unknown side effects.)
[    8.354568] kvm: disabled by bios
[    8.454130] EDAC amd64: Node 0: DRAM ECC disabled.
[    8.454132] EDAC amd64: ECC disabled in the BIOS or no ECC capability, module will not load.
                Either enable ECC checking or force module loading by setting 'ecc_enable_override'.
                (Note that use of the override may cause unknown side effects.)
[    8.563749] usb 1-6: new high-speed USB device number 7 using xhci_hcd
[    8.787762] usb 1-6: New USB device found, idVendor=0424, idProduct=2512, bcdDevice= b.b3
[    8.787765] usb 1-6: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    8.794798] hub 1-6:1.0: USB hub found
[    8.797747] hub 1-6:1.0: 2 ports detected
[    9.119906] usb 1-6.1: new full-speed USB device number 8 using xhci_hcd
[    9.380283] igb 0000:04:00.0 enp4s0: igb: enp4s0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[    9.447801] usb 1-6.1: New USB device found, idVendor=10c4, idProduct=ea70, bcdDevice= 1.00
[    9.447803] usb 1-6.1: New USB device strings: Mfr=1, Product=2, SerialNumber=5
[    9.447804] usb 1-6.1: Product: CP2105 Dual USB to UART Bridge Controller
[    9.447805] usb 1-6.1: Manufacturer: Silicon Labs
[    9.447806] usb 1-6.1: SerialNumber: AH057M5I150145
[    9.474860] cp210x 1-6.1:1.0: cp210x converter detected
[    9.492814] gpio gpiochip2: (cp210x): tried to insert a GPIO chip with zero lines
[    9.492819] gpiochip_add_data_with_key: GPIOs 0..-1 (cp210x) failed to register, -22
[    9.492823] cp210x 1-6.1:1.0: GPIO initialisation failed: -22
[    9.512883] usb 1-6.1: cp210x converter now attached to ttyUSB0
[    9.518844] cp210x 1-6.1:1.1: cp210x converter detected
[    9.536815] gpio gpiochip2: (cp210x): tried to insert a GPIO chip with zero lines
[    9.536820] gpiochip_add_data_with_key: GPIOs 0..-1 (cp210x) failed to register, -22
[    9.536823] cp210x 1-6.1:1.1: GPIO initialisation failed: -22
[    9.556878] usb 1-6.1: cp210x converter now attached to ttyUSB1
[    9.603984] IPv6: ADDRCONF(NETDEV_CHANGE): enp4s0: link becomes ready
[    9.643732] usb 1-6.2: new full-speed USB device number 9 using xhci_hcd
[   10.257935] usb 1-6.2: New USB device found, idVendor=08bb, idProduct=29b3, bcdDevice= 1.00
[   10.257939] usb 1-6.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   10.257941] usb 1-6.2: Product: USB Audio CODEC 
[   10.257943] usb 1-6.2: Manufacturer: Burr-Brown from TI              
[   10.617199] input: Burr-Brown from TI               USB Audio CODEC  as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-6/1-6.2/1-6.2:1.3/0003:08BB:29B3.0005/input/input22
[   10.675981] hid-generic 0003:08BB:29B3.0005: input,hidraw3: USB HID v1.00 Device [Burr-Brown from TI               USB Audio CODEC ] on usb-0000:02:00.0-6.2/input3
[  261.681719] usb 1-6: USB disconnect, device number 7
[  261.681723] usb 1-6.1: USB disconnect, device number 8
[  261.682183] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[  261.682205] cp210x 1-6.1:1.0: device disconnected
[  261.682345] cp210x ttyUSB1: cp210x converter now disconnected from ttyUSB1
[  261.682363] cp210x 1-6.1:1.1: device disconnected
[  261.684703] usb 1-6.2: cannot submit urb (err = -19)
[  261.687945] usb 1-6.2: USB disconnect, device number 9
[  262.024770] usb 1-6: new high-speed USB device number 10 using xhci_hcd
[  262.249337] usb 1-6: New USB device found, idVendor=0424, idProduct=2512, bcdDevice= b.b3
[  262.249340] usb 1-6: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  262.256226] hub 1-6:1.0: USB hub found
[  262.259179] hub 1-6:1.0: 2 ports detected
[  262.580908] usb 1-6.1: new full-speed USB device number 11 using xhci_hcd
[  262.904489] usb 1-6.1: New USB device found, idVendor=10c4, idProduct=ea70, bcdDevice= 1.00
[  262.904492] usb 1-6.1: New USB device strings: Mfr=1, Product=2, SerialNumber=5
[... 33 lines cut ...]
[  798.918750] hub 1-6:1.0: USB hub found
[  798.921695] hub 1-6:1.0: 2 ports detected
[  799.243201] usb 1-6.1: new full-speed USB device number 14 using xhci_hcd
[  799.566795] usb 1-6.1: New USB device found, idVendor=10c4, idProduct=ea70, bcdDevice= 1.00
[  799.566799] usb 1-6.1: New USB device strings: Mfr=1, Product=2, SerialNumber=5
[... cut ...]
[  800.384904] usb 1-6.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  800.384906] usb 1-6.2: Product: USB Audio CODEC 
[  800.384908] usb 1-6.2: Manufacturer: Burr-Brown from TI              
[  800.745185] input: Burr-Brown from TI               USB Audio CODEC  as /devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-6/1-6.2/1-6.2:1.3/0003:08BB:29B3.0007/input/input24
[  800.803239] hid-generic 0003:08BB:29B3.0007: input,hidraw3: USB HID v1.00 Device [Burr-Brown from TI               USB Audio CODEC ] on usb-0000:02:00.0-6.2/input3

Supermagnum, 2022-03-10:
closed due to bad usb cable

closed 2022-03-10 by Supermagnum