Hamlib/Hamlib #987: Rigctld caps-targetable_vfo

PianetaRadio, 2022-03-18:
Using Yaesu FT991A that has caps.targetable_vfo = RIG_TARGETABLE_FREQ
If I connect by usig rigctld, this is what I get:

```
2022-03-18:16:41:44.308355: read_string called, rxmax=1024
2022-03-18:16:41:44.308355: read_string(): RX 19 characters
2022-03-18:16:41:44.308355: 0000    74 61 72 67 65 74 61 62 6c 65 5f 76 66 6f 3d 30     targetable_vfo=0
2022-03-18:16:41:44.308355: 0010    78 33 0a                                            x3.             
2022-03-18:16:41:44.308355: netrigctl_open: unknown setting='targetable_vfo=0x3'
```

and the caps.targetable_vfo is recognized as 0

PianetaRadio, 2022-03-18:
For the sake of clarity, I'm using hamlib v 4.4 and 0x3 comes from:

ft991.c
`    .targetable_vfo =     RIG_TARGETABLE_FREQ | RIG_TARGETABLE_MODE,`

So, rigctld get from backend of FT991 the right value of targetable_vfo, but recognize it as an error!

(In the actual master code RIG_TARGETABLE_MODE was removed, because it was not correct for this radio)

mdblack98, 2022-03-18:
Duplicate of https://github.com/Hamlib/Hamlib/issues/732

closed 2022-03-18 by mdblack98