Hamlib/Hamlib #999: Display frequency does not match get_freq - TS-950SDX

rubienr, 2022-03-27:
# Problem

When `rigctl` connects to TS-950SDX the frequency from the information readout `IF` matches the displayed frequency on the radio, but in contrast when `FA` or `FB` (`get_freq`) is issued the response does not match the displayed frequency. `set_freq` does not set the frequency on the radio display.

# How to reproduce

1. connect `rigctl` to TS-950SDX by `rigctl --model=2013 --rig-file=/dev/ttyUSB0 -vvvvv`
1.  send `IF` command: `W IF; 38`
    1. receive: `IF00014070030`
1. send `FA` command: `W FA; 14`
    1. receive `FA00007074000`, expect `IF00014070030`
1. send `FB` command: `W FB; 14`
    1. receive `FB00010136000`, expect `IF00014070030`

# What I tried

1. Deactivated cache; behaviour did not change:
```
   cache_timeout: "Cache timeout, value of 0 disables caching"
	Default: 500, Value: 500
   	Range: 0.0..5000.0, step 1.0
```
2. Exactly specified the connection parameters; behaviour did not change:
`rigctl --model=2013 --rig-file=/dev/ttyUSB0 --dcd-type=RIG --serial-speed=4800 --set-conf=stop_bits=2,serial_parity=None,serial_handshake=Hardware,data_bits=8 -vvvvv`
3. Configured `grig`, `flrig` or `wsjtx` or so that the frequency/band of the radio can be changed by software.
Whenever a frequency change is made by the GUI it seemingly sends a command to the radio which then beeps but the frequency on the display does not change.
4. Tried with TS-950SDX settings "Transfr" (allows data transfer between two TS-950SDX) and "Direct" (whether signals are directly written to VFO by Transfr function or not) in all variants (all off, all on, only Transfr, only Direct).

I don't know if the behaviour is caused by the radio or `hamlib` library. The RS232 connection is most likely set up correctly (baud rate, flow control, stop bits, parity, rtx/cts/rx/tx signals are inverted by the FT232RL chip, rts/cts are crossed and so are tx/rx) because other commands work as expected: PTT, A=B, A<=>B, s-meter readout, ...

Does `kenwood_transaction: Unknown command or rig busy 'PS'` relate to the described problem?

# Example

```
rigctl --model=2013 --rig-file=/dev/ttyUSB0 -vvvvv
rigctl Hamlib 4.0
Last commit was Fri Dec 25 02:30:30 2020 +0000 SHA=519f82
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
initrigs4_kenwood called
rig_register called
rig_register: rig_register (2012)
rig_register called
rig_register: rig_register (2013)
rig_register called
rig_register: rig_register (2001)
rig_register called
rig_register: rig_register (2025)
rig_register called
rig_register: rig_register (2003)
rig_register called
rig_register: rig_register (2004)
rig_register called
rig_register: rig_register (2016)
rig_register called
rig_register: rig_register (2024)
rig_register called
rig_register: rig_register (2005)
rig_register called
rig_register: rig_register (2007)
rig_register called
rig_register: rig_register (2009)
rig_register called
rig_register: rig_register (2010)
rig_register called
rig_register: rig_register (2022)
rig_register called
rig_register: rig_register (2014)
[... 228 lines cut ...]
Rig command: W FB; 14
rigctl_parse: input_line: W FB; 14
rigctl_parse: vfo_opt=0
rigctl_send_cmd: called
rigctl_send_cmd: backend_num=2
rigctl_send_cmd: KENWOOD
rigctl_send_cmd: arg1=FB;
rigctl_send_cmd: bufcmd=FB;
rigctl_send_cmd: text protocol, send_cmd_term=0x00 0x0d 0x00
rig_flush: called for serial device
serial_flush called
tcflush
rigctl_send_cmd: rigport=3, bufcmd=FB;, cmd_len=3
write_block called
write_block(): TX 3 bytes
0000    46 42 3b                                            FB;             
rigctl_send_cmd: debug Cmd
read_string called, rxmax=15
read_string(): RX 14 characters
0000    46 42 30 30 30 31 30 31 33 36 30 30 30 3b           FB00010136000;  
Reply: FB00010136000;
rigctl_parse: vfo_opt=0
rigctl_parse: retcode=0
rigctl_parse: called, interactive=1
```

mdblack98, 2022-03-27:
Also the PS command won't bother anything.But I have made a change to disable the PS command for the TS-950's since the rig doesn't have that command.
You can download a new DLL later tonight that will be dated 20220328 with that change from here:http://n0nb.users.sourceforge.net/

Mike W9MDB

 
 

   
   - send IF command: W IF; 38      

   - send FA command: W FA; 14      

   - send FB command: W FB; 14

mdblack98, 2022-03-27:
Or rather I see you are running Linux  presumably to you can just clone the github repository and build hamlib yourself.
Remove the hamlib package you have and then do this:
git clone https://github.com/Hamlib/Hamlib.gitcd Hamlib./bootstrap./configuremake install
Mike W9DMB

rubienr, 2022-03-28:
Tested with `grig` compiled form source as of fceac511eadbde504d0e7f24d3b1d262c2bcf09c with same behaviour. However, by fiddling around I found that the radio was set up wrong. The button "QUICK MEMO MR" (recalls the quick memory) had to be pressed once, then setting/reading frequency to/from radio works as expected. Finally removed the installation from git source, re-installed flrig/wsjtx from the official 21.10 Ubuntu repository and it worked as expected. Sorry for the noise. 

For Kenwood TS-950SDX users: the button is described in the instruction manual on page 15 by key number 64 named "QUICK MEMO MR key".

Thank you @mdblack98 for the support! Problem is solved.

closed 2022-03-28 by rubienr

mdblack98, 2022-10-11:
Are you sure you aren't in memory channel mode?  Do you have M-CH lit up?
The idea that IF and FA/FB don't match indicates some problem with your rig.
If you are not in M-CH mode than I'd recommend doing a microprocessor reset.  
Mike W9MDB