Hamlib/Hamlib #1009: Xiegu X6100 via TCP

Links2004, 2022-04-14:
Hi,

is it possible to get a TCP option for the Xiegu X6100?
The X6100 runs linux and I am currently working on a way to control the X6100 complacently via IP / WiFi.
the protocol is the same as serial since `socat` is used on the X6100 to redirect from the physical Serial to a TCP server.

`socat pty,link=/dev/virtualcom0,raw tcp-listen:9999`

in cases it helps here is the link to the OS replacement for the X6100
https://github.com/Links2004/x6100-armbian

mdblack98, 2022-04-14:
TCP can be used on any rig.
Just use the io address with the port like this.
E.g. 127.0.0.1:4321

Mike w9mdb

Links2004, 2022-04-15:
I most likely doing something wrong then.

```
$ rigctl --model=3087 --rig-file=10.22.11.17:9990 -vv m
read_string_generic(): read failed, direct=1 - Connection refused
read_string_generic(): read failed, direct=1 - Connection refused
read_string_generic(): read failed, direct=1 - Connection refused
rig_get_cache: unknown vfo=otherVFO
....
```

vs

```
$ sudo socat pty,link=/dev/ttyX6100Cat,raw,user=${USER} tcp:10.22.11.17:9990 &
$ rigctl --model=3087 --rig-file=/dev/ttyX6100Cat -v m
Opened rig model 3087, 'X6100'

0
```

mdblack98, 2022-04-15:
Try not sending the "m" command -- I suspect the rig may need a pause before sending commands....if so I can make that change.

What version of hamlib are you running?  Would like you to run the latest -- http://n0nb.users.sourceforge.net/rigctl --version
So see if you can just run rigctl normally first and use the m and f commands for example.
rigctl --model=3087 --rig-file=10.22.11.17:9990 -vvvRig command: mRig command: fThen send me the log.txt from this:rigctl --model=3087 --rig-file=10.22.11.17:9990 -vvvvv m >log.txt 2>&1Mike W9MDB

Links2004, 2022-04-15:
hamlib where build from git on latest commit.
```
rigctl --version
rigctl Hamlib 4.5~git Fri Apr 08 11:51:39 2022 +0000 SHA=98c225
```

the `Icom rig UDP network enabled` in the log is a good hint since we are talking about TCP here not UDP.

[log.txt](https://github.com/Hamlib/Hamlib/files/8496028/log.txt)
[no_command.txt](https://github.com/Hamlib/Hamlib/files/8496029/no_command.txt)

Links2004, 2022-04-16:
confirmed working
```
$ rigctl --version
rigctl Hamlib 4.5~git Sa Apr 16 13:18:16 2022 +0000 SHA=e409ef
$ rigctl --model=3087 --rig-file=10.22.11.17:9090 f
7050000
$ rigctl --model=3087 --rig-file=10.22.11.17:9090 m
USB
1800
```

closed 2022-04-16 by mdblack98