Hamlib/Hamlib #1013: Rig model name changes

mdblack98, 2022-04-21:
1004  Yaesu                  MARK-V FT-1000MP        20211014.0      Stable      RIG_MODEL_FT1000MPMKV
  1024  Yaesu                  FT-1000MP               20211113.0      Stable      RIG_MODEL_FT1000MP
  1025  Yaesu                  MARK-V Field FT-1000MP  20211014.0      Stable      RIG_MODEL_FT1000MPMKVFLD

Change models to
1004 FT-1000MP MARK-V
1024 FT-1000MP MARK-V Field

Review other models for similar inconsistences.

closed 2025-08-01 by N0NB

N0NB, 2025-09-19:
Trouble is that the WSJT-X maintainer wants it changed as setting the manufacturer name to W1HKJ put it at the bottom of the list.

arekm, 2025-09-19:
Nice addition to wsjt-x would be adding simple search bar to that setting. That would make order not that important.

Manually reviewing that list was always painful anyway.

arekm, 2025-09-20:
Not that people change radios but change software like use wsjtx with stable 4.6 hamlib then try newer wsjtx built with hamlib 4.7 devel version and so on (rig name changes in hamlib -> wsjtx silently switches to ADAT rig).

Primary blame is on WJST-X for silently ignoring its own config (and a bit on hamlib for not documenting what's stable (ids) and what's not (names)) but well...

N0NB, 2025-09-21:
From my side, the model number is stable.  If the list were keyed by the model number FlRig would have remained the fourth one on the list from the top.  ADAT being at the top is next to useless, IMO.

There is no one good solution.  Leave it to others to do what is unexpected!

The actual order of the model numbers in a backend group have not changed nor have the group numbers changed since Hamlib's inception.  What did change a few years back was the number a possible models in a backend from 100 to 1000.  This did change the apparent model number.  The Yaesu FT-847 went from model number 101 to 1001 and so on.  The Yaesu backen was still group 1 and the '847 was still the first model in that backend.

Hopefully, even though the Icom backend currently has 92 models, I hope that 1000 is plenty (actually 999 as we start counting at one and I doubt we would change that order).