Hamlib/Hamlib #1016: Hypothetical Hamlib Antenna Switch Command Format?

gdjacobs, 2022-04-21:
I'm extending my Node Red flow for remote antenna switching and would like to implement a textual interface to control it (in addition to the Dashboard interface). Ideally, this interface could be exposed via serial as well. With the Hamlib community's indulgence, I'd like to flesh out some ideas concerning what should be included.

gdjacobs, 2022-04-21:
I think the most primitive layer should cover exclusive direct linking of input ports to output ports. In the most basic form, the switch will only have one input port and only allow one output port to be enabled at a time (although N to M matrix switches routing multiple rigs should be included. I also think this layer should support output ports being selectively floated or grounded while inactive, subject to hardware implementation, of course.

gdjacobs, 2022-04-21:
The interface should probably also include a "smart" layer for band mapping and automatic switching parameters, but I don't have any firm thoughts on this.

gdjacobs, 2022-04-21:
Just some initial thoughts. Any thoughts and nput from the community would be appreciated. If anyone feels there's another place more appropriate to take this discussion, please let me know.

Thanks and 73. VA4GDJ

mdblack98, 2022-04-21:
And what antenna switch would this be?  One that has an interface with commands?
Mike W9MDB

gdjacobs, 2022-04-21:
At the moment, the controller is implemented as a Node Red flow, but I want to provide an API and also build a uC version with a serial interface. Most switch automation is highly proprietary (Shacklan, for instance). I could put together an ad hoc command set which would work for my application, but that doesn't seem in keeping with the ideals of an open, extensible interface.