Hamlib/Hamlib #1017: rigctld: symbol lookup error: rigctld: undefined symbol: debugmsgsave

worksmarter, 2022-04-26:
Linux raspberrypi 5.15.34-v7+ build of hamlib 4.4 seems to go OK but when I use rigctld to list the rigs, I get the following error:

rigctld: symbol lookup error: rigctld: undefined symbol: debugmsgsave

This is a fresh OS install with just gpsd and rtc enabled on it.

mdblack98, 2022-04-26:
Make sure you don't have another libhamlib running around.Remove any hamlib packages.
Could be in /usr/lib and /usr/local/lib
By default a build of hamlib will put libhamlib in /usr/local/lib
Mike W9MDB

worksmarter, 2022-04-27:
This is a fresh install of Linux raspberrypi 5.15.34-v7+ With Rasberry Pi's imager. (see Post #1) I have not tried to install hamlib prior to installing it for the first time. I think this is a Bullseye issue as It's not installed twice on my setup for I even double-checked when I found the thread hear suggesting there were two versions of hamlib installed on someone's system. I also found others that are experiencing the same problem/behavior...

mdblack98, 2022-04-27:
Two things....
See what libhamlib is being referenced.ldd rigctl
And then try compiling static.
configure -static
make clean
make install
Mike W9MDB

worksmarter, 2022-04-27:
I am sorry, I am a neophyte with hamlib, so how do I go about answering this question:

 

See what libhamlib is being referenced.ldd rigctl

 

Here is my build configuration:

 

wget https://sourceforge.net/projects/hamlib/files/hamlib/4.2/hamlib-4.4.tar.gz

tar -zxvf hamlib-4.4.tar.gz 

cd hamlib-4.4 

./configure --prefix=/usr/local --enable-static 

make 

sudo make install 

sudo ldconfig 

reboot

 

Thank you in advance for any clarifications you may provide.

 

Kind regards,

Douglas

 

This email, including any attachments, is meant only for the intended 
recipient and may be a confidential communication or a communication 
privileged by law.

mdblack98, 2022-04-27:
If you built with -static there should be no libhamlib in the output of this
ldd rigctld
Are you still seeing the undefined symbol?
Mike W9MDB

worksmarter, 2022-04-27:
Output of 
$ ldd rigctld
ldd: ./rigctld: No such file or directory

Output of:
rigctl -l
rigctl: symbol lookup error: rigctl: undefined symbol: debugmsgsave

mdblack98, 2022-04-27:
Hmmm....ldd is behaving weird....
ldd /usr/local/bin/rigctldldd /usr/local/bin/rigctl
This:which rigctlShould point to /usr/local/bin/rigctl
Remove all libhamlib with this:find / -name "libhamlib*" -exec rm {} \; -print
And make install again.
Also, try the latest hamlib from herehttp://n0nb.users.sourceforge.net/

Mike

N0NB, 2022-04-27:
By way of example, here is the output from my desktop:

```
$ ldd $(which rigctld)
	linux-vdso.so.1 (0x00007fff7812d000)
	libhamlib.so.4 => /home/nate/local/lib/libhamlib.so.4 (0x00007f90aeed0000)
	libreadline.so.8 => /lib/x86_64-linux-gnu/libreadline.so.8 (0x00007f90aee59000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f90aee37000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f90aec72000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f90aec6c000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f90aeb28000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f90aeb08000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007f90aead9000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f90af948000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f90aeab1000)
```

worksmarter, 2022-04-27:
```
$ which rigctld
/usr/local/bin/rigctld

$ ldd $(which rigctld)
linux-vdso.so.1 (0x7eb9d000)
        /usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76f2d000)
        libhamlib.so.4 => /lib/arm-linux-gnueabihf/libhamlib.so.4 (0x76772000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76746000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x765f2000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x765de000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x7656f000)
        libusb-1.0.so.0 => /lib/arm-linux-gnueabihf/libusb-1.0.so.0 (0x76547000)
        /lib/ld-linux-armhf.so.3 (0x76f42000)
        libudev.so.1 => /lib/arm-linux-gnueabihf/libudev.so.1 (0x76513000)
```
So the same results as before and I was using hamlib4.5 because 4.4 did this so I used a backup of my setup from before hamlib install and freshly installed hamlib4.5.
It is still exhibiting the same behavior now on attempt #2.

mdblack98, 2022-04-27:
First take a look at the date on this item/lib/arm-linux-gnueabihf/libhamlib.so.4You should find another one that will be more recent/usr/local/lib/libhamlib.so.4
Then remove rm /lib/arm-linux-gnueabihf/libhamlib.so.4That was installed by some package.

Mike W9MDB

worksmarter, 2022-04-27:
I have all of the items in the /usr/local/lib listed as 4.0.5

```
/usr/local/lib $ ls -l
total 47568
-rw-r--r-- 1 root root   378082 Apr 27 11:57 libhamlib++.a
-rw-r--r-- 1 root root 27975684 Apr 27 11:57 libhamlib.a
-rwxr-xr-x 1 root root     1011 Apr 27 11:57 libhamlib++.la
-rwxr-xr-x 1 root root      973 Apr 27 11:57 libhamlib.la
lrwxrwxrwx 1 root root       20 Apr 27 11:57 libhamlib++.so -> libhamlib++.so.4.0.5
lrwxrwxrwx 1 root root       18 Apr 27 11:57 libhamlib.so -> libhamlib.so.4.0.5
lrwxrwxrwx 1 root root       20 Apr 27 11:57 libhamlib++.so.4 -> libhamlib++.so.4.0.5
lrwxrwxrwx 1 root root       18 Apr 27 11:57 libhamlib.so.4 -> libhamlib.so.4.0.5
-rwxr-xr-x 1 root root   247588 Apr 27 11:57 libhamlib++.so.4.0.5
-rwxr-xr-x 1 root root 20077904 Apr 27 11:57 libhamlib.so.4.0.5
drwxr-xr-x 2 root root     4096 Apr 27 11:57 pkgconfig
drwxr-xr-x 3 root root     4096 Apr  4 07:56 pypy2.7
drwxr-xr-x 3 root root     4096 Apr  4 07:49 python3.9

```
I'm not certain I know how to do what you have requested So I poked around a little bit and it would seem that the symlink is pointing to an older file than 4.0.5 in /lib/arm-linux-gnueabihf

```
aprs@raspberrypi:/lib/arm-linux-gnueabihf $ find libhamlib.so.4
libhamlib.so.4
aprs@raspberrypi:/lib/arm-linux-gnueabihf $ stat libhamlib.so.4
  File: libhamlib.so.4 -> libhamlib.so.4.0.0
  Size: 18              Blocks: 0          IO Block: 4096   symbolic link
Device: b302h/45826d    Inode: 16693       Links: 1
Access: (0777/lrwxrwxrwx)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2022-04-26 14:38:26.000000000 -0400
Modify: 2021-07-16 00:31:14.000000000 -0400
Change: 2022-04-26 14:38:33.194915040 -0400
 Birth: 2022-04-26 14:38:32.835095040 -0400
aprs@raspberrypi:/lib/arm-linux-gnueabihf $ find libhamlib.so.4.0.0
libhamlib.so.4.0.0
aprs@raspberrypi:/lib/arm-linux-gnueabihf $ stat libhamlib.so.4.0.0
  File: libhamlib.so.4.0.0
  Size: 7690428         Blocks: 15024      IO Block: 4096   regular file
Device: b302h/45826d    Inode: 16692       Links: 1
Access: (0644/-rw-r--r--)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2022-04-26 14:38:26.000000000 -0400
Modify: 2021-07-16 00:31:14.000000000 -0400
Change: 2022-04-26 14:38:33.194915040 -0400
 Birth: 2022-04-26 14:38:32.585220041 -0400
aprs@raspberrypi:/lib/arm-linux-gnueabihf $ cd /usr/local/lib/
aprs@raspberrypi:/usr/local/lib $ stat libhamlib.so.4
  File: libhamlib.so.4 -> libhamlib.so.4.0.5
  Size: 18              Blocks: 0          IO Block: 4096   symbolic link
Device: b302h/45826d    Inode: 158901      Links: 1
Access: (0777/lrwxrwxrwx)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2022-04-27 11:57:21.621071549 -0400
Modify: 2022-04-27 11:57:21.621071549 -0400
Change: 2022-04-27 11:57:21.621071549 -0400
 Birth: 2022-04-27 11:57:21.621071549 -0400
```
I don't know how to proceed, There is an issue with the older file and a symlink, but alas I am not well versed with symlinking things.

mdblack98, 2022-04-27:
rm  /lib/arm-linux-gnueabihf/libhamlib*

 
 

libhamlib.so.4
libhamlib.so.4.0.0

worksmarter, 2022-04-27:
This worked:
`sudo rm  /lib/arm-linux-gnueabihf/libhamlib*`
Then cd back to my source 
`cd /home/aprs/Downloads/hamlib-4.5~git`
`./configure --prefix=/usr/local --enable-static`
`make clean`
`make`
`sudo make install`
`sudo ldconfig`
`reboot`

Thanks for the help. I wish I knew what installed the libhamlib file that was present, but all seems to be working now.

closed 2022-04-29 by mdblack98