Hamlib/Hamlib #1018: Enhancement: conditional builds for radio classes

guitarpicva, 2022-04-29:
Concept is:

./configure --with-yaesu --with-icom

Which would build the main parts required by all and only radio types in the yaesu and icom radio classes.

Hamlib takes an eternity to build (particularly on an SOC) and most users have one or two brands of radios and don't need all the rest.

Also, consider --with-rotator, --with-antenna for those who need them, and then the rest of us can avoid the 'noid.

I don't expect any feedback.  I figure if you find the idea worth doing, you'll just do it. ;)

pvrose, 2025-07-31:
Who is the user who would use this? I would think that the main users of the compile flow are going to build hamlib into their generic application, and therefore want all backends.

The real end-user with one or two rigs will then use the generic application which will have hamlib already built-in for them.

guitarpicva, 2025-07-31:
Using cmake, there could be CMakeLists.txt made up for major groups.  

ex: mkdir build && cd build && cp ../CMakeListsYaesu.txt ../CMakeLists.txt && cmake ../

No command line issues.

cmake is another "undertaking" the team may not wish to convert to.  that's a value judgement as well.

N0NB, 2025-07-31:
cmake has come up on the mailing list in the past.  Long ago before the library was this sophisticated I looked into it and redoubled my efforts on understanding the Autotools better.  A few years back a proposal floated through to convert to cmake.  My strong preferences as noted then have not changed, all current environments need to be built and the capability to distribute a source tarball as we do now that does not require the user to install cmake.  So far I've not seen any reference to any work that has been done by the proponents.

I've built cmake projects and I've nothing against it.  I'm not sure that cmake can generate a source archive like the Autotools can where all that is required is a shell to run `configure`, `make`, and any optional build dependencies.  It may be possible, but so far whenever I've encountered a cmake project I've had to have cmake installed.

If someone is motivated to do so, when 5.0 development begins after the 4.7 branch is created, hopefully in a couple of months, would be a prime time to do it.  It's going to be a lot, and I mean a lot of work with maybe a 50/50 chance of acceptance (not just me but other contributors).  I'm certainly willing to evaluate it but I don't have the last word.