Hamlib/Hamlib #1027: RigCtl fails to return valid result when setting FT-991 into C4FM mode

rpoulin, 2022-05-13:
This issue impacts my logging software (Log4OM2) that disconnects my CAT interface due to a timeout or error code returned  from HamLib. I downloaded and tested RigCtl from the command line on Windows with the following observations:

-------------------------------------------
PS C:\Program Files (x86)\hamlib-w64-4.4\bin> .\rigctl.exe -m 1035 -r COM4 -C stop_bits=2 -v -Z -o
Opened rig model 1035, 'FT-991'

Rig command: \get_vfo_info VFOA
Freq: 145690000
Mode: FM
Width: 16000
Split: 0
SatMode: 0

<<---------  Set FT-991 to C4FM mode here ---------->>

Rig command: \get_vfo_info VFOA
Freq: 145690000
Mode: None
Width: 0
Split: -425705312
SatMode: 0
get_vfo_info: error = Protocol error
rig.c(6961):rig_get_vfo_info return(-8)
rigctl_get_vfo_info: vfo=VFOA
rigctl_parse.c(2304):rigctl_get_vfo_info return(-8)

Rig command:
----------------------------------------------

It appears that the C4FM mode may not be implemented.  The man pages indicate that it is not implemented but that "AMS"  is a valid mode token, C4FM is missing from the list of valid modes. In the real world, AMS is an "Automatic Mode Select" option for the FT-991 when C4FM is the selected mode.  C4FM is the digital mode supported by the radio not AMS.

vk4tux, 2022-05-13:
FT-991 Cat manual shows C4FM as a mode choice ;

which means it is doable in rigctld if not already.

73

Adrian Fewster

mdblack98, 2022-05-13:
Can you please run that with 5 v's -- I need to see what the rig is returning.
-vvvvv
Mike W9MDB

closed 2022-05-13 by mdblack98

rpoulin, 2022-05-14:
Mike,

I am sending you a text file containing the trace you requested.  I set the rig into C4FM mode before I ran the commands.  Hope this is what you needed.

Regards,
[trace.txt](https://github.com/Hamlib/Hamlib/files/8691901/trace.txt)

Ron (NG7V)