Hamlib/Hamlib #1028: RigCtl fails to return valid result when requesting VFOB status with FT-991 is put into split operation

rpoulin, 2022-05-13:
This error impacts my logging software (Log4OM) by not returning VFOB frequency when requested even if the FT-991 rigCtl implementation indicates dual VFO capability.

Ran RigCtl using the -o command line to indicate support for VFOA and VFOB. Used the SPLIT feature in the radio to see if I could get a read of VFOB's settings.  This is the result:

PS C:\Program Files (x86)\hamlib-w64-4.4\bin> .\rigctl.exe -m 1035 -r COM4 -C stop_bits=2 -v -Z -o
Opened rig model 1035, 'FT-991'

<<------- put radio split mode off --------->>

Rig command: \get_vfo_info VFOA
Freq: 14027000
Mode: CW
Width: 300
Split: 0
SatMode: 0

<<------- put radio split mod on --------->>

Rig command: \get_vfo_info VFOB
Freq: 7055000
Mode: None
Width: 0
Split: -425705312
SatMode: 0
get_vfo_info: error = Command rejected by the rig
rig.c(6961):rig_get_vfo_info return(-9)
rigctl_get_vfo_info: vfo=VFOB
rigctl_parse.c(2304):rigctl_get_vfo_info return(-9)

<<------- put radio split mode off here -------->>

Rig command: \get_vfo_info VFOA
Freq: 14027000
Mode: CW
Width: 300
Split: 0
SatMode: 0

<<------- put readio split mode on here --------->>

Rig command: \get_vfo_info VFOA
Freq: 14027000
Mode: CW
Width: 300
Split: 1
SatMode: 0

mdblack98, 2022-05-13:
Please download the latest hamlib package from here:http://n0nb.users.sourceforge.net/

The replace the DLL with this one:https://www.dropbox.com/s/7aejdv74nfjxr0q/libhamlib-4.dll?dl=0

This patch will be in tonight's build dated 20220514

Mike W9MDB

mdblack98, 2022-05-13:
This should be fixed now

closed 2022-05-13 by mdblack98

rpoulin, 2022-05-14:
Mike,

Thanks for getting back to me.  I installed the new set of files and the dll you instructed me to download. I received an error from rigctl. See the graphic. The error was from rigctl indicating that hamlib.dll seemed not designed to work on windows or there is an error in the dll.

[image]

rpoulin, 2022-05-14:
Downloaded the nightly build and tested.  Fix works, thank you.

mdblack98, 2022-10-11:
Take the contents on this and replace all the files in the hamlib directory (don't include the "bin" directory...just the files inside it).
https://www.dropbox.com/s/tgeouohpw8x7ezt/bin.zip?dl=0

Mike