Hamlib/Hamlib #1029: testsecurity.exe on Windows

gvanem, 2022-05-14:
This program on Windows, gives this strange output:
```
key1=aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
key2=aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
4145530000166d5c65a320e1ed970da732ea680b10f2d8da7e032eb2adae97fe87998a24902b9a722be0978ca20df01d9af7b6e2b6112dcb2f480411a24279d139
ae3c6800bfcdfdd998586bb2cf9b59e3704e5523
Ciphertext length: 85
Decrypting...
Decrypted plaintext length: 32, aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
```

The values of `key1` and `key2` are not random at all due to the sloppy `gettimeofday()` in `lib/gettimeofday.c`.
But even on MinGW the output is strange. So I've replaced with this improved function
(needs Win-8+):
```c
/*
 * Number of micro-seconds between the beginning of the Windows epoch
 * (Jan. 1, 1601) and the Unix epoch (Jan. 1, 1970).
 */
#define DELTA_EPOCH_IN_USEC  11644473600000000Ui64

static uint64_t FILETIME_to_unix_epoch (const FILETIME *ft)
{
  uint64_t res = (uint64_t) ft->dwHighDateTime << 32;

  res |= ft->dwLowDateTime;
  res /= 10;                   /* from 100 nano-sec periods to usec */
  res -= DELTA_EPOCH_IN_USEC;  /* from Win epoch to Unix epoch */
  return (res);
}

int gettimeofday (struct timeval *tv, struct timezone *tz)
{
  FILETIME ft;
  uint64_t tim;

  if (!tv)
      return (EINVAL);

  GetSystemTimePreciseAsFileTime (&ft);
  tim = FILETIME_to_unix_epoch (&ft);
  tv->tv_sec  = (long) (tim / 1000000L);
  tv->tv_usec = (long) (tim % 1000000L);
  (void) tz;
  return (0);
}
```

which gives these keys instead (with 100 usec sleep for each char) :
``` 
key1=koppppqqqrrrrssssttttuuuuvvvvwww
key2=xxxxyyyyzzzzAAAABBBBCCCCDDDEEEEF
```

Not so random. But with with 1 msec sleep for each char, it seems much more random:
```
key1=Ysid~=ZFV!W7/s#j#H90)#^1HaTb1c9Q
key2=/@efo@Y=_ytl=E$pa2W6oxwrdD>2q~yt
```

N0NB, 2025-07-31:
@gvanem Apologies for the long delayed response.

Is this something you'd be able to rebase against the current master branch and create a pull request?  Also, can you incorporate a run time test for Windows 8+?

gvanem, 2025-08-01:
@N0NB Since Hamlib is so damn hard to tweak for MSVC/clang-cl, I think not. I'm tired of it.

closed 2025-08-01 by N0NB