Hamlib/Hamlib #1030: Node-red Hamlib not working well due to line-feed perhaps?

mdblack98, 2022-05-15:
The node-red tcprequest appears to only take 1 line of response and gets confused between msg packets when more than one line from rigctld is encountered.  Perhaps a tab char as a field separator from rigctld for that situation may improve this behavior.

vk4tux, 2022-05-15:
What is the expected rate of msg requests like 'G UP' via tcp request to 
rigctld etc?

Here it seems if the rate is more than one per 300ms , the requests que up.

closed 2022-05-21 by mdblack98

mdblack98, 2022-10-11:
Band changes can take a while -- very rig dependent.  Flex is like 500ms.  Many rigs > 200ms.There is no "expected" rate.  The rig does what it's capable of.
What you are doing with node-red is really pushing the limit of what your rig is capable of.  I see some queuing on my limited test set right now with 1-per-second queries.
I'm working on a solution for Hamlib where just two tcprequest nodes will be needed (one for query and one for setting) that never disconnect.The stop/start of the tcp port is really a waste of time.
Mike W9MDB

mdblack98, 2022-10-11:
Boy this post took a LONG time to show up.  2022-10-11 right now and it showed up in my inbox.   Anyways...I abandoned trying to fix this in node red -- was getting frustrated with node red's abilities and I had other things I have to work on.
What is needed is a dedicated Hamlib node that stays connected as there is lots of initialization that goes on during startup.