Hamlib/Hamlib #1038: add2debugmsgsave() deadlock

mikaelnousiainen, 2022-05-26:
Noticed another issue yesterday evening when testing the latest master on IC-7300 with busy polling and spectrum enabled: the recent changes on `add2debugmsgsave()` somehow lead to a deadlock. It looks like it may not be the mutex as the line numbers point to the `while` loop. Anyway, this is reproducible easily with busy control traffic and full debug logging enabled.

The lines indicated by the stack traces are:

`rig.c` line 345 is:

https://github.com/Hamlib/Hamlib/blob/8cc4d59cbb06871838c0c41e2491de9f85449a82/src/rig.c#L345

`rig.c` line 347 is:

https://github.com/Hamlib/Hamlib/blob/8cc4d59cbb06871838c0c41e2491de9f85449a82/src/rig.c#L347

So it looks like the loop never exits in some case.

I didn't yet inspect the code much, but could it be multiple simultaneous `add2debugmsgsave()` somehow interfering with each other?

Thread 1:

```
read_string_generic: minlen now 22
Connection opened from 127.0.0.1:43360 <- DEADLOCK HERE
^C

Thread 1 "rigctld" received signal SIGINT, Interrupt.
strchr () at ../sysdeps/arm/armv6/strchr.S:96
96      ../sysdeps/arm/armv6/strchr.S: No such file or directory.
(gdb) bt
#0  strchr () at ../sysdeps/arm/armv6/strchr.S:96
#1  0xb66e64d8 in add2debugmsgsave (s=<optimized out>) at rig.c:347
#2  0x00014928 in main (argc=<optimized out>, argv=<optimized out>) at rigctld.c:1054

(gdb) info threads
  Id   Target Id                               Frame 
* 1    Thread 0xb6ff8040 (LWP 17897) "rigctld" strchr () at ../sysdeps/arm/armv6/strchr.S:96
  2    Thread 0xb63c8440 (LWP 17900) "rigctld" strchr () at ../sysdeps/arm/armv6/strchr.S:96
  3    Thread 0xb5bc7440 (LWP 17901) "rigctld" strchr () at ../sysdeps/arm/armv6/strchr.S:96
  4    Thread 0xb51ff440 (LWP 17926) "rigctld" 0xb6fba214 in strlen () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so
```

Thread 2:

```
(gdb) t 2
[Switching to thread 2 (Thread 0xb63c8440 (LWP 17900))]
#0  strchr () at ../sysdeps/arm/armv6/strchr.S:96
96      in ../sysdeps/arm/armv6/strchr.S
(gdb) bt
#0  strchr () at ../sysdeps/arm/armv6/strchr.S:96
#1  0xb66e64d8 in add2debugmsgsave (s=s@entry=0x46360 <debugmsgsave2> "Connection opened from 127.0.0.1:43360\n") at rig.c:347
#2  0xb671d5ec in read_string_generic (p=p@entry=0x500bc, rxbuffer=0x16 <error: Cannot access memory at address 0x16>, rxmax=22, stopset=0xb63c7a18 "\376\376\340\224'", 
    stopset@entry=0xb68da50c <icom_block_end> "\375", <incomplete sequence \374>, stopset_len=2, stopset_len@entry=0, flush_flag=0, flush_flag@entry=-1233422692, expected_len=1, expected_len@entry=2, direct=direct@entry=1)
    at iofunc.c:1384
#3  0xb671e1e8 in read_string_direct (p=p@entry=0x500bc, rxbuffer=<optimized out>, rxmax=<optimized out>, stopset=stopset@entry=0xb68da50c <icom_block_end> "\375", <incomplete sequence \374>, stopset_len=stopset_len@entry=2, 
    flush_flag=flush_flag@entry=0, expected_len=expected_len@entry=1) at iofunc.c:1483
#4  0xb67b769c in read_icom_frame_generic (p=0x500bc, rxbuffer=0xb63c7a18 "\376\376\340\224'", rxbuffer_len=0, direct=1) at frame.c:490
#5  0xb66e675c in async_data_handler (arg=<optimized out>) at rig.c:7178
#6  0xb6635300 in start_thread (arg=0xb63c8440) at pthread_create.c:477
#7  0xb65b9208 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

Thread 3:

```
(gdb) t 3
[Switching to thread 3 (Thread 0xb5bc7440 (LWP 17901))]
#0  strchr () at ../sysdeps/arm/armv6/strchr.S:96
96      in ../sysdeps/arm/armv6/strchr.S
(gdb) bt
#0  strchr () at ../sysdeps/arm/armv6/strchr.S:96
#1  0xb66e64d8 in add2debugmsgsave (s=s@entry=0x46360 <debugmsgsave2> "Connection opened from 127.0.0.1:43360\n") at rig.c:347
#2  0xb67240a4 in multicast_publisher (arg=0x517a4) at network.c:935
#3  0xb6635300 in start_thread (arg=0xb5bc7440) at pthread_create.c:477
#4  0xb65b9208 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

Thread 4:

```
(gdb) t 4
[Switching to thread 4 (Thread 0xb51ff440 (LWP 17926))]
#0  0xb6fba214 in strlen () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so
(gdb) bt
#0  0xb6fba214 in strlen () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so
#1  0xb66e6520 in add2debugmsgsave (s=s@entry=0x46360 <debugmsgsave2> "Connection opened from 127.0.0.1:43360\n") at rig.c:345
#2  0xb67b7b94 in icom_one_transaction (rig=rig@entry=0x4d680, cmd=cmd@entry=21 '\025', subcmd=subcmd@entry=2, payload=payload@entry=0xb51fd510 "", payload_len=0, payload_len@entry=-1232236336, data=0xb51fd5d8 "\025\002",
    data@entry=0xb67b1078 <icom_get_level+808> "", data_len=0xb51fd50c, data_len@entry=0x0) at frame.c:403
#3  0xb67b8cbc in icom_transaction (rig=0x4d680, rig@entry=0x0, cmd=cmd@entry=21, subcmd=2, payload=0xb51fd510 "", payload@entry=0xb51fd508 "", payload_len=payload_len@entry=0, data=data@entry=0xb51fd5d8 "\025\002",
    data_len=0xb51fd50c, data_len@entry=0xb51fd504) at frame.c:434
#4  0xb67b1078 in icom_get_level (rig=0x0, vfo=3066462504, level=<optimized out>, val=0xb51fd6f8) at icom.c:4020
#5  0x00024a10 in rigctl_get_level (rig=rig@entry=0x4d680, fout=fout@entry=0xb4800738, fin=fin@entry=0xb48005f8, interactive=0, interactive@entry=1, prompt=prompt@entry=0, vfo_opt=vfo_opt@entry=0x51a1c,
    send_cmd_term=send_cmd_term@entry=13 '\r', ext_resp=1, resp_sep=10 '\n', cmd=cmd@entry=0x457bc <test_list+600>, vfo=vfo@entry=536870912, arg1=arg1@entry=0xb51fdb80 "STRENGTH", arg2=arg2@entry=0x31478 "", arg3=arg3@entry=0x31478 "")
    at rigctl_parse.c:3301
#6  0x00029b8c in rigctl_parse (my_rig=<optimized out>, fin=fin@entry=0xb48005f8, fout=fout@entry=0xb4800738, argv=argv@entry=0x0, argc=<optimized out>, argc@entry=0, sync_cb=<optimized out>, sync_cb@entry=0x14ff8 <mutex_rigctld>,
    interactive=<optimized out>, interactive@entry=1, prompt=<optimized out>, prompt@entry=0, vfo_opt=<optimized out>, vfo_opt@entry=0x51a1c, send_cmd_term=<optimized out>, send_cmd_term@entry=13 '\r', ext_resp_ptr=<optimized out>,
    ext_resp_ptr@entry=0xb51fe9e8, resp_sep_ptr=<optimized out>, resp_sep_ptr@entry=0x453a0 <resp_sep> "\n", use_password=<optimized out>) at rigctl_parse.c:1721
#7  0x000153a0 in handle_socket (arg=<optimized out>) at rigctld.c:1224
#8  0xb6635300 in start_thread (arg=0xb51ff440) at pthread_create.c:477
#9  0xb65b9208 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

mdblack98, 2022-05-26:
Try again...I think it was getting stuck on a long debug msg....should clear it out if it sees one.
Mike W9MDB

 
 

https://github.com/Hamlib/Hamlib/blob/8cc4d59cbb06871838c0c41e2491de9f85449a82/src/rig.c#L345

https://github.com/Hamlib/Hamlib/blob/8cc4d59cbb06871838c0c41e2491de9f85449a82/src/rig.c#L347

#1  0xb66e64d8 in add2debugmsgsave ***@***.***=0x46360 <debugmsgsave2> "Connection opened from 127.0.0.1:43360\n") at rig.c:347
#2  0xb671d5ec in read_string_generic ***@***.***=0x500bc, rxbuffer=0x16 <error: Cannot access memory at address 0x16>, rxmax=22, stopset=0xb63c7a18 "\376\376\340\224'", 
    at iofunc.c:1384
#3  0xb671e1e8 in read_string_direct ***@***.***=0x500bc, rxbuffer=<optimized out>, rxmax=<optimized out>, ***@***.***=0xb68da50c <icom_block_end> "\375", <incomplete sequence \374>, ***@***.***=2, 

#1  0xb66e64d8 in add2debugmsgsave ***@***.***=0x46360 <debugmsgsave2> "Connection opened from 127.0.0.1:43360\n") at rig.c:347

#1  0xb66e6520 in add2debugmsgsave ***@***.***=0x46360 <debugmsgsave2> "Connection opened from 127.0.0.1:43360\n") at rig.c:345
#2  0xb67b7b94 in icom_one_transaction ***@***.***=0x4d680, ***@***.***=21 '\025', ***@***.***=2, ***@***.***=0xb51fd510 "", payload_len=0, ***@***.***=-1232236336, data=0xb51fd5d8 "\025\002",
#3  0xb67b8cbc in icom_transaction (rig=0x4d680, ***@***.***=0x0, ***@***.***=21, subcmd=2, payload=0xb51fd510 "", ***@***.***=0xb51fd508 "", ***@***.***=0, ***@***.***=0xb51fd5d8 "\025\002",
    data_len=0xb51fd50c, ***@***.***=0xb51fd504) at frame.c:434
#5  0x00024a10 in rigctl_get_level ***@***.***=0x4d680, ***@***.***=0xb4800738, ***@***.***=0xb48005f8, interactive=0, ***@***.***=1, ***@***.***=0, ***@***.***=0x51a1c,
    at rigctl_parse.c:3301
#6  0x00029b8c in rigctl_parse (my_rig=<optimized out>, ***@***.***=0xb48005f8, ***@***.***=0xb4800738, ***@***.***=0x0, argc=<optimized out>, ***@***.***=0, sync_cb=<optimized out>, ***@***.***=0x14ff8 <mutex_rigctld>,
    interactive=<optimized out>, ***@***.***=1, prompt=<optimized out>, ***@***.***=0, vfo_opt=<optimized out>, ***@***.***=0x51a1c, send_cmd_term=<optimized out>, ***@***.***=13 '\r', ext_resp_ptr=<optimized out>,

mikaelnousiainen, 2022-05-26:
@mdblack98 Thanks, I will test this.

However, would this code still get stuck with a debug message that is long enough?

I'm having difficulties understanding how the loop algorithm really works 🤔

mdblack98, 2022-05-26:
No...if it hits a string that's too long to fit in 50% of the field it truncates the field.
Mike

mikaelnousiainen, 2022-05-26:
Ok, great. Your change seems to fix the issue. Will confirm tomorrow, once the test has been running all night.

closed 2022-05-27 by mikaelnousiainen