Hamlib/Hamlib #1050: FT-990 takes over 3 seconds to get frequency

mdblack98, 2022-06-03:
The FT-990 older ROM v1.2 sends back the whole 1492 bytes of its memory which takes 3.8s with Rigctl and almost 20s with WSJT-X. FYI - I've created modified ft990.c and ft990.h files to handle the differences in the CAT spec between these ROM versions for the FT-990 and recompiled Hamlib etc to get this working with my older version FT-990 ROM.

WSJT-X does its 20s fetch of 1492 bytes at initialisation time and every poll interval thereafter. I've extended the poll interval to 99s which is the longest it goes but it is very inconvenient to have a 20s rig blackout every 99s of course.

closed 2022-06-06 by mdblack98

m0ezp, 2022-06-07:
Many thanks Mike for making the changes and they certainly look more elegant than my version. 

There are still some issues as WSJTX falls over when going into TX with "Target VFO unaccessible". I'm thinking that there needs to be a full UPDATE ALL at rig_open time to bring the rig VFO frequency in. There is a difference on what is returned from UPDATE ALL between ROM 1.2 and ROM 1.3 as well as ROM 1.2 doesn't include the current_back 16 bytes in the structure. I've some personal matters on at the moment but I'll try to get details together properly and come back to you on those over the next week.

mdblack98, 2022-06-07:
I need a new debug log please.
Please place this file as described below
https://www.dropbox.com/s/t52ngcalsgnpm8m/wsjtx_log_config.ini?dl=0

	C:\Users\[username]\AppData\Local\WSJT-X
	The WSJT-X_Rigcontrol.log file will be in the same location
	
For Linux put it in  
	~/.config
	The WSJT-X_Rigcontrol.log file will be here:
	~/.local/share/WSJT-X
	
Restart WSJT-X and duplicate the problem.
Shut down WSJT-X

Then send me the WSJT-X_RigControl.log file
Mike W9MDB

m0ezp, 2022-06-07:
Thanks Mike,

It works in FT8 receive although you don't see the VFO frequency in WSJT-X.

Going into transmit for the log: I set Enable Tx - no message was sent, CAT was displayed on the rig and then in WSJT-X Enable Tx displayed as unset. I repeated this and it worked second time through.

There are two changes I made in my local build to ft990.h to deal with the difference between the UPDATE data returned between ROM v1.2 and v1.3. 
v1.2 length is 16 bytes different:-
`#define FT990_ALL_DATA_LENGTH           1492    /* M0EZP: FT990v12 uses 1492 not 1508 */`
because current_rear wasn't introduced until v1.3:-
`// Update Data Structure
typedef struct _ft990_update_data_t {
  unsigned char flag1;
  unsigned char flag2;
  unsigned char flag3;
  unsigned char channelnumber;
  ft990_op_data_t current_front;
  /* ft990_op_data_t current_rear; M0EZP: field not valid for FT990 ROM v1.2 */
  ft990_op_data_t vfoa;
  ft990_op_data_t vfob;
  ft990_op_data_t channel[90];
} ft990_update_data_t;`

I've also seen pressing Tune (before doing anything like Enable Tx) throwing a 
"Target VFO unaccessible" error.

[WSJT-X_RigControl.log](https://github.com/Hamlib/Hamlib/files/8854749/WSJT-X_RigControl.log)

Regards
David M0EZP

mdblack98, 2022-06-07:
Try this version
https://www.dropbox.com/s/7aejdv74nfjxr0q/libhamlib-4.dll?dl=0

There may a long delay at startup which we can address.
Mike W9MDB

m0ezp, 2022-06-08:
Hi Mike,
Thanks for sending. This one falls over with "Invalid parameter while getting current VFO frequency" unfortunately.
I wouldn't worry about the long startup I think v1.2 users would be okay with that if it only had to do that once. Polling is an issue as that gives a receive black out for 5s too.
Regards
David M0EZP

[WSJT-X_RigControl.log](https://github.com/Hamlib/Hamlib/files/8858594/WSJT-X_RigControl.log)

mdblack98, 2022-06-08:
Try this one please

https://www.dropbox.com/s/7aejdv74nfjxr0q/libhamlib-4.dll?dl=0

Mike W9MDB

m0ezp, 2022-06-09:
Hi Mike,
Thanks for the new one. It falls over with a similar error unfortunately but I'm sure you're close.
Regards
David M0EZ
[WSJT-X_RigControl.log](https://github.com/Hamlib/Hamlib/files/8867233/WSJT-X_RigControl.log)
P

mdblack98, 2022-06-09:
Try this one....we're getting close....

https://www.dropbox.com/s/7aejdv74nfjxr0q/libhamlib-4.dll?dl=0

Send me a log even if it works so I can check the timing.

Mike W9MDB

mdblack98, 2022-06-12:
> Hi Mike,
>
> Thanks, you can't be far off. This never got to receive unfortunately. CAT was on a lot longer for 39s and then it crashed with the old faithful "Invalid parameter while getting current VFO frequency".
>
> Have you still got ft990_op_data_t current_rear in the update_data? That was the 16 byte difference which was added with ROM v1.3 but its not in the fetch for v1.2.

Try this one...
I expect to see a 3 second delay on startup as it reads the entire block just once.
https://www.dropbox.com/s/7aejdv74nfjxr0q/libhamlib-4.dll?dl=0

Mike W9MDB

m0ezp, 2022-06-12:
Thanks Mike, sorry but there are still problems unfortunately. It’s sending multiple updates, I think VFOs, Op, etc, and doesn't return the rig frequency. There are a few timeouts and then it throws a Hamlib error. I can see its reading 1492 bytes which is good.

I’ve managed to make my local version stable which brings back the rig frequency and settles in 5 seconds. It only caters for the old ROM version of course and the way I did it is probably too inflexible to cover both ROM versions. I swapped update_op_data to read_flags in rig_open as even when I forced it to read all_data the get_freq that follows wanted to do that again. In get_update_data for all the update_data calls, other than read_flags, I’m overwriting ci with update_all_data and using a counter to only allow one fetch of everything and then ignoring all the future update_data calls. 

[WSJT-X_RigControl.log](https://github.com/Hamlib/Hamlib/files/8886588/WSJT-X_RigControl.log)

mdblack98, 2022-06-12:
I like your counter idea....this one does one call to get the data and uses it for the rest of the time.
Try this one

https://www.dropbox.com/s/7aejdv74nfjxr0q/libhamlib-4.dll?dl=0

m0ezp, 2022-06-13:
Thanks Mike, I'm sure you're getting closer but its not quite right yet. It took about 5 seconds before CAT came on. CAT then stayed on for seeming 20s or so but it didn't bring the rig frequency back and the waterfall never started for some reason although stations were coming in.

Your code looks much neater than mine but I've attached my ft990.c in case it helps.

Best, David

[ft990-M0EZP.zip](https://github.com/Hamlib/Hamlib/files/8888373/ft990-M0EZP.zip)

[WSJT-X_RigControl.log](https://github.com/Hamlib/Hamlib/files/8888335/WSJT-X_RigControl.log)

m0ezp, 2022-06-13:
Thanks Mike, I'm sure you're getting closer but this one threw the rig freq out of band.

[WSJT-X_RigControl.log](https://github.com/Hamlib/Hamlib/files/8892403/WSJT-X_RigControl.log)

mdblack98, 2022-06-13:
It doesn't look like you picked the UNI model

Mike W9MDB