Hamlib/Hamlib #1058: Command Reject Error when changing frequency on IC9700

mpippins, 2022-06-08:
### Version Info
* Testing done in Hamlib V 4.4
* Radio:
     * IC-9700 V 1.31
### Issue
I am unable to successfully execute  the *set_freq* command.  Hamlib returns:
     <code>icom_set_freq: set freq failed: Command rejected by the rig</code>

I can execute *set_freq* in versions 4.0 and 4.1.  It fails starting on version 4.2 through 4.4

### Log
Included is the debug output using the following command:
<code>"C:\Program Files (x86)\hamlib-w64-4.4\bin\rigctl.exe" -r COM8 -s 115200 -m 3081 -vvvvv V MainA f F 144000000 V SubA f F 440000000</code>
[log.txt](https://github.com/Hamlib/Hamlib/files/8857967/log.txt)

mdblack98, 2022-06-08:
Please try the latest
http://n0nb.users.sourceforge.net/

Mike W9MDB

 
 

   

log.txt

mpippins, 2022-06-08:
Appears to have the same results.  Everything executes but *set_freq*.

### Log:
Command:
<code>"C:\Program Files\hamlib-w64-4.5~git\bin\rigctl.exe" -r COM8 -s 115200 -m 3081 -vvvvv V MainA f F 144000000 V SubA f F 440000000</code>
[log.txt](https://github.com/Hamlib/Hamlib/files/8858048/log.txt)

mdblack98, 2022-06-08:
Do you have the latest 9700 firmware?https://www.icomjapan.com/support/firmware_driver/3251/

And might you be in memory mode?
Seems neither get_freq or set_freq doesn't work
Here's a set_freq getting rejected
write_block(): TX 12 bytes, method=20000    fe fe a2 e0 25 00 00 00 00 44 01 fd                 ....%....D..    read_string_generic called, rxmax=200 direct=1read_string_generic: minlen now 12read_string_generic(): RX 12 characters, direct=10000    fe fe a2 e0 25 00 00 00 00 44 01 fd                 ....%....D..    icom_one_transaction: DEBUG retval=12, frm_len=12, cmd=0x25read_string_generic called, rxmax=200 direct=1read_string_generic(): RX 6 characters, direct=10000    fe fe e0 a2 fa fd                                   ......           2:frame.c(352):icom_one_transaction returning(-9) Command rejected by the rig

And here's get_freq being rejected000    fe fe a2 e0 25 00 fd                                ....%..         icom_one_transaction: DEBUG retval=7, frm_len=7, cmd=0x25read_string_generic called, rxmax=200 direct=1read_string_generic(): RX 6 characters, direct=10000    fe fe e0 a2 fa fd                                   ......              4:frame.c(352):icom_one_transaction returning(-9) Command rejected by the rig

Mike W9MDB

mpippins, 2022-06-08:
When upgrading from HL 4.0 today, I was on IC9700 V1.3, as you linked.
The latest is V 1.31, that I am currently on with the same results as 1.3

mdblack98, 2022-06-08:
What about memory mode?  If the 9700 has been broken since Hamlib 4.4 I would think somebody would have mentioned it.

Or are you in satellite mode maybe?
The two standard VFOs are MainA and MainB
But I see MainA is getting rejected which doesn't make sense at all unless the rig is on the Sub VFO, satellite mode, or memory mode.

mpippins, 2022-06-09:
I am using it with Gpredict, and a piece of software I wrote for two wheel controllers to adjust Rx and RIT, using the Extended Response Protocol into the same rigctld socket.

I tested the commands in the 9700's normal mode, and they do work.  Except once Gpredict engages, it ends up in SAT mode.

I have used this for some time in SAT mode with V 4.0.  But once I switched to 4.4 the commands wouldn't work with Gpredict or my program.  Although, I use *Main* and *Sub* in my code, not *MainA* or *SubA* as my example logs showed, through the rigctl test.  They both seem to work in all versions, its just the *set_freq* function that doesn't work in SAT mode.

closed 2022-07-04 by mdblack98