Hamlib/Hamlib #1060: TT599 - seg fault when getting SWR reading

peterdulowsky, 2022-06-08:
The format of command is quite different between Orion 565 and Eagle 599. For Eagle 599:
@STF<fwd watts>R<ref watts*10><CR>

I guess fwd watts and ref watts must be read and then divided to obtain SWR reading.

See: http://www.tentec.com/wp-content/uploads/2016/05/tt_599_program_ref_v1.001.pdf 

https://github.com/Hamlib/Hamlib/blob/24c18938c284300ff4cab7a1b516058dffa4de36/rigs/tentec/orion.c#L1296

peterdulowsky, 2022-06-08:
A quick fix:

```
int tt565_get_level(RIG *rig, vfo_t vfo, setting_t level, value_t *val)
{
    int retval, cmd_len, lvl_len;
    char cmdbuf[TT565_BUFSIZE], lvlbuf[TT565_BUFSIZE];
    float fwd,ref;
```

```
       if (rig->caps->rig_model == RIG_MODEL_TT599)
        {
        /* in Xmit, response is @STF99R10<cr> 99 watts forward,1.0 watt reflected
            uu = fwd watts
            vv = rev watts x 10

           in Rcv,  response is @SRM16<CR> Indicates 16 dbm
            uuu = 000-100 (apx) Main S meter
        */

        if (lvlbuf[1] != 'S' || lvl_len < 5)
        {
            rig_debug(RIG_DEBUG_ERR, "%s: unexpected answer '%s'\n",
                      __func__, lvlbuf);
            return -RIG_EPROTO;
        }

        if (lvlbuf[2] == 'T')
        {
            ref = atof(strchr(lvlbuf + 2, 'R') +1)/10.0;      /* reflected power */
            fwd = atof(strchr(lvlbuf + 2, 'F') +1);           /* forward power */

            if (fwd == 0.0)
            {
                 val->f = 0.0;                                 /* no forward power */
            }
            else if (fwd == ref)                               /* too high SWR */
            {
                val->f = 9.99;
            }
            else
            {
                val->f = (1+sqrt(ref/fwd))/(1-sqrt(ref/fwd)); /* calculate SWR */
            }

            if (val->f < 1.0) { val->f = 9.99; }              /* high VSWR */
        }
        else { val->f = 0.0; }  /* SWR in Receive = 0.0 */

        }
        else
        {
        /* in Xmit, response is @STFuuuRvvvSwww (or ...Swwww)
            uuu = 000-100 (apx) fwd watts
                        vvv = 000-100       rev watts
                        www = 256-999  256 * VSWR
           in Rcv,  response is @SRMuuuSvvv
            uuu = 000-100 (apx) Main S meter
                        vvv = 000-100 (apx) Sub  S meter
        */
```
Tested with rigswr and seems to be working fine.

mdblack98, 2022-06-08:
What's your callsign so I can give you credit for this?

Mike W9MDB

peterdulowsky, 2022-06-08:
> What's your callsign so I can give you credit for this? Mike W9MDB >

OM4AEI

mdblack98, 2022-06-08:
It's been committed...please test.

Mike W9MDB

OM4AEI

closed 2022-06-19 by mdblack98