Hamlib/Hamlib #1061: TT599 noise reduction level get level fails (will apply to set levels too)

peterdulowsky, 2022-06-08:
The command format for TT599 is different: 

```
Noise Reduction
The EAGLE provides a Noise Reduction (nr) feature which can be turned on or off, and is also level selectable
The command prefix is *RMNN.
COMMAND: *RMNN<noise_reduction_value><CR>
< noise_reduction_value > = ON/OFF Selection / level
0 = Noise Reduction Off
1..9 = Noise Reduction ON and at the specified level
Examples: *RMNN1<CR> Turns on the Noise Reduction at level 1
*RMNN0<CR> Turns off the Noise Reduction
QUERY: ?RMNN<CR>
Examples: PC Query Eagle Response Meaning
?RMNN<CR> @RMNA7<CR> Noise Reduction on level 7
?RMNN<CR> @RMNA0<CR> Noise Reduction off
```

https://github.com/Hamlib/Hamlib/blob/24c18938c284300ff4cab7a1b516058dffa4de36/rigs/tentec/orion.c#L1643

mdblack98, 2022-06-14:
Have you tested the last version to see if this is fixed now?

peterdulowsky, 2022-06-14:
No, due to a work obligation I have not tested this change yet. Will aim to test it till end of the week.

peterdulowsky, 2022-06-14:
Took me less than I anticipated. The result is it mostly works however there are some non conformance between the rig behaviour and programming reference.

The CAT description says:
```
COMMAND: *RMNN<noise_reduction_value><CR>
< noise_reduction_value > = ON/OFF Selection / level
0 = Noise Reduction Off
1..9 = Noise Reduction ON and at the specified level
```

However, the rig behaves differently. The NR level can be set from the front panel between 1..10. When reading ?RMNN the response is @RMNN0 for NR off, and @RMNN1 to **@RMNN10** for NR on with appropriate level.

Sending *RMNN0 disables NR. NR symbol is not shown on the front panel. However, when NR button is pressed, NR symbol appears but the NR itself does not work. Sending ?RMNN returns @RMNN0. Small imperfection but who cares.

Sending *RMNN1-9 correctly sets NR level and enables it.

Sending *RMNN10 sets NR level 1. This is both confirmed via front panel and also via ?RMNN command. I guess the rig does not handle two characters in level number.

I would suggest to:
1. Assume 10 levels for NR and change the code accordingly ( /9.0 and *9.0 change to 10.0 in set_level and get_level)
2. Get NR level will correctly return the NR level between 0.0 (off) and 1.0 (level 10 set)
3. Setting the NR level will work correctly between values 0.0 (off) and 0.9 (level 9).
4. Level 10 will not be settable via CAT interface due to a bug in the rigs firmware

I guess 4. is not going to be an issue. This whole NR settings is not used very often.

mdblack98, 2022-06-14:
Try another pull...I think it should behave now...level 9 will set level 9 and should return 0.9 for the value.Level 10 manual set on the rig should return 1.0.
Mike W9MDB

peterdulowsky, 2022-06-14:
Unless I am missing something I think this is not going to work. 

[image]

The rig returns an integer 0..10 and that needs to be converted to float 0 .. 1.0 so division by 10 is needed for 599 Eagle. For Orion 565 it needs to stay as it was, i.e. divided by 9. Orion seems to be different to Eagle. However, I am unable to test since I have 599 Eagle only.

mdblack98, 2022-06-14:
Thanks -- fixed it

peterdulowsky, 2022-06-14:
Thank you. Will give it a go later today

mdblack98, 2022-06-17:
Is this all fixed now?

closed 2022-06-19 by mdblack98