Hamlib/Hamlib #1065: FT-897D communication issue: frequency may be set/get incorrectly

mikaelnousiainen, 2022-06-14:
We've discovered an issue with FT-897D where setting the frequency to `144295000` returned a frequency of `1442950` (divided by 100) right after setting it. This has happened a couple of times, but we don't have proper debug log info from Hamlib yet, so we need more testing, but filing this here if it is a known issue. Serial speed has been lowest, 4800 bps. Otherwise CAT control has been reliable at all times.

After the frequency issue, Hamlib reported Communication timed out (error code -5) for a short time.

mdblack98, 2022-06-14:
Need debug please...

mikaelnousiainen, 2022-07-12:
I will try to get back to testing this in a couple of days.

mikaelnousiainen, 2022-07-28:
The fix (for 100 MHz frequencies) had to be reverted, it is not correct. The issue has not surfaced after the first time. It looks like this is not an issue in Hamlib, but may have been caused by some serial port communication glitch or similar.

closed 2022-07-28 by mikaelnousiainen