Hamlib/Hamlib #1072: Kenwood TS-480 issues: split frequency not set (always sets VFOA) and RFPOWER not working

mikaelnousiainen, 2022-06-21:
Kenwood TS-480 issues:

* Setting split frequency always sets VFOA frequency
* RFPOWER setting not working anymore

I'm looking into these issues, hopefully fixing them this week.

mikaelnousiainen, 2022-06-21:
I've got a fix for RFPOWER and some questions about how split freq set (in rig.c, withOUT set_split_freq caps function) should work -- there might be a bug there. Will post my findings shortly.

mikaelnousiainen, 2022-06-21:
RFPOWER fix in #1073

mikaelnousiainen, 2022-06-21:
@mdblack98 Regarding split issue with Kenwood:

* It seems that Kenwood rigs do not implement `set_split_freq()` in rig caps so it is emulated in `rig_set_split_freq()`
* When split is NOT enabled, `set_split_freq` seems to simply alter VFOA if current VFO is VFOA
* When split IS ENABLED (after `set_split_vfo` for VFOB), `set_split_freq` still alters VFOA, but `get_split_freq` gets VFOB frequency

Based on the following debug logs, it looks like `rig_set_split_vfo()` should set `state->tx_vfo` correctly to VFOB, but `rig_set_split_freq()` still somehow ends up with `VFOA` here: https://github.com/Hamlib/Hamlib/blob/0b6161c8e44f5c31680b41fb87e34eeb81ad039d/src/rig.c#L3856

Kenwood TS-480 sets the freq using: https://github.com/Hamlib/Hamlib/blob/0b6161c8e44f5c31680b41fb87e34eeb81ad039d/src/rig.c#L3897 <- Uses `tx_vfo`, but `vfo_fixup` (here https://github.com/Hamlib/Hamlib/blob/0b6161c8e44f5c31680b41fb87e34eeb81ad039d/src/rig.c#L3883) figures out VFOB correctly, storing the result in `vfo`-> Should the call use `vfo` instead of `tx_vfo`? 

Debug log for `set_split_vfo`:

```
rigctl_parse: cmd=set_split_vfo handle=8
mutex_rigctld: client lock engaged
rigctl(d): S 'currVFO' '1' 'VFOB' ''
rigctl_parse: vfo_opt=0
                     208:rigctl_parse.c(2981):rigctl_set_split_vfo entered
rig_parse_vfo called
rigctl_set_split_vfo(3000): rx_vfo = currVFO, tx_vfo = VFOB
                     209:rig.c(4683):rig_set_split_vfo entered
rig_set_split_vfo: rx_vfo=currVFO, split=1, tx_vfo=VFOB, cache.split=0
vfo_fixup:(from rig_set_split_vfo:4724) vfo=VFOA, vfo_curr=VFOA, split=1
vfo_fixup:(from rig_set_split_vfo:4725) vfo=VFOB, vfo_curr=VFOA, split=1
vfo_fixup: final vfo=VFOB
rig_set_split_vfo: final rxvfo=VFOA, txvfo=VFOB, split=0
rig.c(4734) trace
rig.c(4757) trace
kenwood_set_split_vfo called VFOA,1,VFOB
                     210:rig.c(4850):rig_get_split_vfo entered
rig_get_split_vfo: cache check age=951ms
rig_get_split_vfo: cache miss age=951ms
rig.c(4901) trace
                     211:kenwood.c(1562):kenwood_get_split_vfo_if entered
                     212:kenwood.c(1084):kenwood_get_if entered
kenwood_safe_transaction called
kenwood_transaction called
kenwood_transaction: cmdstr = IF
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1
write_block(): TX 3 bytes, method=2
0000    49 46 3b                                            IF;             
read_string_generic called, rxmax=128 direct=1
read_string_generic: minlen now 38
read_string_generic(): RX 38 characters, direct=1
0000    49 46 30 30 30 30 33 37 30 32 30 30 30 20 20 20     IF00003702000   
0010    20 20 20 30 30 30 30 30 30 30 30 30 30 31 30 30        0000000000100
0020    30 30 30 38 30 3b                                   00080;          
kenwood_transaction: read_string(len=38)='IF00003702000      000000000010000080;'
kenwood_transaction: returning RIG_OK, retval=0
kenwood.c(612):kenwood_transaction returning2(0) 
kenwood.c(682):kenwood_safe_transaction returning2(0) 
                     212:kenwood.c(1086):kenwood_get_if returning(0) 
kenwood.c(1628) trace
kenwood_get_split_vfo_if: priv->tx_vfo=VFOA, split=0
                     211:kenwood.c(1681):kenwood_get_split_vfo_if returning(0) 
rig_get_split_vfo: cache.split=0
                     210:rig_get_split_vfo: elapsed=23ms
                     210:rig.c(4914):rig_get_split_vfo returning(0) 
kenwood_set_split_vfo(1386): tsplit=0, split=1
kenwood.c(1400) trace
kenwood.c(1406) trace
kenwood_transaction called
kenwood_transaction: cache invalidated
kenwood_transaction: cmdstr = FR0;FT1
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1
write_block(): TX 8 bytes, method=2
0000    46 52 30 3b 46 54 31 3b                             FR0;FT1;        
write_block(): TX 3 bytes, method=2
[... 85 lines cut ...]
0000    49 46 3b                                            IF;             
read_string_generic called, rxmax=128 direct=1
read_string_generic: minlen now 38
read_string_generic(): RX 38 characters, direct=1
0000    49 46 30 30 30 30 33 37 30 33 30 30 30 20 20 20     IF00003703000   
0010    20 20 20 30 30 30 30 30 30 30 30 30 30 31 30 30        0000000000100
0020    31 30 30 38 30 3b                                   10080;          
kenwood_transaction: read_string(len=38)='IF00003703000      000000000010010080;'
kenwood_transaction: returning RIG_OK, retval=0
kenwood.c(612):kenwood_transaction returning2(0) 
kenwood.c(682):kenwood_safe_transaction returning2(0) 
                     242:kenwood.c(1086):kenwood_get_if returning(0) 
kenwood_get_vfo_if: priv->tx_vfo=VFOB
                     241:kenwood.c(1746):kenwood_get_vfo_if returning(0) 
                     240:rig_get_vfo: elapsed=17ms
                     240:rig.c(2819):rig_get_vfo returning(0) 
                     239:rig.c(2642):rig_set_vfo returning(0) 
                     238:rig_set_freq: elapsed=28ms
rig.c(1881):rig_set_freq returning2(0) 
                     238:rig_set_split_freq: elapsed=29ms
rig.c(3909):rig_set_split_freq returning2(0) 
                     238:rigctl_parse.c(2815):rigctl_set_split_freq returning(0) 
```

```

mikaelnousiainen, 2022-07-05:
@mdblack98 Putting my status update here too:

The RFPOWER bug is fixed, but the VFO is not. I tested using vfo instead of tx_vfo for set_freq like I suggested but it did not fix all of the behavior. This needs more work, as I don't understand yet fully how the Kenwood backends should work without actual set_split_freq being implemented but only being emulated.

Would you have time to take a look at the debug logs here?

mikaelnousiainen, 2022-07-08:
@mdblack98 Setting the split frequency works now correctly, but *ONLY* when SPLIT is enabled. When SPLIT is DISABLED, setting the split freq seems to alter the "current" VFO, which is kind of weird -- and should not be the case as VFO freq is targetable in Kenwood rigs.

I still have to do some testing later today (and maybe verify Icom split still works ok), will confirm here once this is fully tested.

mdblack98, 2022-07-08:
Calling rig_set_split_freq when split is not turned on has several possibilities.  I believe the intent was that it would not be called if split was not on.

The current vfo is what will transmit if split is not on -- the function probably should have been called rig_set_tx_freq but you will get that effect if you use RIG_VFO_TX as the vfo in the function call.

There are several ways to treat this....

#1 If split is not on set the current vfo (as that is the TX vfo at this state of the rig) -- current behavior
#2 If split is not on return an error
#3 If split is not on set the vfo requested
#4 If split is not on set the TX vfo (which takes current vfo into account) and ignore the requested vfo -- this is reverse split if you then turn split on after this.

I think the current behavior is the best  as you can just call this function with RIG_VFO_TX and whatever the TX vfo is will get set.

Now the problem comes in when the VFO gets selected for some rigs as that changes the split direction and why this function takes that into account.

I think the current behavior is the best we can get.

One should be able to do this

rig_set_split(rig, RIG_VFO_A, 1, RIG_VFO_B);  // set split as you want it
rig_set_freq(rig, RIG_VFO_RX, freq);
rig_set_freq(rig, RIG_VFO_TX, splitfreq);

I believe that should work on all rigs now as VFO_A/B are mapped to whatever the rig uses (e.g. Main/Sub)

Mike W9MDB

mikaelnousiainen, 2022-07-08:
@mdblack98 Yeah, maybe it's good to leave this as it is, because everything works correctly when split is enabled. Everything looks good with my tests.

closed 2022-07-08 by mikaelnousiainen