Hamlib/Hamlib #1076: Provide Support for Gemini DX1200 and HF-1K Amps

RafalLukawiecki, 2022-07-07:
I use Gemini DX1200 and a friend has HF-1K. It supports comms using a simple TCP/IP protocol. I am new to hamlib having only used the command line tools. I am not sure what kind of effort is required to add support of a new amp, but happy to help. Thank you.

73 Rafal EI6LA

mdblack98, 2022-07-07:
Do you have a manual for the protocol?  I don't see any such documentation on their website.
TCP/IP is supported by default for all devices in Hamlib.  Just use addr:port like this example "-r 192.168.1.100:7000" to connect to port 7000 on that IP address. 
Mike W9MDB

RafalLukawiecki, 2022-07-07:
Thanks, Mike. Yes, this PDF describes the current API protocol.

[NetworkConnectionProtocol_3.pdf](https://github.com/Hamlib/Hamlib/files/9063647/NetworkConnectionProtocol_3.pdf)

73 Rafal EI6LA

vk4tux, 2022-07-07:
This project may be of interest ; 
https://www.oe3ide.com/gemini-multi-control-by-oe3ide/

RafalLukawiecki, 2022-07-07:
Thanks for sharing about the OE3IDE. Is there something like that available for Unix/Linux/macOS? If not, I suspect getting hamlib to do it would be a good option.

N0NB, 2025-07-31:
Closing per request.