Hamlib/Hamlib #1094: Cannot set frequency on FT857D using Python bindings

chrism0dwk, 2022-07-26:
Hello All,

As part of an effort to look at a local source of RFI, I'm trying to write a script that sweeps my transceiver (FT857D) across a range of frequencies, recording the signal strength at each frequency step.  Having written programmes with `hamlib` in the past in C, I thought I'd give the Python bindings a go.

Interestingly, although the script appears to connect to my rig, I cannot get consistent results when continuously polling the transceiver to update the frequency and read the S meter.  The guts of my approach are as follows
```python
def get_sig_strength(rig, frequency):
    logging.info(f"Setting frequency {frequency}")
    if rig.set_freq(RIG_VFO, int(frequency)) != Hamlib.RIG_OK:
        logging.warning("Cannot set frequency")
    sig_level = rig.get_level_i(Hamlib.RIG_LEVEL_STRENGTH)
    return sig_level

def freqsweep(rig, low, high, step):
    """Perform a frequency sweep on `rig`
       between `low`Hz and `high`Hz in steps
       of `step`Hz.
       :returns: a `numpy` matrix of signal strength versus frequency
    """
    freqs = np.arange(start=low, stop=high, step=step)
    return [[f, get_sig_strength(rig, f)] for f in freqs]
```
with the full code [here](https://gitlab.com/-/snippets/2377116).

It feels like Hamlib is not waiting long enough between successive commands for the rig to respond, but other than instrumenting my code with a bunch of `sleep()` statements (which work unreliably for reading `RIG_LEVEL_STRENGTH`, not for setting the frequency) I'm short for ideas.

A general question: many of the Python bindings don't work as expected given the Hamlib C++ doc, particularly wrt return codes -- above is a guess at error checking.  Is there a separate doc for the Python bindings to set me on the right path? 

73,

Chris

OS: Linux Mint 20.3
Python 3.8
Hamlib 3.3 (`python3-libhamlib2/focal 3.3-10build1 amd64)
Serial cable: USB-A to Yaesu mini-DIN connector, fine at 38400 baud with `rigctl` etc)
Rig: FT857D (hamlib ID 122)

dforsi, 2025-04-20:
Chris,

have you had a chance to try a version newer than Hamlib 3.3?

Can we close this issue or is still relevant for you? You can reopen it or open a new one at any time.

N0NB, 2025-07-22:
Done

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

closed 2025-07-22 by N0NB