Hamlib/Hamlib #1098: Rig type for QRP Labs QCX series

ncw, 2022-08-04:
I'm currently preparing a patch for hamlib for the QRP Labs QCX series

These use the Kenwood TS480 protocol and it is by an large compatible.

However sending morse leaves long pauses between each sent item because of this code

https://github.com/Hamlib/Hamlib/blob/36c777e8bbd61600b89253656d1c6263206310cd/rigs/kenwood/kenwood.c#L4968-L4975

What I want is hamlib to treat the QCX like an Elecraft rig for the purposes of sending morse. The QCX doesn't work as an Elecraft rig though - it fails to initialise - for reasons I haven't worked out yet.

https://github.com/Hamlib/Hamlib/blob/36c777e8bbd61600b89253656d1c6263206310cd/rigs/kenwood/kenwood.c#L4961-L4966

I've made a patch adding a QCX rig type, mainly just to fix this one issue, but maybe there is a better way?

Here is a preview https://github.com/ncw/Hamlib/commit/fa277d37e0cb5d275840cd3ca1a88ca4c3c9e7e0

mdblack98, 2022-08-04:
I think this is the correct approach.
What about the KY command does it not like?  The padding?
Mike W9MDB

 
 

https://github.com/Hamlib/Hamlib/blob/36c777e8bbd61600b89253656d1c6263206310cd/rigs/kenwood/kenwood.c#L4968-L4975

https://github.com/Hamlib/Hamlib/blob/36c777e8bbd61600b89253656d1c6263206310cd/rigs/kenwood/kenwood.c#L4961-L4966

ncw, 2022-08-04:
> I think this is the correct approach.

Great - I'll continue.

> What about the KY command does it not like?  The padding?

Yes, the padding causes a looong delay between sends which isn't ideal for operating in a contest!

mdblack98, 2022-08-04:
They need to fix their firmware.  The KY command is fixed length and the trailing spaces are not supposed to be sent.
From the TS-480 manualThe fixed 24-byte length is used for the P2 parameter.“ ” (space) character must be used for the unused characters.These space characters will not be converted.
Mike W9MDB

ncw, 2022-08-04:
> They need to fix their firmware.  The KY command is fixed length and the trailing spaces are not supposed to be sent. From the TS-480 manualThe fixed 24-byte length is used for the P2 parameter.“ ” [...]

That is perfectly true... However the Elecraft rigs use a variable length KY command as you can see in `kenwood.c` and I think it is that the QCX rigs are based on.

https://github.com/Hamlib/Hamlib/blob/36c777e8bbd61600b89253656d1c6263206310cd/rigs/kenwood/kenwood.c#L4961-L4966

So defining a QCX rig type and adding it to that case wil lfix the QCX morse sending.

mdblack98, 2022-08-04:
True...but they claim TS-480 compatible don't they?  Elecraft never claimed that.
Mike W9MDB

 
 

https://github.com/Hamlib/Hamlib/blob/36c777e8bbd61600b89253656d1c6263206310cd/rigs/kenwood/kenwood.c#L4961-L4966

ncw, 2022-08-04:
> True...but they claim TS-480 compatible don't they?  Elecraft never claimed that.

Also true...

I guess this could be fixed in the QCX by ignoring trailing spaces in the KY command...

I think I prefer patching hamlib though as it enables us to be more precise about the QCX's capabilities - ie it is CW only, 5W has a 300 Hz filter only etc...

What do you think?

mdblack98, 2022-08-04:
Oh yeah...let's put the rig in as you are planning.
Mike

ncw, 2022-08-04:
> Oh yeah...let's put the rig in as you are planning.

:+1:

ncw, 2023-04-09:
Adding a new rig type for QCX was done here: 64c566cd14b743c91563c986cd70c0a3be431b9d

This PR https://github.com/Hamlib/Hamlib/pull/1269 fixes the final part of the morse sending.

ncw, 2023-04-09:
I'll close this now as the new rig type was added as per the title of this issue.

closed 2023-04-09 by ncw