Hamlib/Hamlib #1104: FTDX101 takes very long to time out when powered off

mikaelnousiainen, 2022-08-21:
FTDX101 rig open (and any command) takes VERY long (30-60s) to time out when the rig is powered off.

I'm investigating this and will post either a solution or logs.

I will check whether other rigs (like FT-991A that I have) suffer from the same issue -- this is possible, as both use newcat protocol,

mdblack98, 2022-08-22:
Things can be sped up a bit by changing line 1246 in rig.c

From
rs->rigport.retry = 1;
To
rs->rigport.retry = 0;

That should be an OK change.

Have to be careful about aborting any/all the startup commands due to timeouts.  Might be reasonable to do.

But what's the goal of having it respond quicker when the rig is off?

mikaelnousiainen, 2022-08-23:
Solved in #1106.

It was not really about retries, but more about EPOWER response causing rig to be re-opened, which was super slow because of Yaesu newcat ID queries for every single command.

mikaelnousiainen, 2022-08-23:
Also regarding the "goal" here: I'd expect to be able to power on a rig without waiting for 2-3 minutes :) Also, commands should time out as fast as possible (but not faster :D).

closed 2022-08-23 by mikaelnousiainen