Hamlib/Hamlib #1111: Add TCP ability

mdblack98, 2022-08-30:
Add ability to connect to Hamlib as a TCP client -- e.g. N1MM now can connect to a rig via TCP so having a TCP passthru capability would allow N1MM work with WSJTX/JTDX without additional software.
If HRD adds TCP connection capability then HRD should also be able to share the rig with other programs too.

mikaelnousiainen, 2022-08-30:
@mdblack98 What kind of protocol is the one N1MM would use over TCP? Any docs?

mdblack98, 2022-10-11:
It would just be whatever rig you have selected over TCP instead of serial.  More rigs/packages are starting to add this.
Hamlib can already connect to any rig via TCP if it's supported like Thetis does now.  You just use "-r 127.0.0.1:13013" for example.  So this allows remote operation, multiple clients, and faster than serial.
Mike