Hamlib/Hamlib #1120: PTT Configuration for IC-7100

BobSegrest, 2022-09-25:
The Icom-7100 USB interface only creates 1 serial port.  This port is used by the Hamlib daemon to control rig frequency and other settings.  How do I configure Hamlib for PTT on my IC-7100?

Bob
KO2F

mdblack98, 2022-09-25:
IC-7100 supports CAT PTT -- are you saying that isn't working?
Or are you wanting to do RTS or DTR keying?
If you want to do RTS/DTR keying add this switch to rigctld to allow it to share the com port.
--set-conf=ptt_share=1

Mike W9MDB

BobSegrest, 2022-09-25:
Hi Mike, Thank you for the quick response!

I am trying to get Pat Winlink working on a Windows 10 system.  Pat uses Hamlib for rig control.  When I tell Pat to connect, it sets the rig frequency (this is working) and indicates it is transmitting.  But the rig never keys up (no little red light).

By contrast, Winlink Express (which does not use Hamlib or Flrig) has no problems.

I'm hoping that I have just missed some little something I have to configure on the Hamlib side, as there is nothing PTT specific in Pat Winlink.

I'm starting Hamlib with the command:  rigctld -m 3070 -r com5 -s 19200

Is there a way to test the PTT using rigctl ?

Bob, KO2F

mdblack98, 2022-09-25:
With rigctl by itself.
rigctl -m 3070 -r com5 -s 19200

T 1
Also with rigctld running
rigctl -m 2
T 1
Mike

mdblack98, 2022-09-25:
And what versione of Hamlib?

rigctl --version

Mike W9MDB

BobSegrest, 2022-09-25:
Hi Mike,

Thank you!  Using the T 1 command does indeed assert the PTT and a following T 0 command drops the PTT.

I suspect this means Hamlib is working correctly and I'm going to have to pursue the Pat Winlink side further.

And, I am using the current Hamlib 4.4 for Windows (hamlib-w64-4.4.exe) kit.

Bob, KO2F

mdblack98, 2022-09-25:
Is that testing rigctl with rigctld too?
There was a bug which I'm not sure if it affected that.You can try the latest from here
http://n0nb.users.sourceforge.net/

Mike

BobSegrest, 2022-09-25:
Yes,

I tried with and without rigctld running.  Seemed to work with both configurations.

Bob
KO2F

closed 2022-09-25 by BobSegrest