Hamlib/Hamlib #1121: Add DCD Output Support for CM108

wb9vxy, 2022-09-26:
The PAUL board from KitsforHams has a Yellow DCD Led attached to CM108 PIn 15 GPIO4

I am using the board for Direwolf and it would be great if I could configure this LED to show
decoded packets (DCD output in Direwolf)

Software that allows access to the other GPIO pins would likely be useful to someone 
in the future as well.

Thanks,
WB9VXY

wb9vxy, 2022-09-26:
> Are you proposing some new API call?
> We already have GPIO support in hamlib but its centered on PTT
> control and not anything "generic".
> How do you envision doing this? You're not really talking about CAT
> control here but some "data monitoring" capability which seems to me
> to be outside the scope of Hamlib without add some new API.

I would think that something like

DCD CM108  {pin number} in Direwolf would be nice.

T3

mdblack98, 2022-09-26:
I could see something more like
gpio 2 1
Which would turn on pin 2 for the connected gpio device.
gpio 2 0
To turn it off.
Would be a new api call in addition to a new rigctl command.

wb9vxy, 2022-09-26:
Probably even better why to do it.

T3

N0NB, 2025-07-31:
@wb9vxy This has been implemented, have you had a chance to test it?