Hamlib/Hamlib #1128: ./configure doesn't exist

ckuhtz, 2022-10-21:
the instructions in README and INSTALL refer to the three step with ./configure.  That file doesn't exist after cloning the repo.

Update instructions or repo?

ckuhtz, 2022-10-21:
Is the missing ./configure intended? happy to submit pr to ref autoconf first.

ckuhtz, 2022-10-21:
and if you run autoconf you get this:

```
configure.ac:33: error: possibly undefined macro: AM_INIT_AUTOMAKE
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:92: error: possibly undefined macro: AM_CONDITIONAL
configure.ac:111: error: possibly undefined macro: AM_PROG_CC_C_O
configure.ac:303: error: possibly undefined macro: AC_MSG_ERROR
configure.ac:566: error: possibly undefined macro: AM_PATH_PYTHON
configure.ac:580: error: possibly undefined macro: AM_PYTHON_CHECK_VERSION
```

ckuhtz, 2022-10-21:
#1129 #1130

N0NB, 2022-10-21:
As `configure` is a generated script and not a source file, it is not included in the Git repository.  The `bootstrap` script correctly runs the needed Autotools programs.

The README.developer file should contain the information needed.  Also, README.betatester is good background as well.

mdblack98, 2022-10-21:
What OS are you building on?  I can't reproduce the undefined macros problem.

closed 2022-10-21 by mdblack98

ckuhtz, 2022-10-21:
> What OS are you building on? I can't reproduce the undefined macros problem.

The bootstrap script is a mandatory prereq for the build instructions in README and INSTALL. The macro error is a result of autoconf failing without being run through the bootstrap script. Generally, you'd run autoconf to create configure from configure.ac, but that's not the case in this repo.  README.* is a poor place to place this detail in when the instructions in README and INSTALL describe the build from source process in three steps and are incomplete.

ckuhtz, 2022-10-21:
> As `configure` is a generated script and not a source file, it is not included in the Git repository. The `bootstrap` script correctly runs the needed Autotools programs.
> 
> The README.developer file should contain the information needed. Also, README.betatester is good background as well.

bootstap belongs mentioned in the build instructions in README and INSTALL and not some random README.*.  The first step in the build instructions cannot succeed as written.

N0NB, 2022-10-21:
I hate to be rude but you've come barging into the project yesterday without introduction proclaiming we're wrong when the project has followed the Best Practices outlined by the Gnu Project and the Gnu Autotools project for _decades_.  The documentation is correct for the way this project is structured and maintained.  While I think Mike's addition to the INSTALL file is not necessary I am willing to leave it be as he wrote it.

Unlike other projects you may have encountered, Hamlib's Git repository is not distribution ready nor is it ready to simply compile and go.  The reasons are many and technical but a quick summary is that not just the `configure` script but several support files are required as well.  These files will change depending on what version of Autotools was used to generate them.  Someone on Debian is going to have a different set of files from someone on Arch, and for anyone developing on MinGW they will all be different yet again.  That is just for Autoconf, Automake requires all of the `Makefile.in`s be present and `libtool` has its own pile of files it needs present and all of those can vary depending on the Autotools version and whatever patches a distribution has applied.  It just isn't logical to maintain a project of this scope will all of the build system files changing from developer to developer.  If we didn't do it the way we do applying a simple patch from one developer to the next would be a nightmare.

What you fail to understand is that `bootstrap` is not included in a release tarball nor should it.  The README and INSTALL files are included in the source tarbal.l as are others for completeness, but those two are required by Autotools to create a release source tarball and are intended to only explain what is needed to build Hamlib from that release source tarball.  I invite you to compare the files contained in the latest daily source tarball available from https://n0nb.users.sourceforge.net/ found in the first section titled, "Hamlib Git master branch source snapshot" to what you find in the Git repository.  We intend for the Git repository to be a source only archive containing no files generated by the build system.

I hope this explanation is useful.